evcc-io/evcc #5441: 1/3-Phasen Umschaltung geht trotz Sponsoring nicht

Naiki92, 2022-12-11:
Da hat man mal wieder Sonne und die 3-Phasenumschaltung geht trotz hinterlegten Sponsoring nicht :/ 
Ich nutze den GO-E V3 Lader. Neustart, neu hinterlegen des Token funktionieren nicht und die Möglichkeit zum Phasenwechsel wird in der UI nicht geboten.

[image]

DerAndereAndi, 2022-12-11:
Eine Umschaltmöglichkeit in der UI gibt es nicht.

Hast du in der Konfiguration des Loadpoints auch `phases: 0` stehen? Siehe Doku: https://docs.evcc.io/docs/reference/configuration/loadpoints/#phases

premultiply, 2022-12-11:
Es gibt keinen Knopf in der UI zum Phasenwechsel.
Wenn die Konfiguration korrekt ist und mehr als 4,6 kVA Überschuss(!) abzgl. momentaner Ladeleistung vorhanden sind wird nach entsprechener Wartezeit umgeschaltet.
All dies lässt sich aus dem riesigen Screenshot nicht ablesen.

closed 2022-12-11 by premultiply