evcc-io/evcc #5457: Add minSocEnergy setting

andig, 2022-12-13:
A regular complaint of users is that their EV unexpectedly charges due to `minSoC` in combination with vehicle api outages. While we can't really fix that, I suggest to introduce an additional `minSocEnergy` setting that can be used to limit the amount of energy used for `minSoc`.

Adding the `minSocEnergy` settings will have the following properties:

- energy amount in kWh to limit `minSoc charging to`
- applicable to loadpoint and vehicle, where loadpoint is catch-all for undetected vehicles and vehicle for offline and default vehicles
- only active if `minSoc` is configured to a non-zero value (this is needed to identify that `minSoc` charging is to be used at all- needs review)

I'd also propose to start `minSoc` charging only 15 minutes _after_ connecting the vehicle to give the Api some time to settle and to allow a `soc` value to be retrieved. I propose to name the setting `minSocEnergy` (and not `minEnergy` do clearer communicate that it still requires `misSoc` to be set. Needs be verified.

naltatis, 2022-12-13:
Puh, this sounds complicated and I guess it's very hard to communicate this setting to new users. What's an appropriate kWh value to choose? What guidance would we give a user to pick a good value here? We would have to explain the underlying problem (unreliable vehicle API) before the settings makes sense for users. I'd rather think about better default behaviour and/or better communication first.

I really like the idea of introducing a fixed 15min grace period time to give the API some time. This way we would also handle short API outages or strange vehicle behaviour in low SoC ranges. For users that own a vehicle with flaky API support or have bad connectivity at your charging point we should recommend not to use MinSoC instead of giving them fallback controls.

Regarding communication: I think we could also improve the SoC information in the UI. Currently we show the last known SoC of the vehicle without any context. If evcc was unable to refresh the value for a given time we should indicate that (icon, comment, color, ...). This makes it easier to understand why e.g. MinSoC does not work as expected. I'm thinking of something like a `lastUpdated` timestamp or a `refreshFailed` flag in the API.

andig, 2022-12-13:
Default: `20kWh` which is roughly 100km

> We would have to explain the underlying problem (unreliable vehicle API)

To be honest: we are already doing that all then time when people complain about minsoc! We could even chose to _not_ document this at all and use it as internal fallback if that is a concern.

premultiply, 2022-12-13:
We already have the capacity. It would be no problem to calculate the minSoC as kWh as fallback.

But main feature would be a reliable and fast detection for valid up-to-date api values.

andig, 2022-12-13:
One scenario I‘d like to address is the unidentified vehicle. How would that give us a battery size? We‘d still need fallback max energy, at least for the loadpoint.

thierolm, 2022-12-13:
> One scenario I‘d like to address is the unidentified vehicle. How would that give us a battery size? We‘d still need fallback max energy, at least for the loadpoint.

We could persist the maximum charged energy of a single load session per vehicle VIN in the sqllite settings.
Over the time we would get a close estimate of the battery size.

andig, 2022-12-13:
> Over the time we would get a close estimate of the battery size.

To the battery size of what considering multiple vehicles? What do we do when that history is not available.

> One scenario I‘d like to address is the unidentified vehicle. 

To add to what I said above: the opposite case of an _identified_ vehicle that then _doesn't_ provide its soc is probably pretty scarce, so we really wouldn't gain much except for default vehicles.

premultiply, 2022-12-13:
`minSoC` is only available for identified vehicles.

andig, 2022-12-13:
Please read my post:

> applicable to loadpoint and vehicle, where loadpoint is catch-all for undetected vehicles and vehicle for offline and default vehicles

premultiply, 2022-12-13:
Offline vehicles do have capacity property.
All other types are not relevant to minSoC.

andig, 2022-12-18:
No consensus

closed 2022-12-18 by andig