evcc-io/evcc #5460: Increase visibility of vehicle SoC limit marker in GUI

RTTTC, 2022-12-13:
When car is charged up to the vehicle limit - limit becomes "invisible".
Could this be made more visible? Like a thin vertical line that extends above and below the SoC bar?

naltatis, 2022-12-29:
I think it's not a big issue and I don't see a simple and good solution here. The current implementation as a "gap" in the progress bar is deliberately very subtle. When charging is in progress it gets quite clear that the light green area (energy that needs do be added) does not extend across the gap. I think replacing the gap with a second coloured bar (like the evcc soc limit) would create more problems and confusion.

I'd rather work towards a solution where we eliminate the one of the limits. E.g. having the slider control the vehicle limit directly. Keeping this reliably in sync is not easy, but from a users perspective I think this is where we should work towards.

I'll close this issue for now.

closed 2022-12-29 by naltatis

RTTTC, 2023-03-20:
I was talking about making "the gap" itself more visible, such as gap being orange color or something like that.
NOT adding another bar, that would be a bad idea :)
Currently, as the car is charged up to the vehicle limit, gap becomes invisible, and visually it is less clear why charging stopped.
Gap is only visible when battery charge is less than vehicle limit.