evcc-io/evcc #5463: Elli charger connect on max 7,4kW charging capable car

roto11r, 2022-12-14:
Hi I'm not sure if it's a request or a bad configuration on my evcc server:

My Elli connect charger is 11kW capable but my VW e-Up can only accept 7,4 kW at max 

EVCC is showing 11kW when charging at maximum speed, probably putting an error with the total kWh chargeds in every session.

Have i misconfigured something? Perhaps the Evcc team has to add an option in the configuration menu with each car added ?

Also can it be possible to add a possibility to start heating the car from the evcc menu at about 30 min before the end of a charging session ? ( Like we can do with we connect )

Thanks for reading and thanks for this awesome development

Francisco from Switzerland

RTTTC, 2022-12-14:
I believe E-Up has a 2 phase charger 16A ?
How do you have it configured in evcc? Probably as a 3 phase car.

roto11r, 2022-12-14:
I don't know how is the internal charger except the car accept 7.4 kW. My Elli charger is wired in 3 phases

RTTTC, 2022-12-14:
Check your configuration file, find your vehicle, and change phases to 2

Sillium007, 2022-12-15:
Log file could help a bit since we would see what is reported by the charger.

And yes eUp has a 2 Phase Charger

premultiply, 2022-12-15:
Looks like your charger has no integrated metering device so power is only "estimated" by `maxCurrent * loadpoint.phases * 230V`.

Your vehicle configuration needs `phases: 2` if your vehicle is limited to 2p as evcc has no chance to detect it.

Anyway its always better to use a full-featured charger including a metering device.

closed 2022-12-15 by premultiply