evcc-io/evcc #5478: Individual pv/battery logging on mqtt for each defined meter

thomas-hutterer-tik, 2022-12-16:
**Is your feature request related to a problem? Please describe.**
Integrating evcc with the Watt Analytics platform we want support multiple pv and battery meters to be defined in Evcc and their data being stored, analyzed and displayed in the Watt Analytics Platform individually per meter not only in the aggregated form. 

**Describe the solution you'd like**
on the mqtt messages in addition to the total values
evcc/site/pvPower 0
evcc/site/batteryPower -204
evcc/site/batterySoC 24
also the individual values per meter like:
evcc/site/pv/1/power 0
evcc/site/pv/2/power 0
evcc/site/pv/3/power 0
evcc/site/battery/1/power -204
evcc/site/battery/1/SoC 24
evcc/site/battery/2/power -204
evcc/site/battery/2/SoC 24

**Describe alternatives you've considered**
Could not find an alternative

**Additional context**

andig, 2022-12-18:
Sounds good. I‘d also like to harmonize the device indices. Currently, REST starts at 0 while mqtt starts at 1. Does anyone have strong opinions which is preferred or idiomatic for both api types?

VolkerK62, 2022-12-18:
for me it would be logical to start with 1.

thomas-hutterer-tik, 2022-12-18:
Both is fine for me, harmonization between REST and MQTT is definitely a good thing

andig, 2022-12-20:
Follow-up: once done we can also start publishing individual vehicle data.

thomas-hutterer-tik, 2022-12-20:
Sounds good

Thomas

thomas-hutterer-tik, 2022-12-22:
Is there a nightly build? sudo apt-get install evcc told me nothing new since yesterday.

closed 2022-12-23 by andig

David-Toeltsch, 2022-12-23:
Hi we tested it with version 0.110.1 and it seems like the values are not available yet:

evcc/site/homePower 6.5302
evcc/updated 1671794130
evcc/loadpoints/1/chargePower 0
evcc/site/pvPower 0
evcc/site/batteryPower 10.022
evcc/site/gridPower -10.022
evcc/site/batterySoC 0
evcc/loadpoints/1/mode off

Best regards,
David (from Watt-Analytics)

andig, 2022-12-23:
Thats expected. They will be part of nightly release tomorrow. Due to the large amount of recent changes please don‘t expect a new release soon.

David-Toeltsch, 2022-12-23:
Ah okay we thought due to having a new version available that the changes might have been included already. Nvm then we´ll test it next week.
Thanks and merry christmas everyone :-)