evcc-io/evcc #5484: Limit power drawn from grid

Glennvd, 2022-12-17:
**Is your feature request related to a problem? Please describe.**
Belgium is introducing a capacity tariff in 2023. You pay net fees based on your highest kwh usage (15 min) of the past month. It would be great if evcc could help control this.

**Describe the solution you'd like**
In an ideal world, evcc would know the max draw you've had in the past month, and optimise for that. However, defining a "max kw from grid" in the config would already be great. EVCC would then actively lower the charging rate when the energy consumption of the house is rising. 

**Describe alternatives you've considered**
At the moment I have a simple pauze/resume automation for charging through home assistant and the tesla app.

**Additional context**
I'm happy to look into implementing a feature like this myself. I'm mainly wondering if a PR for something like this would be accepted or if it falls outside the scope of EVCC.

VolkerK62, 2022-12-17:
might be possible with the loadmanagement version of @hoermto 
see #1719

andig, 2022-12-18:
That will be load management.

closed 2022-12-18 by andig