evcc-io/evcc #5488: Live Netzpreis im UI statt ausschließlich hier: Auswertung Ladeenergie

Hofyyy, 2022-12-18:
**Describe the solution you'd like**
Es wäre schön den Strompreis statisch, oder ggf. live (Tibber) im Haupt-UI zusehen.
Ideen dazu sind im angehängten Screenshot angezeichnet.

Der Vorteil bei uns zu Hause wäre, das wir gleich sehen können, ob man ggf. mit Sofort laden möchte,
oder ob man z.b. die Waschmaschine einschaltet, da EVCC dauerhaft auf einem iPAD sichtbar ist.

**Describe alternatives you've considered**
An welcher Stelle man die Information am besten anzeigt und wie genau ist glaube ich nicth so wichtig. Möglich wäre bei den dynamischen Tarifen natürlich schone 6 Stunden Ansichten mit netter Grafik. Für uns wäre aber der aktuelle live Wert der Stunde (Tibber) voll ausreichend.

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

naltatis, 2022-12-18:
Bei statischen Strompreisen sehe ich den Mehrwert aktuell nicht so groß. Bei den meisten Nutzern ändert der sich ja nur einmal im Jahr - bei Einspeisung meist gar nicht. Bei dynamischen Preisen sehe ich den Mehrwert auf jeden Fall. Finde auch die Stelle am Netzbezug (vielleicht sogar im schwarzen Balken oben) für den Ist-Preis gut. Alternativ könnten wir so auch den aktuellen CO2 Wert https://github.com/evcc-io/evcc/pull/5454 darstellen.

Den Forecast (Werte für die nächsten Stunden) werden wir vermutlich erstmal im Zielladen Dialog einbauen um die Entscheidungen des neuen Planners https://github.com/evcc-io/evcc/pull/5445 besser verstehbar zu machen.

StevieC121176, 2022-12-18:
Könnte man in dem Zusammenhang nicht auch die Ladekosten für den geplanten bzw. Aktuellen Ladevorgang anzeigen?

Hofyyy, 2022-12-18:
Bei den statischen Strompreisen könnten man ja live mit der Solarenergie verrechnen. #Energiepreis, aus dem anderen Panel. Aber ja macht erst doppelt Sinn bei Tibber o.ä.  Ganz eigennützig würde mir auch der Momentanwert reichen :-) Der Verlauf ist denke ich im Plan UI besser aufgehoben und das momentane UI clean zu halten.

RTTTC, 2022-12-18:
Cool ideas!

To add constantly visible current price of either grid/solar or dynamic / solar calculation

Also would be cool to see each charge cost, this could be added to the sessions?

naltatis, 2022-12-31:
Ich überlege ob es sinnvoll ist die Preisdarstellung in der Main-UI konfigurierbar zu machen. Für Nutzer mit festen Tarifen ist das glaube ich nur so mittelspannend. Mit Zeittarifen oder dynamischen Preisen halte ich das für ne wichtige Info. Alternativ könnte man auch CO2-Infos zum Netzstrom anzeigen (sofern wir die dann später mal haben).

Quasi so ein Setting analog zu Kilometer/Meilen:

[image]

Das würde natürlich nur die Daten in der Energieanzeige und er Legende betreffen. Im Ersparnisdialog und bei den Ladevorgängen zeigen wir natürlich immer alle verfügbaren Daten.

premultiply, 2022-12-31:
Ist die Einheit so wichtig? Im Prinzip ist da ja immer ein relativer Indexwert je nachdem was man da als Datenquelle zugrunde legt.

StevieC121176, 2023-01-06:
Ich habe einen fixen Preis, könnte man in dem Zuge nicht auch anzeigen, wieviel man z.Bsp bei jedem Ladevorgang durch die Sonne spart? Ich lade gerade im Winter jetzt meistens mit min+PV und da gibt es meistens einen mix aus Netz und Sonnenenergie.

naltatis, 2023-02-04:
👆 Der PR schafft jetzt einige Vorbedingungen mit denen wir mehr Sessiondaten erfassen können (%Eigenstrom, co2, Preis).

PR #6048 (2023-02-06): Option to show current price and co2 info
Introducing an option to show current price and co2 data in the expanded energy flow. 

**Changes**
- 💵🌱 New UI option to add `price` or `co2` data to the energy flow. (fixes #5488)
  - The option is off by default since we consider it a more enthusiast feature
  - Options are only available of the datasources (`tariffs`) are configured.
  - Tooltip shows all available data (price, co2) including their full units. For brevity we've opted to show short units in the overview (`g` instead of `gCO2e/kWh` and `ct` instead of `ct/kWh`).
- 🔋 Added additional  [... cut]
closed 2023-02-06 by andig