evcc-io/evcc #5492: Planner: add schedule

andig, 2022-12-18:
Zielladen ist ein super Feature, jedoch wäre es auch für mich extrem hilfreich, wenn man das Zielladen nicht jedes mal neu aktivieren müsste. 
Bei mir würde jeden Morgen 90% (gerade im Winter wo eh nichts mit PV los ist) passen. Irgendwann vergesse ich das Zielladen zu aktivieren und stehe dann morgens fast blank da. Den MinSOC habe ich nur auf 30%. Wegen meinen dynamischen Strompreisen ist der Strom wenn ich komme einfach zu teuer und will das Sofortladen möglichst unterdrücken. So will ich immer in der Nacht laden. Ein Häkchen für immer um z.B. 5:00 Uhr Zielzeit würde genügen. Wenn dann noch die Funktion "lade zum günstigsten Preis kommt" bin ich fast wunschlos glücklich.

_Originally posted by @ThiloBaWue in https://github.com/evcc-io/evcc/discussions/1433#discussioncomment-4425881_

andig, 2022-12-18:
@naltatis I'm already working on zoned tariffs. Those need a day schedule like (Mo-Fr and Sa-So). On that basis we could also add a week schedule for target charging (like 80% Mo-Fr 7:00 instead of 80% on 16th 7:00). If you want to look at the UI part I'll work out the api.

magnifikus, 2023-01-10:
Eine Option für Wochentage wäre aber auch ideal, sowas wie MO-DO:17:00,FR:13:00,SA-SO:11:00, alternativ könnte man ja auch einfach 7 Werte für jeden Wochentag erlauben 17,17,17,17,13,11,11 z.B. alternativ nur 7 für immer den gleichen Wert.

schenlap, 2023-01-16:
Als Referenz, zoned tariffs: https://github.com/evcc-io/evcc/pull/5583

schedule könnte dann so aussehen:
```
tariffs:
  planner:
    - days: Mo-Fr
      targettime: 7:00
    - days: Sa
      targettime: 9:00
```
oder sollte es am loadpoint liegen?

andig, 2023-01-16:
Vielleicht sogar eher am Fahrzeug?

premultiply, 2023-01-16:
Oder sowohl als auch?

andig, 2023-03-08:
Aus https://github.com/evcc-io/evcc/discussions/6695#discussioncomment-5242762:

> Die zweite Anregung: täglich X Stunden oder X kWh ist eigentlich eine Spielart von https://github.com/evcc-io/evcc/issues/5492 die dann nicht auf den Soc sondern einfach auf die Menge schaut. Im Prinzip wiederholtest Zielladen ohne Soc wie bei den Offlinefahrzeugen.

maf-soft, 2023-03-27:
Falls noch nicht bedacht: ein einmaliger Plan (nur nächsten Samstag 9:00) wäre noch praktisch.

maf-soft, 2023-03-27:
@andig Aber nur wenn das Auto schon angesteckt ist und bis dahin auch bleibt - oder habe ich was übersehen?

andig, 2023-03-27:
Das ist ein guter Punkt:

	// reset plan once charge goal is met
	lp.setTargetTime(time.Time{})
	lp.setPlanActive(false)

Müssten wir überlegen ob wir das ändern wollen. Wenn Du aber eh noch 8x los fährst- warum dann heute schon den Plan für Samstag einstellen?

maf-soft, 2023-03-27:
Weil ich jetzt schon weiß, dass ich Samstag eine große Tour mache und rechtzeitig an alle wichtigen Voraussetzungen denken will. Eh noch 8x los fahre ich bis dahin für spontane Kleinigkeiten, die nicht geplant werden müssen. Am Ende möchte man das natürlich im UI einstellen, aber ich dachte das wäre jetzt hier die Gelegenheit, die Voraussetzungen zu schaffen.

kakulukia, 2023-04-13:
Interessant wäre auch eine Option zur Schonung des Akkus hinzuzufügen. Dann muss ich da selber gar nicht drauf achten und stecke einfach immer mein Auto an.

Hier mal eine Referenz: https://www.e-mobileo.de/10-tipps-wie-der-akku-laenger-haelt/

Ergo: Es wird per default immer nur bis 80% geladen und wenn ich dann nur mal kurz zum Supermarkt fahre, das Auto direkt wieder anstecke und er mit 75% erfasst wird, lässt evcc das Auto in Ruhe bzw. würde erst wieder bei einer konfigurierbaren Schwellen anfangen zu laden.

Marty56, 2023-04-16:
Vielleicht könnte man das Ziel minimum SOC nach jahreszeitabhängig machen.
Das habe ich mit meiner Homeautomation und dem Rest API so gemacht. 
In den Wintermonaten deutlich höher, als im Sommer. Monatsabhängig ist m.M. von der Granularität ausreichend.

diver77, 2023-04-21:
> Falls noch nicht bedacht: ein einmaliger Plan ("nur nächsten Samstag 9:00") wäre noch praktisch.
> @andig Aber nur wenn das Auto schon angesteckt ist und bis dahin auch bleibt - oder habe ich was übersehen?

ja das fehlt mir sehr oft und verstehe ich auch nicht.
Hatte es heute morgen schon wieder.
Gestern den Plan eingeschalten, auf 80% laden bis heute morgen 6.00Uhr

Gestern abend ist dann meine Schwiemu nochmal losgefahren und hat das Auto auch wieder angesteckt.
Leider war das Auto heute morgen bei 30% weil der Plan nicht mehr vorhanden war.

wisst ihr denn schon bis wann der Wochenplaner kommen soll?

Danke für Eure super Arbeit.

Gruß Diver

andig, 2023-04-21:
Das ist dann https://github.com/evcc-io/evcc/issues/5271

micw, 2023-04-27:
Heute morgen war das Auto leer, weil wir vergessen haben, die Zielzeit einzugeben. Zum Glück war das andere noch genügend geladen. Wäre es denkbar, erst einmal eine ganz einfache Standard-Zielzeit umzusetzen und all die konfigurierbaren Dinge wie Wochentage Stück für Stück nachzurüsten?
Beim Loadpoint kann ich ja schon mit "mode", enable und disable viel festlegen. Wenn jetzt noch "Zielzeit" als Config-Parameter hinzukäme, wären bestimmt 80% der Use-Cases erfüllt.

schenlap, 2023-04-27:
Als Workaround kannst du mein Script aus https://github.com/evcc-io/evcc/issues/5271#issuecomment-1517484370 verwenden.

diver77, 2023-04-28:
oh, das wäre wirklich cool, vor allem weil ich mich mit scripten nicht sehr gut auskenne.
Schmeisse mal in den Raum, das clever-pv das auch kann.

micw, 2023-04-28:
https://github.com/evcc-io/evcc/discussions/7710

hermann-59, 2023-07-10:
Auch für mich wäre die Wochenplanung das wichtigste nachzurüstende Feature. Dann könnte ich meinen Plugin-Hybrid auf PV-Laden stellen und über den Plan einstellen, dass er wochentags morgens um 6:00 voll sein soll. Das würde auch im Winter funktionieren und ich müsste nach einmaliger Einstellung (möglichst per Konfiguration, damit der Plan restarts übersteht) nie mehr etwas ändern. Wäre echt genial.

maetheagle, 2023-07-31:
Hi, ich bin hier ganz neu, und evcc liefert genau das was man sich als PV Besitzer mit Plugin Hybrid und E-Zweitwagen wünscht. Aber was wirklich fehlt ist ein Wochenplaner (einmal eingestellt, immer gültig) Daher bin ich auf den Thread gestoßen, und kann meinem Vorredner(schreiber) nur zustimmen. 
Alles was die Sonne tagsüber nicht schafft, wird wochentags von 2-6Uhr nachgeladen.
Das Batteriemanagement ist bei mir so eingestellt, das aus dem Hausakku nachts nur 300W rauskommen, um die Grundlast zu decken, so würde das Laden des Autoakkus auch nicht den Hausakku leersaugen.

@andig wird daran schon gearbeitet? oder wo steht der Planer auf der Prioliste?

naltatis, 2023-08-01:
Aktuell wird hieran noch nicht gearbeitet. Ersteinrichtung via UI ist die große Baustelle, die wir zuerst lösen wollen.

Wir sind uns aber einig, dass das ein sinnvolles Feature ist was wir über kurz oder lang auf jeden Fall haben wollen.

wattnpapa, 2023-08-04:
Moin
Als Anreiz ein paar Ideen wie es Tibber umgesetzt hat in der App

Das Auto ist bekannt und der SOC aktuell bei mir mit fiktiven SOC da keine Love Anbindung vorhanden.

Damit weiß Tibber wie viel KWh und wie schnell das Auto Max laden kann um auszurechnen wie viel Ladezeit es benötigt 

[image]

An dem jeweiligen Ladepunkt kann man diese Einstellungen machen 

[image]
[image]
[image]
[image]

Vielleicht hilft das als Anregung ;) man muss ja nicht immer alles neu erfinden 

Mit der Funktionalität bin ich aktuell zufrieden würde es auch gerne in evcc machen

hermann-59, 2023-08-31:
Angenommen, es gäbe die Möglichkeit einer Planung, dass das Auto immer um 6:00 morgens zu 80% voll sein soll.
Um ein optimales Ladeverhalten zu implementieren, reicht es wahrscheinlich nicht, den Zeitpunkt zu bestimmen, wann evcc spätestens beginnt zu Laden. Das wäre dann wahrscheinlich irgendwann in der Nacht, völlig ohne PV-Strom.

Um möglichst viel Strom aus der PV-Anlage zu nutzen, bräuchte man so etwa wie eine Vorhersage der Sonneneinstrahlung bis morgen um 6:00 verbunden mit der Vorhersage des PV-Ertrages. Dann müsste evcc an Hand des prognostizierten PV-Ertrages bestimmen, wann die beste Ladezeit ist, um den PV-Ertrag optimal zu nutzen, auch wenn dieser dann unter der Minimal-Ladeleistung liegen würde. Nur einen Teil der Ladeleistung aus der PV-Anlage zu beziehen wäre immer noch besser, als Nachts ohne PV-Strom zu laden.

ThiloBaWue, 2023-09-01:
Hallo hermann-59, das siehst Du vielleicht zu kompliziert. Wenn Du den Modus auf PV stellst und dann das Zielladen bis zu Deiner gewünschten Uhrzeit einstellst, dann hast Du das was Du willst. Er lädt mit PV, sobald ein Überschuss da ist und wenn das nicht reicht, lädt er rechtzeitig mit dem dann vermutlich aus dem Netz stammenden Strom nach. Was eben nur nach wie vor wünschenswert wäre, wenn man nicht jeden Tag die Zeit bis zur Abfahrt, von Hand neu stellen müsste. Ich hoffe immer noch stark auf das baldige Erscheinen der Funktion.

hermann-59, 2023-09-01:
Ich hatte es bisher so verstanden, dass evcc bei Einstellung "PV" erst zu Laden anfängt, wenn der Mindestladestrom (bei mir also 6A, macht also 1440W) als Überschuss vorhanden ist.
Habe ich an einem Wintertag nie 1440W Überschuss, würde evcc also erst nachts laden (obwohl tagsüber Überschuss vorhanden wäre, wenn auch nicht 1440W).
Sehe ich das falsch?

schenlap, 2023-09-01:
das ist so. Wenn nicht genug PV Überschuss vorhanden ist startet er nicht. Unter 6A ladet sein Auto vermutlich auch gar nicht. Du kannst aber dennoch einstellen dass er auch nur z.B bei 1kW Überschuß lädt, dann kommt halt ein Teil aus dem Netz. Ich habe ein Setting für Sommer und eines für Winter, das stelle ich dann einfach um.

hermann-59, 2023-09-01:
Es geht ja gar nicht darum, ob das Auto mit weniger als 6A lädt. Evcc sollte idealerweise dann laden, wenn voraussichtlich die höchste PV-Produktion ist, unabhängig davon, ob der gesamte Ladestrom aus der PV kommt.
Wie unterscheiden sich Deine beiden Settings?

schenlap, 2023-09-02:
es sind einfach nur unterschiedliche thresholds:
```
#summer
  enable: # pv mode enable behavior
    delay: 2m # threshold must be exceeded for this long, 2m
    threshold: -1500 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 2m # threshold must be exceeded for this long, 2m
    threshold: 300 # maximum import power 
  #spring
  #enable: # pv mode enable behavior
  #  delay: 1m # threshold must be exceeded for this long, 2m
  #  threshold: -1100 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  #disable: # pv mode disable behavior
  #  delay: 2m # threshold must be exceeded for this long, 2m
  #  threshold: 500 # maximum import power (W # MIN = 1380-enable(600) = 780)
  #winter
  #enable: # pv mode enable behavior
  #  delay: 1m # threshold must be exceeded for this long, 2m
  #  threshold: -600 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
  #disable: # pv mode disable behavior
```

naltatis, 2023-09-02:
@hermann-59 ja, es ist genau so wie du sagt. "Kleiner Überschuss" wird aktuell nicht mitgenommen. Allerdings gibts hier schon einen Proof of Concept, um aufgrund von PV-Vorhersagedaten bessere Entscheidungen treffen zu können:
https://github.com/evcc-io/evcc/pull/9454

naltatis, 2023-10-04:
Erster Entwurf

- Fokus auf Wochentag + Uhrzeit
- Aktivieren/Deaktivieren bestehender Pläne (analog iPhone/Android Wecker)
- Pläne können auf "wiederholen" gesetzt werden

[image]

[image]

[image]

ThiloBaWue, 2023-10-04:
Super, genau so habe ich mir das gewünscht!!! Bitte in dieser Weise umsetzen!!!

diver77, 2023-10-05:
Genau so wäre es perfekt, zumindest für mich.
Ich weiß Geschmäcker sind verschieden aber für mich passt es. :-)

maetheagle, 2023-10-07:
Perfekt, danke das ihr das angegangen seid!

ThiloBaWue, 2023-11-01:
Gibt es schon etwas neues? Kann man das schon in einer Beta testen?

stevoboy, 2023-11-12:
Würde mich auch brennend interessieren!

alex90alex, 2023-11-19:
Ich hänge mich mal ran und freue mich auch schon auf die wiederkehrende Ladeplanung. 
Generelle Frage: wie verhält sich denn die Ladeplanung wenn man zwei Fahrzeuge an zwei Ladepunkten angeschlossen bzw. wie sollte es sich verhalten? Ich hatte heute Nacht die Ladeplanung für zwei Fahrzeuge aktiv. Als ich die Ladeplanung für ein Fahrzeug aktiviert hatte ermittelte EVCC eine Ladezeit von ca. 3h und 5 min. Nachdem ich die Planung für das zweite Fahrzeug eingestellt hatte, wurde die Ladezeit vom ersten Fahrzeug nicht erhöht. 
Hintergrund: die beiden Wallboxen teilen sich einen Anschluss mit insgesamt max. 11kW. Gibt es eine Möglichkeit dies bei der Ladezeitberechnung mit einfließen zu lassen?
Beste Grüße Alex

naltatis, 2023-11-20:
evcc geht bei der Berechnung von der maximalen Ladegeschwindigkeit (+ etwas Puffer) aus. Lastmanagement o.ä. ist aktuell nicht abgebildet.

Update zum Thema: In https://github.com/evcc-io/evcc/pull/10335 haben wir ne ganze Menge im Unterbau geändert (Trennung Ladeziel und Ladelimit, ...). Wenn das durch ist, ist auch der Weg zu wiederholenden Plänen nicht mehr sehr weit. Zeitschätzung können wir aber natürlich nicht abgeben.

michiproep, 2023-12-25:
Super Feature!
Es sollte über die Config ein Standard gesetzt werden können (bsp: charger -> min 30kwh bis 06:00 Uhr oder vehicle -> min 150km bis 06:00Uhr)
Allerdings sollte es über die UI möglich sein, den Standard zu überschreiben (bsp: morgen -> voll laden)

Aber ich denke, ihr macht das schon so dass es passt.

uerlenwein, 2023-12-28:
Ich hätte gerne auch noch die Option mit einer Startzeit für das Laden, also um wieviel Uhr wird gestartet mit dem Laden.

docolli, 2024-01-03:
https://github.com/evcc-io/evcc/issues/5492#issuecomment-1746926443 sieht schon sehr vielversprechend aus!

Hier mal meine Gedanken dazu, 

Grundlegend sind die Ladeparameter `immer in Plänen` hinterlegt.
Jedes Vehicle (auch das Gastfahrzeug) hat seine eigenen Pläne, sie sind also Eigenschaften des Fahrzeugs.

Die Pläne bekommen Prioritäten/Hierarchieebenen zugeordnet.
Werte, die in einer höheren Ebene nicht definiert sind, werden von der darunterliegenden Ebene vererbt.

Grundsätzlich bräuchte man:

1) Unterste Ebene / Default-Plan (Basiswerte, die immer gelten)
2) Standard Ebene / Normale Ladepläne
3) Oberste Ebene / Ad-Hoc Ladeplan

Jeder Plan kann folgende Werte enthalten:

A) Zeitpunkt bzw. gültiger Zeitraum (für wiederkehrende Pläne) wie "jeden Tag / bis 8 Uhr", "Montag / bis 7 Uhr", "Montag bis Freitag / bis 6 Uhr", oder auch "November" oder "November bis März"
B) Ladelimit, bis zu dem geladen wird
C) Mindestladung, bis zu der immer mit max. Ladeleistung geladen wird

A) und B) sind ja heute schon im Plan. C) würde ich der Vollständigkeit halber mit aufnehmen, um mehr Flexibilität zu haben. So könnte man sich dann Jahreszeitenpläne mit unterschiedlichen Werten erstellen und muss nicht immer nachjustieren.

Ich würde dann ein Planer GUI machen, in dem es einen (1) nicht löschbaren Default-Plan gibt, der einen nicht-editierbaren Zeitraum "Immer" hat und bei dem man die Werte B) und C) eingeben kann.
Dann gibt es die "Normalen Ladepläne", so wie heute auch mit den Werten A) und B), sowie C). A) muss in diesem Plantyp definiert sein, ist B) bzw. C) nicht definiert, wird der Wert vom Default-Plan übernommen.
Zusätzlich gibt es einen (1) "Ad-Hoc" Ladeplan, der ein temporäres Übersteuern der normalen Pläne erlaubt. Wenn dieser erreicht ist (oder die Ladung anderweitig beendet wird), wird er wieder gelöscht. Dieser wird direkt über das GUI / Slider auf der Hauptansicht definiert.

Die Plan-Ebenen könnte man im GUI über Reiter trennen ("Default" / "Abfahrt" / "Ad-Hoc"). So wie heute "Abfahrt" / "Ankunft".

Es gilt auch die Nicht-Soc Fahrzeuge zu berücksichtigen. Heute gibt es meines Wissens für diese nur den Ladeplan mit einer einstellbaren Energiemenge in kWh. Und eine Ad-Hoc Energiemenge, die man über den Slider nach dem Einstecken und Start der Ladung einstellen kann.
Würden für diese Fahrzeuge Default-Energiemengen Sinn machen, die immer verwendet werden, wenn man das Fahrzeug einsteckt?
Also so eine Art wiederkehrender Plan mit definierter Energiemenge (anstatt einem Soc-Ladelimit) "jeden Tag / bis 6 Uhr / 12 kWh". Als Pendler weiß man dann, dass man jede Nacht z.B. 12kWh nachladen muss, um den Tagesverbrauch ungefähr wieder auszugleichen. So steckt man jeden Abend nur an und evcc kümmert sich um des Rest. Am nächsten Morgen (zum geplanten Zeitpunkt) sind die 12kWh optimal (günstigster Preis / wenig CO2) geladen.

bolisaf, 2024-02-03:
@naltatis my German isn’t that good. It looks like this is the thread where you are discussing implementing a recurring plan per day of the week etc. Would you mind providing an update? Looking forward to this great new feature :)

hermann-59, 2024-02-03:
Ich habe noch nicht verstanden, wie ein Plan mit dem Lademodus (PV, min+PV, max) zusammenspielen soll. Wäre es nicht vernünftig, den Lademodus mit in den Plan aufzunehmen?
Dann könnte ich für meinen PHEV einen Plan für Sommer machen (PV), einen für die Übergangszeit(min+PV) und einen für den Winter (max?). So muss ich immer von Hand umstellen.

JoernHei, 2024-02-03:
> Ich habe noch nicht verstanden, wie ein Plan mit dem Lademodus (PV, min+PV, max) zusammenspielen soll. Wäre es nicht vernünftig, den Lademodus mit in den Plan aufzunehmen? Dann könnte ich für meinen [...]

Der Plan spielt immer eine Rolle, nach meinem Verständnis:
Bei "schnell": Es wird geladen bis zum Ladelimit.

Bei "min + PV": Es wird mit der Mindestladeleistung bis zum Ladelimit geladen. Sollte das Ziel aus dem Plan (in der Mitte unten) so nicht erreicht werden können, wird der Ladestrom so spät wie möglich auf das Maximum gestellt um die im Plan eingestellte Lademenge zu dem im Plan eingestellten Zeitpunkt zu erreichen.

Bei PV: Es wird nur mit PV Überschuss geladen. Sollte so der Plan nicht erreicht werden können, wird spät wie möglich auf das Maximum gestellt um die im Plan eingestellte Lademenge zu dem im Plan eingestellten Zeitpunkt zu erreichen.

Also stellst du im Plan einfach ein wann du losfahren möchtest und lässt den Modus auf PV. Dann wird so viel PV wie möglich genutzt und dein Auto ist zum Abfahrzeitpunkt geladen.

Beste Grüße

bolisaf, 2024-02-03:
Would it make sense to add above explanation to the help?Thanks!L

hermann-59, 2024-02-03:
Damit sind aber Plan und Lademodus entkoppelt und müssen separat eingestellt werden. Ein Sommer- und ein Winterplan sind damit nicht möglich.

Apropos Plan: wie sieht es dann mit variablen Stromtarifen und PV-Ertragsprognosen aus?

JoernHei, 2024-02-03:
Der Modus steht über dem Plan. Der Plan hat nur Einfluss auf den PV Teil.

Kannst du bitte noch einmal genau deinen Anwendungsfall schildern, damit ich dein Problem besser verstehen kann? Am besten mit Werten, wie PV Überschuss, Abfahrtszeit, Lademenge. Vielleicht kommen wir so zu einer bequemeren Awendungsmöglichkeit bei dir.

hermann-59, 2024-02-03:
Wie bereits erwähnt fahren wir einen Plugin-Hybrid mit relativ kleinem Akku (13kWh). Normalerweise fährt meine Frau ca. 20 km täglich (morgens). Im Sommer kann dann das Auto zu 100% mit PV-Überschuss geladen werden. Es reicht, wenn das Akku am nächsten morgen geladen ist.
Im Winter oder in der Übergangszeit liefert die PV-Anlage aber nicht immer genügend Strom und ich muss von Fall zu Fall auf min+PV umschalten, damit das Auto am nächten morgen voll ist (und es muss voll sein, da die elektrische Reichweite des PHEV gerade mal 45km beträgt).
Für einen ersten Schritt wäre es gut, wenn das Auto (mit möglichst viel PV-Überschuss) bis zum nächstem morgen vollständig geladen wäre. Des weiteren könnte ich mir vorstellen, dass eine PV-Prognose verwendet wird, um die optimale Ladezeit und den optimalen Ladestrom zu ermitteln.
Ziel sollte es sein, dass der Akku mit möglichst geringen Kosten so geladen wird, dass wir meist rein elektrisch fahren können.

JoernHei, 2024-02-03:
Wie groß ist die PV? Gibt es eine Hausbatterie? Bisher hört es sich so an wie bei mir, dann könnte ich dir meine Konfiguration erläutern.

JoernHei, 2024-02-03:
Ladeleistung ist bestimmt 3,7 kW? Ist zumindest bei meinem Passat so.

hermann-59, 2024-02-03:
Ich habe keinen Hausspeicher (lohnt sich bei uns nicht, wir sind beide tagsüber zuhause) und unsere PV hat 9,2kWp. Und ja, die max. Ladeleistung ist 3,7kW. Die minimale ist übrigens 1,4kW.

JoernHei, 2024-02-03:
Ok, ich habe die gleiche Konfiguration. Leider bin ich nicht an allen Tagen so früh zu Hause.

Ich habe bei mir beobachtet, dass das Auto mit 3,7 kW am effizientesten lädt. Aber wenn PV Überschuss ist, dann möchte ich den ja nutzen, egal wie effizient.

Daher habe ich in der Konfigurationsdatei (yaml, genaueres dazu steht in der Doku, die man über das Menü oben rechts unter der Hilfe findet) folgendes eingestellt: Start bei 500W Überschuss, Stopp bei 1000W Bezug für 10 Minuten. 
Das hatte ich mal für den Winter so eingestellt, läuft aber auch im sonst mit erstaunlich wenig Netzbezug.

Im Plan gibt man dann nur noch die nächste Abfahrtszeit (oder 15 Minuten früher zur Sicherheit) ein. Lademenge immer voll.

Somit wird jeder PV Überschuss genutzt und falls dieser nicht reicht, wird das Auto mit Netzstrom erst kurz vor Abfahrt geladen. Ist auch besser für den Akku, wenn er nicht so lange vollgeladen steht.

Das Einstellen der Abfahrtszeit muss man leider immernoch jeden Tag vornehmen. So wie ich es verstanden habe sind die Entwickler aber daran einen "Wochenplan" zu entwickeln, der sich automatisch wiederholt. 

Ich hoffe ich konnte dir weiterhelfen.

hermann-59, 2024-02-03:
Start bei 500W Überschuss würde bei mir bedeuten, dass 900W aus dem Netz kommen (da mit mindestens 6A, also knapp 1,4kW geladen wird). Finde ich nicht so optimal. Auch bedeutet es, dass die ersten 500W eingespeist werden und damit für das Auto verloren sind (geht wahrscheinlich nicht besser). Stopp bei 1000W Bezug heißt dann auch noch, dass viel mit gekauftem Strom geladen wird. Geht das nicht besser?

Ich habe im Sommer "PV" eingestellt, da dann genügend Überschuss da ist, um das Auto komplett (mit max. 3,7kW) zu laden. In Winter stelle ich auf "min+PV", was bei mir bedeutet, dass normalerweise mit 1,4kW geladen wird, wobei dabei der größte Anteil vom Stromanbieter kommt. Da das Auto üblicherweise so zwischen 10 und 11:00 eingesteckt wird, funktioniert das auch ohne Plan ganz gut.

JoernHei, 2024-02-03:
Ja, man muss eine Entscheidung treffen wann man anfängt zu laden. Ich war überrascht wie gut diese Kompromisslösung läuft. Habe es deswegen so gelassen. An einem sonnigen Tag nimmt die PV Leistung stetig zu und der Netzbezug ist trotzdem sehr gering. (Grafisch ist es ja die Fläche zwischen der Ladeleistung und des PV Überschusses. Wenn da für 60 Minuten im Schnitt 500W bezogen werden sind das 0.5 kWh -> hier könnte man ca. 10 - 15 Cent sparen, aber nur an einem sonnigen Tag, an bewölkten Tagen muss der Rest sowieso aus dem Netz kommen, -> im Jahr ca. 180 Tage x 0,15€ = 27€ pro Jahr macht im Monat bisschen was über 2€)Die 10 Minuten kommen daher, dass die Wallbox nicht bei jeder Wolke gleich abschalten soll.Wenn du die (gefühlt) beste Planung haben möchtest, musst du davor sitzen und es selbst steuern. Das habe ich auch eine Weile gemacht. Bin zu dem Schluss gekommen, dass es besser ist den Computer soetwas machen zu lassen.Ich kann nur empfehlen mit den Einstellungen herumzuspielen, dann findest du eine akzeptable Lösung. Ich hatte auch eine Weile eine prognosebasierte Ladung, die lief aber auch nicht viel besser. Vielleicht reicht es dir auch einfach min+PV als Standard zu setzen. Und wenn du viel Sonne erwartest auf PV umzustellen.Wir Hybridfahrer mit dem kleinen Akku haben es auch etwas schwerer als die anderen, glaube ich zumindest.

hermann-59, 2024-02-03:
Die optimalen Einstellungen sind wahrscheinlich bei jedem ein bisschen anders. Wenn ich im Sommer mit "PV" lade, ist der Akku eigentlich immer bei Sonnenuntergang voll, ohne Strom dazuzukaufen.
In der restlichen Zeit lade ich mit "min+PV". Bleibt mehr als die 1,4kW übrig, nutzt das evcc, ansonsten wird mit 1,4kW geladen. Auch da ist bei Sonnenuntergang der Akku voll, aber ein Teil kommt eben vom Stromversorger.
Schön wäre es, wenn evcc beim Betrieb mit "min+PV" eine Prognose hätte, mit welcher errechnet werden könnte, wann heute mit dem höchsten Ertrag zu rechnen ist und die Ladezeit/-leistung dementsprechend anpassen könnte.

Ach ja: Das Umschalten des Lademodus per UI sollte natürlich persistent sein, der aktuell eingestellte Lademodus also auch bei einem Neustart von evcc nicht verloren gehen.

JoernHei, 2024-02-03:
Bei min+PV sollte die Leistung bei genügend Überschuss angehoben werden.

Den Standardmodus beim Anstecken des Fahrzeugs kannst du selbst festlegen. Siehe hier https://docs.evcc.io/docs/reference/configuration/loadpoints unter dem Punkt "mode".

Vielleicht könnte uns einer der Entwickler (@andig ) sagen, ob es die Möglichkeit "Persistent" künftig auch noch geben wird. :) Oder wie man das sonst erreicht?

schenlap, 2024-02-04:
Ich poste hier nochmals mein Skript dass beim Anstecken eines Fahrzeugs einen Plan erstellt.

Änderung an der evcc.yaml:
```
# push messages
messaging:
  events:
    connect: # vehicle connect event
      title: Car connected
      msg: "${vehicleTitle} Car connected, mode:${mode}"
    guest:
      title: Guest connected
      msg: "${vehicleTitle} Guest connected, mode:${mode}"
  services:
    - type: script
      cmdline: /home/pi/evcc/evcc_message # ev. anpassen
      timeout: 50s
```

Das script liegt am server, wo auch evcc läuft, unter /home/pi/evcc/evcc_message und muss ausführbar sein:
```
ip=localhost:7070
log="/tmp/evcc_targettime.log"
now=$(date)
car=Audi  # Fahrzeugname laut evcc.yaml

echo "message received: $1 - $2 - $now" >> $log
if [[ "$2" == *"connected, mode:pv"* ]] # guest and car
then
        soc=50
        week_day=$(date +%w) #get the week day as a number from 0 to 6 starting with 0 as Sunday
        if [[ $week_day -ge 0 && $week_day -le 4 ]] # from monday to thursday
        then
                targettime=$(date -d "+1 day" -u +"%Y-%m-%d")T05:00:00Z
                soc=95
        else
                targettime=$(date -d "+1 day" -u +"%Y-%m-%d")T07:00:00Z
                soc=60
        fi
        curl -X POST http://localhost:7070/api/vehicles/$car/plan/soc/$soc/$targettime >> $log
        curl -X POST http://localhost:7070/api/vehicles/$car/limitsoc/100 >> $log
        echo "Setting targettime: " $targettime to $soc >> $log
else
        echo "Wrong message: " $2 >> $log
fi
```

Ist der nächste Tag ein Wochentag, dann wird per plan auf 95% geladen. Ist der nächste Tag ein Wochenende, dann auf  60% (Reserve für pv Ladung). Der Fahrzeugname muss mit der yaml übereinstimmen. Mit dem Script brauchte ich nur selten eingreifen.

DK4FT, 2024-02-04:
Habe die gleichen Bedingungen wie herman-59: Golf GTE (Hybrid) mit kleinem Akku, keinen Haus-Akku und tägliche Fahrten unter 40km, für die 80% Ladestand ausreichen, wenn er auch geladen ist und ich es nicht mal wieder vergessen habe zu aktivieren...

@schenlap top, danke fürs Teilen.
Klingt nach einer guten Lösung, bis persistente Pläne es in die Applikation schaffen.

fuxae, 2024-02-08:
> evcc geht bei der Berechnung von der maximalen Ladegeschwindigkeit (+ etwas Puffer) aus. Lastmanagement o.ä. ist aktuell nicht abgebildet.
>
> Update zum Thema: In [...]

@naltatis sorry wegen der Nachfrage, aber ist der wiederkehrende Plan etwas an dem ihr gerade arbeitet oder ist das momentan zurückgestellt aufgrund anderer wichtigere Themen? 
Sollte das Thema momentan keine Prio haben, werde ich mit mit Home Assistant den Planer zwischenzeitlich selber bauen. Wenn es aber ohnehin in einem der nächsten Releases angedacht ist, würde ich mir das sparen...
Danke :)

JoernHei, 2024-02-13:
Ich verstehe das als Laden mit min Ladeleistung bis der Plan erfüllt ist und bei PV Überschuss die Ladeleistung erhöhen und/oder den Plan überschreiten. Korrekt?

docolli, 2024-02-13:
Gelduld, wird kommen und sieht vom Ansatz her sehr gut aus!
Siehe Screenshots https://github.com/evcc-io/evcc/issues/5492#issuecomment-1746926443

[11 more replies]

PR #16091 (2024-12-01): Add schedule for planner
Fix #5492

Hi,
dieser Draft soll den Support für das Planen für Wochentage hinzufügen.
Dafür fehlt noch folgendes: *Ich kenne mich im `evcc`-System nicht so gut aus. Wenn etwas fehlt, dann bitte mir mitteilen.*

- [x] **Backend-Implementierung**
- [x] **Tests (um-)schreiben**
- [x] **Bug fixen:** `Uncaught (in promise) Maximum recursive updates exceeded in component <ChargingPlanRepetitiveSettingsEntries>. This means you have a reactive effect that is mutating its own dependencies and thus recursively triggering itself. Possible sources include component template, rend [... cut]
closed 2024-12-01 by andig