evcc-io/evcc #5506: Adding Meatrol ME631 Meter

stewartwatson321, 2022-12-19:
Hi

I am using Meatrol ME631 Meter with EVCC; I find it useful because it is accurate and allows the use of current clamps (so I can simply clamp onto existing wiring, rather than having to cut/connect wires). Most of the other meters you have as 'compatible'  with EVCC require direct connection to wires, which is much harder to achieve with an existing installation.

I have written the following template to get ME631 working [one phase is used as grid meter, second phase is used as PV meter, third phase (address 2159) is currently unused, but could be, for example, battery - which I don't have]:

This is the me631.yaml file I created:

template: me631
products:
  - brand: me631
    description:
      generic: me631
params:
  - name: usage
    choice: ["grid", "pv"]
    allinone: true
  - name: host
  - name: port
    default: 502
render: |
  type: custom
  power:
    source: modbus
    uri: {{ .host }}:{{ .port }}
  {{- if eq .usage "grid" }}
    id: 1
    register:
      address: 2155 # grid power
      type: holding
      decode: float32
    scale: 1000
  {{- end }}
  {{- if eq .usage "pv" }}
    id: 1
    register:
      address: 2157 # PV-sensor power
      type: holding
      decode: float32
    scale: 1000
  {{- end }}

I can obviously add this to my compile, but what is the procedure for getting this meter adopted into 'mainline' EVCC?

Thanks.

andig, 2022-12-19:
Please format posting

premultiply, 2022-12-19:
Ein Fall für mbmd @andig ?
http://www.rogowski.cn/Private/Files/20190709/6369826453235993354459719.pdf

andig, 2022-12-20:
Closing here- lets move this to mbmd!

closed 2022-12-20 by andig