evcc-io/evcc #5516: Änderung der enable und disable threshold per api

StefanSchoof, 2022-12-20:
**Is your feature request related to a problem? Please describe.**
#5501

**Describe the solution you'd like**
Eine REST api method um enable und disable threshold während der Laufzeit zu ändern.

**Describe alternatives you've considered**
Weiter jedes mal das Auto abstecken und evcc neu starten.

**Additional context**
- Wenn es hilft, wäre bereit zu versuchen einen PR zu erstellen.
- Aktuell ist der Wert nicht in dem state Endpunkt
- `/api/loadpoints/<id>/enablethreshold` vs `/api/loadpoints/<id>/enable/threshold`?
- Sollte `delay` sich auch anpassen lassen?

andig, 2022-12-20:
Schöner wäre eine bessere Automatik für den Winter? Perspektivisch wollen wir den MinPV Modus ja loswerden und durch etwas clevereres ersetzen.

closed 2023-01-15 by github-actions[bot]

andig, 2023-01-22:
Wir haben uns entschlossen, die Apioberfläche für diese Detailparameter nicht zu vergrößern. Unser Fokus liegt weiterhin darauf, gute Automatiken zu bauen und uns nicht in Details zu verzetteln. Danke für den PR!