evcc-io/evcc #5518: SOC calculated incorrectly with multiple batteries

kscholty, 2022-12-20:
### Describe the bug

If multiple batteries with different capacities are configured in the system, it's impossible to calculate the real overall SOC. 
Currently it's calculated as the average SOC over all batteries, which is only correct if all batteries have the same capacity,
See calculation of variable `socs` in function `sitePower()` in `core/site.go`

Example: 
Battery 1, capacity 10kWh, SOC 10%
Battery 2, capacity 30kWh, SOC 20%
Resulting SOC on UI: 15%, Real SOC: (1+6)/(10+20) = 23,3%

Possible solution:
- Add parameter "Capacity" to Battery meters and extend battery API accordingly
- Calculate correct SOC using the actual capacities

### Steps to reproduce

1. Configure a system with 2 batteries and different SOC
2. Check SOC on the UI. It will show the average SOC.
3.

### Configuration details

```yaml
meters:
- type: template
  template: e3dc
  usage: battery
  host: s10
  port: 502
  name: battery1
- type: template
  template: victron-energy
  usage: battery
  host: multiplus
  port: 502
  name: battery2
```

### Log details

```text
n/a
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.109.2

premultiply, 2022-12-20:
This is by design to keep it simple.

kscholty, 2022-12-20:
I see. It's simpler indeed. 
However, showing a wrong SOC is not nice and using this wrong value with parameters like bufferSoC and prioritySoC makes these kind of useless, or at least sketchy in such a scenario.

premultiply, 2022-12-20:
How do you control that your batteries charge and discarge with power exactly linear to their capacity ?
The whole SoC thing is a huge guess and because of this and because the cases where multiple batteries at all or even multiple batteries with different capacities used are more than rare and there is only one SoC display we decided to keep it as simple as possible.

Maybe https://github.com/evcc-io/evcc/pull/5497 will help you?

Even if we would provide an capacity input option for the users users would tend to enter wrong (to nice) values from marketing etc. It's the same with the vehicle batteries where we have this config option to set a startup value (just) for estimator initialization.

kscholty, 2022-12-20:
> How do you control that your batteries charge and discarge with power exactly linear to their capacity ? 

The battery controller (a BMS or ESS system) gives me a SoC. This SoC is per definition related to the remaining capacity, since it's simply `RemainingCapacity/TotalCapacity * 100` 
A good battery controller is able to count the energy going into and out of the battery, so the SoC is quite accurate.
(But I agree that in many cases it is not)

>The whole SoC thing is a huge guess

I'd say that depends. Using values from a cheap BMS is probably worse than guessing, but e.g. a Victron SmartShunt is quite accurate.

> and because of this and because the cases where multiple batteries at all or even multiple batteries with different capacities used are more than rare

Ok, that's probably true

> and there is only one SoC display we decided to keep it as simple as possible.

Here I strongly disagree. If multiple batteries are supported, the functionaly should work correctly, at least in theory.
The input possibly being wrong, is no excuse for adding another layer of "wrongness".
Especially if the result of the calcualtion is itself an input for another functionality.

> Maybe #5497 will help you?

Thanks for the hint. My problem is not so much the display, it's the `bufferSoC `and `prioritySoC ` not working correctly due to the wrong calculation. I know the individual data quite well and don't use evcc as a data source for the batteries.

> Even if we would provide an capacity input option for the users users would tend to enter wrong (to nice) values from marketing etc. It's the same with the vehicle batteries where we have this config option to set a startup value (just) for estimator initialization.

It might happen, agreed. But then it's the user's responsibility. However, the current implementation does not even allow me, as an informed user, to achieve the correct result, which is my whole point.
And I guess you shouldn't underestimate your users. Probably most of them are enthusiasts in the one or other way who know quite a lot about their systems. Regarding car batteries, it's indeed often unclear what the capacity really is, but for PV-Batteries this can be different.

However, to shorten the discussion (which I appreciate, by the way)

**Would a fix to this be accepted or shouldn't I bother, because the design decision would lead to a rejection of a respective pull request anyway?**

Hofyyy, 2022-12-20:
I like the idea to have an optional parameter capacity like we have it in the cars. at the end it is the same without wheels :-)

andig, 2022-12-20:
Agreed this can be improved, but not a priority. Happy to look at a PR.

Hofyyy, 2022-12-20:
> Agreed this can be improved, but not a priority. Happy to look at a PR.

Hey Andi,

magst du mir mal 1 bis 2 einstiegspunkte nennen? Dann gucke ich mir das mal an.
Ich habe eine Victron Battery am laufen und würde mich durch den "schönen" go sourcecode hangeln.
In der Theorie müsste das ja sehr einfach gehen, wenn man nicht gerade C++ verliebt ist.

andig, 2022-12-21:
We would need something like:
- `api.Battery`, maybe `api.BatteryCapacity`
- this needs be (optional) part of all meters, maybe similar to `vehicle.embed` (`embed.go`)
- then `site.go` would look if `api.BatteryCapacity` is available and- if yes- use those values for better calculation results

Just an idea, depends more insights when prototyping.

closed 2022-12-29 by andig