evcc-io/evcc #5520: DaheimLaden Wallbox ohne Cloud

VolkerK62, 2022-12-20:
Die DaheimLaden Wallbox kann jetzt auch im lokalen Netzwerk betrieben werden.
Diese Info habe ich von DaheimLaden erhalten:
```
die Wallbox kann seit Anfang Dezember 22 lokal mittels Modbus TCP gesteuert werden. 
Die Schnittstelle ist aktuell im Public Beta Test. 
Die ModBus TCP und OCPP Schnittstelle können parallel genutzt werden.
 
Mit der Modbus TCP Schnittstelle kann jedoch auch ohne DaheimLaden Cloud-Server Verbindung
 mit der Wallbox im eigenen Netzwerk kommuniziert werden.
 
Anbei finden Sie hierfür die detaillierte Dokumentation.
```
[02.12.2022 DaheimLaden ModBus TCP Spezifikation.pdf](https://github.com/evcc-io/evcc/files/10270481/02.12.2022.DaheimLaden.ModBus.TCP.Spezifikation.pdf)

Lässt die sich als lokale Version implementieren?

andig, 2022-12-20:
@utsavanand2 is this something you'd like to address?

premultiply, 2022-12-21:
Könnte ich wohl bauen. @VolkerK62 würdest du das mit mir probieren & testen?

VolkerK62, 2022-12-21:
gerne, aber ich kann nur nightly 
Selber bauen und kompilieren ist mir völlig unbekannt

kaluxx, 2022-12-29:
... würde mich auch zu Test anbieten. 
Ich nutze allerdings die docker software. Könnte mir vorstellen, dass die Modbus steuerung hier nochmal schwieriger ist.

sma-hm -  evcc - daheimladen v1 - hyundai I5

StevieC121176, 2023-01-01:
ich könnte mich auch zum Testen anbieten.

Daheimladen Touch, Kostal WR + Kostal Smartmeter

@VolkerK62 - Welche Software Version hast du auf deiner Wallbox hast?

VolkerK62, 2023-01-01:
@StevieC121176   k.A. sieht man das irgendwo? ich habe die erste Generation (Smart)

StevieC121176, 2023-01-01:
> @StevieC121176 k.A. sieht man das irgendwo? ich habe die erste Generation (Smart)

Ah ok, ich hab die Touch. bei mir sieht man die unten rechts auf dem Display. Bei der Smart wird das mit Sicherheit anders sein und auch die Firmware Version unterschiedlich.

VolkerK62, 2023-01-04:
@StevieC121176  (ich mach mal hier weiter) 
funktioniert mit der neuen Firmware evcc über cloud noch?
Ich habe vorhin mit Jonas von DaheimLaden telefoniert und er war sich nicht sicher, ob die neue Firmware mit Modbus auch schon mit evcc nutzbar ist.

Bin mir nicht sicher, ob ich die Dokumentation richtig verstehe, aber kann man die Box dann in 0,1 A Schritten regeln?

StevieC121176, 2023-01-05:
Also mit der Firmware für die Touch AC3_1.08 geht Modus und EVCC ohne Probleme, das ist keine Beta. Diese ist allerdings für die Touch. Vorher hatte ich die AC3_1R2DA. Damit ging EVCC aber Modus nicht.

Ich hab mir erstmal die ganzen Register im Modbus Adapter angelegt. Diese werden auch alle korrekt angezeigt. 
Die Doku ist etwas verwirrend, laut meinem Verständnis ist das der Faktor der dort beschrieben ist. Heißt Ausgabe ist 2400, der tatsächliche Wert ist dann 240 (2400*0,1)

PV-Heini, 2023-01-05:
Ich habe auf der Smart-WB aktuell die Firmware AC_SCE_3.r1DN (steht zumindest im Display) drauf und habe vom DaheimLaden-Kundenservice folgende Auskunft erhalten:

_Sie haben aktuell noch eine Nachladesoftware auf der Wallbox installiert. Die gibt es derzeit noch nicht mit Modbus Spezifikation. Ich kann Ihnen die Software mit Modbus aufspielen, dann aber ohne Nachladefunktion. Oder Sie gedulden sich nochmal ca eine Woche, dann haben wir auch die Nachladesoftware mit Modbus Schnittstelle fertig._

Ich warte dann gerne noch, bis die Nachladesoftware fertig ist ...

StevieC121176, 2023-01-05:
Das ist nur ein Häkchen was in den Einstellungen einmal manuell gesetzt werden muss...

VolkerK62, 2023-01-07:
@premultiply  wenn die neue Firmware da ist, könnte man sowohl die Smart, als auch die Touch Version testen.
Hast du dir das schon mal genauer ansehen können?

premultiply, 2023-01-07:
Noch nicht, steht aber auch meiner ToDo-Liste ziemlich weit oben.

PV-Heini, 2023-01-14:
Die Versions-Nr. im Display meiner Smart WB hat sich übrigens noch nicht verändert. Wie sieht es denn bei den anderen Benutzern der Smart-WB so aus?

VolkerK62, 2023-01-15:
Ich habe die Nachlade-Modbus-Version bei mir drauf, aber ich habe auch die " ....DA" Version. Deine "....DN" Version ist wohl noch in der Mache. Das ist ein kleiner Laden, gib ihnen etwas Zeit. Die Modbus-Version ist ja auch in evcc noch nicht implementiert.

stephanschwebe, 2023-01-15:
Wir haben wir für alle User inzwischen eine Modbus Firmware-Software fertig. Um ein bissl Dunkel ins Licht zu bringen, die DA Software sind DaheimLader Smart Wallboxen mit integrierten FI, die DN Wallboxen (DaheimLader Smart) haben einen externen FI. Volker, du hast gestern den Fehler in deiner Nachladesoftware (3.R3DA) gefunden. In der Software 3.R2DA ist Modbus ebenso enthalten. Für die 3.R2DN User können wir es auch lauffähig bekommen. 

Auf dem DaheimLader Touch benötigt ihr mind. Firmware 1.08DA oder 1.R1DA. 

Daher würden wir alle, die die Beta EVCC Modbus Implementierung von Stevie testen wollen, dabei unterstützen, ihre Wallbox richtig zu konfigurieren. Bitte schickt uns per E-Mail an info@daheimladen.de eine kurze E-Mail (Station-ID + Beta Modbus EVCC Test). Dann installieren wir morgen bei euch allen die richtigen Firmwares. 

Mit der Nutzung des Modbus Protokolls kann EVCC dann zu 100% lokal ohne unsere Web-API bei euch betrieben werden. Nach einem Beta Test freuen wir uns auf einen offiziellen Push in das nächste EVCC Release. Mit der Migration auf Modbus für EVCC haben wir dann die Möglichkeit unabhängig von EVCC nachfolgend die neue Dev. Web-API für eure Use-Cases auszurollen. 

Gruß, Stephan

VolkerK62, 2023-01-15:
@stephanschwebe Vielen Dank. 
Verstehe ich das richtig, dass der Fehler in der 3.R2DA nicht enthalten ist?

Und dann noch ein paar Fragen, die möglicherweise für die Implementierung in evcc wichtig sind:
- Kann der Ladestrom in 0,1A Schritten geregelt werden?
- Laut Dokumentation (Beschreibung Register 0093) muss sich die Box für die Modbus-Steuerung im Modus "Plug and Charge" befinden. Das ist als Dauerzustand nicht immer gewünscht. Könnte man beim Fahrzeug connect diesen Modus einschalten und beim disconnect wieder in den "RFID" Modus zurückschalten?

stephanschwebe, 2023-01-15:
Hallo Volker, 

ja, wir installieren bei allen Usern DaheimLader Smart / V1 / V2 Usern als stabiles Release 3.R2DA. Bei den Usern die bislang eine "DN" Software haben, installieren wir die gleiche Software und installieren nachfolgend zusätzlich einen Hotfix (so dass kein FI Fehlertest auf der Wallbox selbst durchgeführt wird). 

Der Ladestrom kann in 1A Schritten gesteuert werden. Wenn der Ladevorgang pausiert werden soll, kann auch ein 0A Kommando geschickt werden. Alternativ ermöglichen wir mit RSDA auch ein Stoppen und späteres Fortsetzen (=Neustarten) des Ladevorgangs. 

EVCC muss in die Lage versetzt werden den RFID Kartenschutz zu umgehen. Daher kann mit Register 0093 der Kartenleser deaktiviert werden. Wichtig ist, wie du schreibst, das nach Beendigung des Ladevorgangs der alte Status zurückgeschrieben wird. Dies lässt sich sehr gut über das Register 0000 / Status erkennen.

StevieC121176, 2023-01-19:
> Hallo Volker,

> 
> Der Ladestrom kann in 1A Schritten gesteuert werden. Wenn der Ladevorgang pausiert werden soll, kann auch ein 0A Kommando geschickt werden. Alternativ ermöglichen wir mit RSDA auch ein Stoppen und späteres Fortsetzen (=Neustarten) des Ladevorgangs.
> 

Gibt es da womöglich einen Fehler in der Doku? Dort steht unter Beschreibung des Registers 0034 das in 0,1 A Schritten eingestellt werden kann.

stephanschwebe, 2023-01-19:
Ja, der Fehler liegt in der Doku. Es muss dort 1A Schritte heißen. Wir haben es korrigiert.

premultiply, 2023-01-19:
Kann man die Phasenstrom- und -spannungsregister en block lesen?

Das ist irgendwie seltsam. 16 bit Register mit je einem undefinierten Leerregister dazwischen.

Die alle einzeln zu lesen ist relativ ineffizient.

stephanschwebe, 2023-01-19:
Die Gesamtladeleistung in Summe aller drei Phasen (in Watt) siehst du im Register 0012 (Total Charging Power). Die jeweiligen Werte pro Phase sind für L1 in 0006, L2 in 0008 und L3 in 0010.

premultiply, 2023-01-19:
Schon klar, ich brauche aber die Phasenströme und -spannungen und wille diese nur ungern alle einzeln abfragen müssen.
Normalerweise liegen diese bei anderen Wallboxen direkt hintereinander und können somit in einem Rutsch gelesen werden.

Bei DaheimLaden liegen hier jedoch zwischen den uint16-Registern noch weitere undokumentierte/ungenutzte Register dazwischen.
Hier könnte man auch notfalls jeweils 6 uint16 Register in einem Rutsch lesen und die zwischenliegenden ignorieren.
Dabei darf die Modbus-Implementierung der Box aber nicht über diese "Zwischenregister" stolpern.
Da ich so eine Box nicht habe bin ich da vorsichtig.
Im Draft wird bisher jedes Register einzeln gelesen was somit insgesamt 6 statt 2 Leseoperationen bedeutet und somit auch fast dreimal so lange dauert.

stephanschwebe, 2023-01-20:
Ich würde vorschlagen deine Draft Implementierung zu nutzen. Das Modbus Protokoll wird aktuell für einen weiteren Use-Case genutzt. Hier können wir die Register nicht durcheinander bringen.

VolkerK62, 2023-01-20:
@premultiply 
>Da ich so eine Box nicht habe bin ich da vorsichtig.

Kann ich dir mit einer Freigabe weiterhelfen?

premultiply, 2023-01-20:
Ja, bitte.

VolkerK62, 2023-01-21:
ok. da brauch ich aber Hilfe ;-(  
Ich habe eine mail an info@evcc.io geschickt.

premultiply, 2023-01-22:
@stephanschwebe Die Doku stimmt diesbezüglich mit der Realität überein: 0,1A

Alle Register der Box lassen sich in einem Rutsch lesen. Auch über laut Doku nicht definierten Registern hinweg.

andig, 2023-02-12:
Was fehlt denn hier noch?

stephanschwebe, 2023-02-12:
@andig Aus unserer Sicht ist die Implementierung fertig und auch von uns getestet. Utsav hatte letzte Woche finales Feedback an Premultiply geschickt. Es muss lediglich als Standard ID Tag anstelle "0" > "evcc" aufgenommen werden. Die bestehende Implementierung könnte damit in das nächste Release. Wie angemerkten Änderungen (v.a. fehlende individuelle ID Tags) werden im Rahmen eines weiteren Release bearbeitet werden.

VolkerK62, 2023-02-12:
Aber, mWn sollte es als zusätzliche Wallbox aufgenommen werden und nicht die bestehende Implementierung ersetzen, oder?

stephanschwebe, 2023-02-12:
Ja, genau. Als "Daheimladen (Modbus)". Aktuell steht erst die Ethernet Modbus TCP Schnittstelle zur Verfügung. Wir planen bis Ende des Monats auch die Wifi Schnittstelle zur Verfügung zu stellen. Mit der bisherigen Cloud Schnittstelle kann dann sukzessive auf Modbus umgestellt werden.

andig, 2023-02-12:
Im Moment würde das die bestehende Implementierung ersetzen- da müssten wir also nochmal ran. Wäre Eure Strategie, die Cloudlösung im Zusammenspiel mit evcc vollständig abzulösen?

> Es muss lediglich als Standard ID Tag anstelle "0" > "evcc" aufgenommen werden

Das scheint noch nicht drin- wo müsste das hin? Einmalig beim Start?

stephanschwebe, 2023-02-12:
Ja, mittelfristig möchten wir die Dev-API in Q2-23 komplett überarbeiten. Dafür müssen bis dahin alle User auf Modbus migriert werden. In einer ersten Migrationsphase möchten wir die LAN User auf Modbus Umstellen. Anfang März wären dann die Wifi User dran.

Wir gehen davon aus, dass bis Ende März dann alle EVCC-User migriert sind und wir dann die Cloud-Schnittstelle deaktivieren bzw. komplett als offene Cloud-API umbauen können. 

Der ID-Tag wird für die Zuordnung / Auswertung der Ladevorgänge genutzt. Aktuell überträgt EVCC hier eine "0". Für eine Abgrenzbarkeit wäre hier "evcc" für die Wallbox App / Online-Portal super.

andig, 2023-02-12:
Meine Frage war eher: wie übertragen wir das? In welches Register muss das rein? Reicht einmalig beim Start von evcc?

VolkerK62, 2023-02-12:
@stephanschwebe verstehe ich das richtig?
Wenn ich mit RFID Karte freischalte, wird die Kartennummer in Register 54 übertragen. 
Wenn evcc die Ladung startet, soll in dieser Register "EVCC" eingetragen werden?

StevieC121176, 2023-02-12:
> @stephanschwebe verstehe ich das richtig?
> Wenn ich mit RFID Karte freischalte, wird die Kartennummer in Register 54 übertragen.
> Wenn evcc die Ladung startet, soll in dieser Register "EVCC" eingetragen werden?

So hab ich das auch verstanden. Damit kann dann in der Ladehistorie der Daheimladen App auch nach EVCC gefiltert werden.

VolkerK62, 2023-02-13:
Dann stellt sich die Frage, was passiert, oder soll passieren, wenn ich die RFID in evcc zur Fahrzeugidentifizierung nutze.

Philipp23, 2023-02-13:
Es wäre natürlich nice, wenn es die Option gäbe, dass der konfigurierte Identifier des Vehicles an die Wallbox übergeben wird. Dann könnte man auch über daheimladen.com die Ladevorgänge auswerten.

stephanschwebe, 2023-02-13:
Die Daten der Ladevorgänge werden mittels OCPP an die App / Online-Portal übertragen. Dies erfolgt unabhängig von EVCC. Eine vollständige Integration zwischen dem Modbus Protocol und OCPP muss noch folgen. Aktuell werden die Werte aus Register 0054 nur gelesen und nicht geschrieben. 

@andig Gibt es bei EVCC bereits andere Implementierungen von RFID Lesern? Theoretisch könnte die Legitimierung auch für EVCC initial durch die RFID Karte erfolgen. EVCC müsste dann auf den Charging Start "Starting" warten und ab diesen Zeitpunkt mittels Modbus die Steuerung übernehmen (und nicht mittels Remote-Start). Damit würde zunächst über das Setting "Current = 0A" der Ladevorgang pausiert werden. Bei dieser Implementierung wäre aber die Wallbox stets im Lademodus und würde dann auf den PV-Strom warten. 

Mit der jetzigen Implementierung wird der Ladevorgang erst beim PV Überschuss gestartet (was m.E. die saubere Variante ist). Aber das Requirement die RFID Karten bzw. Nutzer den EVCC Ladevorgängen ist bei uns angekommen.  

Im ersten EVCC Modbus Release würde ich zunächst das RFID Feature entfernen und dann im nächsten Release nachreichen. Das gibt euch allen dann auch die Möglichkeit die Implementierung auf Herz-und-Nieren zu testen.

VG,

Stephan

premultiply, 2023-02-13:
Ja, RFID-Implementierungen gibt es schon bei anderen Boxen.

Für RFID ist es obligatorisch, dass das Session-Management sauber implementiert ist.
Das ist derzeit bei dieser Firmware noch Verbesserungswürdig.

Für evcc essentiell ist, dass der Charger
1. eine Session mit der Authentifizierung beginnt und mit der Trennung der Kabelverbindung endet. Die Authentifizierung darf auch automatisch erfolgen (durch Anstecken des Kabel, Open-Access).
2. Während der Session die dafür verwendete RFID-Kennung (Identifier) über ein Register abrufbar ist.
3. Die Ladefreigabe (hier über die Stromregelung) unabhängig von der Session bleibt. D.h. der zuletzt von evcc als steuernde Instanz gesetzte Zustand muss auch nach dem Sessionende erhalten bleiben und nicht eigenmächtig von der Box mit dem Beginn einer neuen Session verändert/zurückgesetzt werden. Auch muss diese Freigabe jederzeit ausserhalb einer Session les- und schreibbar sein.

Der Charger kann/darf eine eigene RFID-Authentfizierung zum definierten Start einer Session durchführen (z. B. interne Liste der erlaubten Karten). Notwendig ist das nicht. Es muss nur sichergestellt sein, dass evcc das zuletzt gelesene Token bzw. das letztlich freigegebene Token während einer aktiven Session lesen kann und mit den in der Konfiguration hinterlegten Werten abgleichen kann. Hiermit wird dann z. B. auch eine manuelle Fahrzeugzuordnung anhand von verschiedenen Karten ermöglicht.

Mit dem Punkt 3 gibt es in der Firmware derzeit noch Probleme. Daher habe ich derzeit als Workaround eingebaut, dass in jeder Ladepause die Session komplett beendet wird. Das ist aber absehbar inkompatibel mit einer zukünftigen RFID-Authentifizierung.

VolkerK62, 2023-02-15:
Sorry, dass ich nachfrage. Wie geht es jetzt weiter? Wartet irgendjemand auf irgendetwas? 
Könnte man trotz der RFID-Probleme implementieren oder braucht es erst das Firmware Update?

StevieC121176, 2023-02-15:
Es wäre schön wenn zukünftig auch in 0,1A Schritten gesteuert werden könnte. Die Regelung in 1A Schritten ist bei 3phasig schon recht "grob"

stephanschwebe, 2023-02-16:
Aus unserer Sicht kann die getestete Modbus Implementierung als eigene Ladestation "DaheimLaden Modbus" aufgenommen werden. Die RFID Implementierung, wie auch ID-Tags gehen wir im nächsten Release gesondert an. 

@andig Könnt ihr bitte die Changes im nächsten Release als neues Wallbox Template mit aufnehmen? 

@StevieC121176 Wir müssen uns bei den Implementierungen an eine möglichst hohe Kompatibilität zu allen Elektrofahrzeugen orientieren. Allein die Differenzierung beim Minimum-Ladestrom zwischen 6A vs. 8A (alle Renault & Smart Onboard-Lader) bringen heute in der Praxis schon immer wieder große Probleme (nicht nur in der Technik, sondern bei den Anwendern). 

In dem folgenden Artikel sind die Grundlagen der Ladekommunikation zwischen Wallbox und Fahrzeug gut zusammen gefasst: https://de.wikipedia.org/wiki/IEC_62196_Typ_2

Wir planen aktuell daher nicht, zeitnah eine 0,1A Schritte-Steuerung zu implementieren.

premultiply, 2023-02-16:
Sobald ich etwas Luft habe werde ich den PR entsprechend finalisieren.

Hier wäre es aber noch gut wenn @stephanschwebe auch noch an meine Anmerkungen zu den Implementierungsproblemen eingehen könnte.
Ich würde nur ungern im Zuge einer zukünftigen RFID-Erweiterung an der sonstigen Freigabelogik größere vollziehen müssen.

@stephanschwebe Die feingranulare Stromvorgabe hat keinerlei Einfluss auf irgendeine Kompatiblität und ist seit jeher Bestandteil des zugehörigen IEC-Standards. Es ist umgekehrt nirgends spezifiziert dass nur ganze Ampere-Stufen verwendet werden dürfen. Ein PWM-Signal ist immer ein nicht wertdiskretes Signal. Im Standard ist daher auch nur eine lineare Umrechnungsformel sowie ggf. Beispielwerte angegeben.
Das ist leider ein vielfach kopiertes Missverständnis von vielen Wallbox-Herstellern, oft auch aus der Historie heraus wo diese Größe über DIP-Schalter etc. fest eingestellt wurde.

Wenn ihr daran einfach etwas ändern könnt bitte tut es in eurem eigenen kommerziellen Interesse.
Die Kunden schauen heute gezielt nach diesem Feature. Boxen die dies technisch nicht können fallen direkt durch das Suchraster.
Ihr müsst dies ja nicht über eure eigene App manuell so fein einstellbar machen aber mindestens über die Modbus-Schnittstelle für externe Steuerungen sollte es möglich sein. Hier wird Stand heute unnötig gezielt auf ganze Ampere abgerundet obwohl das betreffende Register alle Voraussetzungen bietet.
Wäre daher eine sehr einfache und abwärtskompatible Modifikation in eurer Firmware.

stephanschwebe, 2023-02-16:
@premultiply Eure Anforderungen haben wir verstanden. Mit der aktuellen Firmware der Wallbox müssen signifikante Veränderungen für das EVCC-RFID Handling gemacht werden. Das Modbus Protokoll wird auch für andere Zwecke genutzt. Damit müssen wir abwägen, wie wir alle Anforderungen zusammen bringen. 

Wir werden im nächsten Release das Read-only Register 54ff (alle Ziffern der Online-RFID Karte werden aktuell als DEC Wert angezeigt) in ein Read-Write Register ändern. Im YAML File könnte der User einen statischen selbst gewählten ID Tag hinterlegen. EVCC ist dann in der Lage im Startvorgang den ID Tag zu überschreiben. Die Ladevorgänge können damit je nach Wunsch des User einem Zweck zugeordnet werden. 

Für eine komplexere Ladesteuerung muss der User an der Station entscheiden können, ob er die Box in den reinen EVCC-PV-Steuerungsmodus schickt. Mit dem Touch Display lässt sich das einfacher realisieren. Für die DaheimLader Smart / V1 User müsste dieses Setting auf dem Konfigurations-Hotspot der Wallbox gesetzt werden. Damit muss als Frontend stets EVCC genutzt werden.

Die Ladesteuerung in kleineren Abständen werden wir in den kommenden Wochen in unsere regelmäßigen Fahrzeug-Tests mit einfließen lassen. 

Es wäre super, wenn der jetzigen Stand - ohne RFID Karten / als eigenes Template - im nächsten Release aufgenommen werden kann.

Philipp23, 2023-02-17:
@stephanschwebe Gibt es schon ein Datum, wann das Firmware-Update der Box kommt, um dieses Feature auch über WLAN zu nutzen?

stephanschwebe, 2023-02-17:
Die Firmware mit Modbus TCP Wifi für den DaheimLader Touch ist seit gestern im internen funktionalen Beta Test. Die Firmware für den DaheimLader Smart / V1 wird bis Mitte kommender Woche nachgezogen. Wir planen dann in KW10 mit dem Public Beta Test zu starten.

VolkerK62, 2023-02-19:
Hat mit dem Issue nur insofern zu tun, dass die Umsetzung wahrscheinlich ist die Lösung ist.
Mir ist in den vergangenen Wochen aufgefallen, dass die Wallbox besonders Sonntags gerne mal die Verbindung zum Server verliert (500er Fehler) und dann nicht mehr durch evcc regelbar ist und dann wird gnadenlos mit der zuletzt eingestellten Leistung weitergeladen.

StevieC121176, 2023-02-19:
Dieses Phänomen hatte ich auch als ich meine Box über WLAN angeschlossen hatte. Seit der Verbindung über LAN zu einem FritzWLANRepeater hatte ich es bis jetzt nicht mehr. Ich denke aber auch, sobald das mit Modbus läuft sollten Verbindungsprobleme der Vergangenheit angehören.

VolkerK62, 2023-02-20:
Ist per LAN angeschlossen. Power off, Power on und dann geht es wieder. Es scheint als würden Box und Daheimladen-Server die Verbindung zueinenander verlieren. Ping und lokale Modbus-Abfragen funktionieren.

closed 2023-02-22 by premultiply

stephanschwebe, 2023-02-22:
Liebe DaheimLaden EVCC User, 

wir haben zwischenzeitlich die Nutzung der WLAN Modbus Schnittstelle für den DaheimLader Touch für einen Public Beta Test freigegeben. Bitte schreibt uns eine kurze E-Mail (info@daheimladen.de) mit dem Hinweis EVCC Test mit Firmware 1.11. Wir installieren euch dann gern die Beta-Firmware zum Testen. 

Gleiches planen wir für die WLAN Modbus Schnittstelle für den DaheimLader Smart (ehemals V1 / V2) am Freitag (24.2.). 

Viele Grüße,

Stephan

Philipp23, 2023-02-24:
@stephanschwebe Ich habe seit heute die Firmware 1.11 auf meiner Wallbox drauf. Modbus über WLAN scheint aber nicht zu funktionieren.

Diagnose: Ich habe einen Portscan über das gesamtes Netzwerk gemacht und konnte kein Gerät finden (ausgenommen meinen Wechselrichter) wo der Port 502 offen ist.
Darüber hinaus ist es sehr schwer festzustellen, welche IP-Adresse die Wallbox eigentlich haben soll. Deaktiviert man DHCP auf der Wallbox, dann gilt die dort fix eingestellte IP-Adresse offensichtlich nur für den LAN Anschluss. Die IP-Adresse für den WLAN Anschluss kann ich nicht beeinflussen.
Mit etwas Analyse über den DHCP-Server meines Routers konnte ich die IP-Adresse der Wallbox am WLAN Anschluss herausfinden - zumindest glaube ich das. 
Der WLAN Anschluss zeigt sich in meinem Portscanner als "Zhongshan BT technology.co.,ltd" - allerdings ist dort der Port 502 nicht offen.

Bei Verbindung über den LAN Anschluss funktioniert es nach wie vor ohne Probleme.

Kann mir da bitte jemand weiterhelfen ...

PV-Heini, 2023-02-25:
Ich habe auf der Smart-WB mit externem FI-Schalter die Firmware ...DA aufgespielt bekommen. Gestern habe ich auf 0.113.0 aktualisiert und in der YAML das Template daheimladen-mb konfiguriert. Das Laden funktioniert für geschätzte 10s und bricht dann mit Fehlermeldungen ab (siehe unten). Ich vermute ´mal, das ist ein Thema für den Service von DaheimLaden, oder? @stephanschwebe hatte ja ausgeführt, dass die DN-Boxen zwar die DA-Firmware bekommen, aber zusätzlich wegen des externen FI noch ein Patch notwendig ist. Macht sonst jemand die gleichen Beobachtungen mit DN-Box?

[image]

stephanschwebe, 2023-02-25:
> @stephanschwebe Ich habe seit heute die Firmware 1.11 auf meiner Wallbox drauf. Modbus über WLAN scheint aber nicht zu funktionieren.
>
> Diagnose: Ich habe einen Portscan über das gesamtes Netzwerk [...]

Hallo Philipp,

nach einigen Tests sehen wir aktuell, dass die statische IP mit WLAN nicht funktioniert. Daher bitte für die Test auf DHCP umstellen und sich vom Router eine neue IP Adresse vergeben lassen (in der Fritzbox wird kein Name der Station sondern die IP Adresse als PC-192.168.xxx.yyy / etc. angezeigt). 

In einzelnen Fällen scheint der Port 502 sich wieder zu schließen (bzw. mit anderen Services wie IOBroker in Konflikt zu stehen). Nach einem Neustart der Wallbox ist dieser wieder da und ansprechbar. Wir analysieren aktuell alle Fehlermeldungen von euch ...

Gruß,

Stephan

stephanschwebe, 2023-02-25:
> Ich habe auf der Smart-WB mit externem FI-Schalter die Firmware ...DA aufgespielt bekommen. Gestern habe ich auf 0.113.0 aktualisiert und in der YAML das Template daheimladen-mb konfiguriert. Das [...]

@PV-Heini Bei dir ist die Firmware 3.10DA installiert? Der Unterschied DA / DN dürfte bei der Steuerung von EVCC keinen Unterschied machen. Der einzige Unterschied ist, dass die Software DA zusätzlich den internen FI steuert. Da du keine internen, sondern einen externen FI hast, muss hierfür bei dir ein Hotfix installiert werden. Aber das hat wie gesagt, nix mit EVCC zu tun. 

Die IP Adresse deiner Wallbox wird mittels DHCP vergeben? Bei statischen IP Adressen gibt es wie oben geschrieben Probleme. Der Fehler ist bekannt.

Philipp23, 2023-02-26:
> Hallo Philipp,
>
> nach einigen Tests sehen wir aktuell, dass die statische IP mit WLAN nicht funktioniert. Daher bitte für die Test auf DHCP umstellen und sich vom Router eine neue IP Adresse vergeben [...]

Hallo Stephan,
habe das gestern noch einmal mit DHCP ausprobiert. Unter der IP-Adresse, die vom Router vergeben wurde, ist definitiv kein Port 502 offen. Habe die Wallbox auch mehrfach neu gestartet, ohne erfolgt. Es konnte nie eine Verbindung zu Port 502 hergestellt werden. Zur Sicherheit habe ich dann noch einen Test mit einem anderen Router und einem anderen WLAN gemacht, um sicherzugehen dass es keine störenden Einflüsse anderer Geräte gibt, hat aber leider auch nicht funktioniert. Keine Verbindung möglich. Bei Verbindung über LAN klappt es jedes mal sofort und ohne Probleme.

Schöne Grüße, Philipp

PV-Heini, 2023-02-27:
@stephanschwebe Ich habe auf EVCC 0.113.2 aktualisiert und habe bisher keine Probleme. Ich beobachte ´mal weiter ...

simowNgithub, 2023-03-08:
Ich habe die 114.1 und die WLAN modbus firmware auf der Wallbox. Verbindung steht, Ladevorgang konnte auch starten, jedoch bekomme ich weiterhin die Meldungen: 

[image]

TheAndi, 2024-06-16:
> @stephanschwebe Die Doku stimmt diesbezüglich mit der Realität überein: 0,1A

Was ist denn aktuell hier der Stand? Kann/macht die WB (Smart) **0,1 A Schritte** oder nicht? Bzw. kommt das noch?
Die aktuellste "DaheimLaden ModBus TCP Spezifikation" die ich online finde, ist VERSION 1.4 vom MÄRZ 2023 und sagt 0,1 A...

Sind die andere Bugs bzw. Unzulänglichkeiten der Firmware (die man hier so liest) mittlerweile behoben oder würdet ihr eher von der Box abraten?
Alternativ hab ich eine günstige Entratek Dot Pro 2 im Auge (leider ohne LAN). 

Da ich aktuell vor dem Kauf stehe und das für mich ein Kriterium ist, wäre ich für eine kurze Antwort dankbar :)

[2 more replies]

PR #5836 (2023-02-22): Add DaheimLaden ModbusTCP
Fixes #5520

Breaking Change: Replaces DaheimLaden Cloud implementation by local ModbusTCP implementation.
Recent firmware supporting ModbusTCP is required.
Needs small modification of evcc config: Remove `token:` and `stationid:` and add local `host:` entry.

Some pending issues needs to be kindly investigated by DaheimLaden please:  /cc @stephanschwebe @utsavanand2
- [ ] Session management and charging state is weird. Setting current to 0 and pausing the charge does not change the state to "connected, not charging" (2). It will stay at "charging" (4). Even the charg [... cut]