evcc-io/evcc #5525: Shelly 3EM Meter (grid), Verbrauch wird als Einspeisung erkannt

thomas-bauer-19, 2022-12-21:
Hallo,
ich schreibe jetzt mal in Deutsch, wenn nicht gewünscht dann kommt gerne nochmal eine Version in Englisch ;-)

Nach einem Update auf die aktuelle Version 0.110.0 wird bei mir der Verbrauch meines Shelly 3EM in der WebGUI als Einspeisung anzeigt was aber nicht korrekt ist.

evcc config:
meters:
- name: grid
   type : template
   template: shelly-3em
   usage: grid
   host: x.x.x.x

Output evcc meter grid:
Power:      -1364W
Energy:         380.0kWh
Current: xxx

Wurde in den letzten Versionen dazu was angepasst was ich übersehen habe?

premultiply, 2022-12-21:
`evcc meter --log trace` bitte

thomas-bauer-19, 2022-12-21:
root@evcc:/home/user# evcc meter grid --log trace
[main  ] INFO 2022/12/21 12:10:24 evcc 0.110.0
[main  ] INFO 2022/12/21 12:10:24 using config file: /etc/evcc.yaml
[db    ] INFO 2022/12/21 12:10:25 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2022/12/21 12:10:25 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2022/12/21 12:10:25 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2022/12/21 12:10:25 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2022/12/21 12:10:25 SELECT * FROM `settings` 6 <nil>
[eebus ] INFO 2022/12/21 12:10:25 Local SKI:  xxx
[eebus ] DEBUG 2022/12/21 12:10:25 mdns: announce
[eebus ] DEBUG 2022/12/21 12:10:25 mdns: using zeroconf
[eebus ] DEBUG 2022/12/21 12:10:25 starting websocket server on :4712
[eebus ] ERROR 2022/12/21 12:10:25 error during websocket server starting: listen tcp :4712: bind: address already in use
[http  ] TRACE 2022/12/21 12:10:25 GET http://xxxx/status
[http  ] TRACE 2022/12/21 12:10:25 {"wifi_sta":{"connected":true,"ssid":"xxx","ip":"xxx","rssi":-76},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":true},"time":"12:10","unixtime":1671621024,"serial":44478,"has_update":false,"mac":"485519C9B10A","cfg_changed_cnt":0,"actions_stats":{"skipped":0},"relays":[{"ison":false,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"input"}],"emeters":[{"power":-883.03,"pf":-0.89,"current":4.24,"voltage":234.79,"is_valid":true,"total":76495.4,"total_returned":1566609.3},{"power":-88.75,"pf":-0.64,"current":0.60,"voltage":236.28,"is_valid":true,"total":36462.4,"total_returned":998410.0},{"power":-8.32,"pf":-0.08,"current":0.46,"voltage":239.50,"is_valid":true,"total":266151.6,"total_returned":611096.2}],"total_power":-980.10,"emeter_n":{"current":0.00,"ixsum":3.94,"mismatch":false,"is_valid":false},"fs_mounted":true,"v_data":1,"ct_calst":0,"update":{"status":"idle","has_update":false,"new_version":"20221027-110030/v1.12.1-ga9117d3","old_version":"20221027-110030/v1.12.1-ga9117d3"},"ram_total":49928,"ram_free":30216,"fs_size":233681,"fs_free":154867,"uptime":3707126}
[http  ] TRACE 2022/12/21 12:10:25 GET http://xxxx/status
[http  ] TRACE 2022/12/21 12:10:25 {"wifi_sta":{"connected":true,"ssid":"xxxx","ip":"xxx","rssi":-76},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":true},"time":"12:10","unixtime":1671621025,"serial":44478,"has_update":false,"mac":"485519C9B10A","cfg_changed_cnt":0,"actions_stats":{"skipped":0},"relays":[{"ison":false,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"input"}],"emeters":[{"power":-883.03,"pf":-0.89,"current":4.24,"voltage":234.79,"is_valid":true,"total":76495.4,"total_returned":1566609.3},{"power":-88.75,"pf":-0.64,"current":0.60,"voltage":236.28,"is_valid":true,"total":36462.4,"total_returned":998410.0},{"power":-8.32,"pf":-0.08,"current":0.46,"voltage":239.50,"is_valid":true,"total":266151.6,"total_returned":611096.2}],"total_power":-980.10,"emeter_n":{"current":0.00,"ixsum":3.94,"mismatch":false,"is_valid":false},"fs_mounted":true,"v_data":1,"ct_calst":0,"update":{"status":"idle","has_update":false,"new_version":"20221027-110030/v1.12.1-ga9117d3","old_version":"20221027-110030/v1.12.1-ga9117d3"},"ram_total":49928,"ram_free":30036,"fs_size":233681,"fs_free":154867,"uptime":3707126}
[http  ] TRACE 2022/12/21 12:10:25 GET http://xxxxemeter/0
[http  ] TRACE 2022/12/21 12:10:25 {"power":-883.03,"pf":-0.89,"current":4.24,"voltage":234.79,"is_valid":true,"total":76495.4,"total_returned":1566609.3}
[http  ] TRACE 2022/12/21 12:10:25 GET http://xxxx/emeter/1
[http  ] TRACE 2022/12/21 12:10:25 {"power":-88.75,"pf":-0.64,"current":0.60,"voltage":236.28,"is_valid":true,"total":36462.4,"total_returned":998410.0}
[http  ] TRACE 2022/12/21 12:10:25 GET http://xxxx/emeter/2
[http  ] TRACE 2022/12/21 12:10:25 {"power":-8.32,"pf":-0.08,"current":0.46,"voltage":239.50,"is_valid":true,"total":266151.6,"total_returned":611096.2}
Power:          -980W
Energy:         379.1kWh
Current L1..L3: 4.24A 0.6A 0.46A
[eebus ] DEBUG 2022/12/21 12:10:25 mdns: stop announcement

VolkerK62, 2022-12-21:
Bei meinem Shelly 3EM wird weiterhin alles richtig gemessen und angezeigt (aktuelle Nightly). Ich habe aber kein EEBUS.

thomas-bauer-19, 2022-12-21:
Bei mir wurde am Anschlusspunkt umgebaut, ich werde nochmal kontrollieren ob die Ausrichtung der Current Sensoren richtig ist.
Wird dir bei Verbrauch ein positives oder negatives Ergebnis angezeigt? Mich irritiert der negative PF...

VolkerK62, 2022-12-21:
Verbrauch= power positiv. So wie ich das beobachtet habe, kann dabei "pf" sowohl positiv, als auch negativ sein. Hat irgendwas mit Blind/Schein/Wirk Leistung zu tun.

premultiply, 2022-12-21:
> ```
> [http  ] TRACE 2022/12/21 12:10:25 GET http://xxxxemeter/0
> [http  ] TRACE 2022/12/21 12:10:25 [...]

Da werden die Wandler falschrum montiert bzw. angeschossen worden sein.

closed 2022-12-21 by premultiply