evcc-io/evcc #5543: Vehicles OVMS minSoc, targetSoc und Mode wird nicht automatisch gesetzt

Tailer67, 2022-12-22:
### Discussed in https://github.com/evcc-io/evcc/discussions/5481

MinSoc, taregetSoc and Mode is not set when the vehicle is plugged into the Loadpoint.
```
vehicles:
- type: template
  template: ovms 
  title: e-Up  
  user: *****
  password: *****
  vehicleid: SI-CT67E  
  capacity: 16  
  name: ev1
  phases: 1
  mode: pv
  minSoC: 30
  targetSoC: 80
```
```
root@evcc:~# evcc vehicle
[main  ] INFO 2022/12/22 10:38:09 evcc 0.110.0
[main  ] INFO 2022/12/22 10:38:09 using config file: /etc/evcc.yaml
[mqtt  ] INFO 2022/12/22 10:38:09 connecting evcc-749441449 at tcp://192.168.178.12:1883
SoC:           80%
Charge status: C
Range:         75km
Odometer:      23693km
Finish time:   2022-12-22 11:43:00 +0000 UTC
Position:      xxx,xxx
Capacity:      16.0kWh
```

The vehicle is not queried with the onIdentify parameter

```
vehicles:
- type: ovms
  #template: ovms 
  title: e-Up  
  user: xxx
  password: xxx
  vehicleid: SI-CT67E  
  capacity: 16
  name: ev6
  phases: 1
  onIdentify:
    mode: pv
    minSoC: 30
    targetSoC: 80
```

```
root@evcc:~# evcc vehicle
[main  ] INFO 2022/12/22 11:31:54 evcc 0.110.0
[main  ] INFO 2022/12/22 11:31:54 using config file: /etc/evcc.yaml
[mqtt  ] INFO 2022/12/22 11:31:54 connecting evcc-937573924 at tcp://192.168.178.12:1883
.SoC:          Get "https://:6869/api/cookie?username=xxx&password=xxx": dial tcp :6869: connect: connection refused
Charge status: Get "https://:6869/api/cookie?username=xxx&password=xxx": dial tcp :6869: connect: connection refused
Range:         Get "https://:6869/api/cookie?username=xxx&password=xxx": dial tcp :6869: connect: connection refused
Odometer:      Get "https://:6869/api/cookie?username=xxx&password=xxx": dial tcp :6869: connect: connection refused
Finish time:   Get "https://:6869/api/cookie?username=xxx&password=xxx": dial tcp :6869: connect: connection refused
Position:      Get "https://:6869/api/cookie?username=xxx&password=xxx": dial tcp :6869: connect: connection refused
Capacity:      16.0kWh
OnIdentified:  Mode:pv, MinSoC:30, TargetSoC:80
root@evcc:~# 
```

The vehicle query and the setting of the loading parameters only work when the server "dexters-web.de" is specified

```
vehicles:
- type: ovms
  title: e-Up
  user: xxx
  password: xxx
  vehicleid: SI-CT67E
  capacity: 16
  name: ev6
  phases: 1
  onIdentify:
    mode: pv
    minSoC: 30
    targetSoC: 80
  server: dexters-web.de
```

```
root@evcc:~# evcc vehicle
[main  ] INFO 2022/12/22 14:49:56 evcc 0.110.0
[main  ] INFO 2022/12/22 14:49:56 using config file: /etc/evcc.yaml
[mqtt  ] INFO 2022/12/22 14:49:56 connecting evcc-1456246850 at tcp://192.168.178.12:1883
SoC:           64%
Charge status: B
Range:         65km
Odometer:      23732km
Finish time:   2022-12-22 16:34:00 +0000 UTC
Position:      xxx,xxx
Capacity:      16.0kWh
OnIdentified:  TargetSoC:80, Mode:pv, MinSoC:30
root@evcc:~# 
```

andig, 2022-12-22:
> connect: connection refused

Please provide logfiles without errors.

andig, 2022-12-22:
Nvm, fixed in https://github.com/evcc-io/evcc/commit/374c9907fc43b16c7c6dfd85a611cbb573d2c514

closed 2022-12-22 by andig