evcc-io/evcc #5544: charger logic error mit openWB, evcc macht keine Anstalten das Problem zu beheben

MarkusGH, 2022-12-22:
### Describe the bug

Nach Anstecken des Fahrzeugs lädt openWB mit 3,5 kW, evcc meldet "charger logic error: disabled but charging" macht aber keine Anstalten openWB anzusteuern. Wenn man dann im UI auf "Schnell" und "Aus" umschält wird openWB korrekt angesteuert und das Problem ist weg.

### Steps to reproduce

Siehe oben

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: localhost
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 3s # control cycle interval

log: info
levels:
  porsche: trace
  lp-1: debug
  lp-2: debug
  openwb: trace

mqtt:
  broker: mqtt.xxx.de:1883
  topic: evcc

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.1570 # [currency]/kWh
  feedin:
    type: fixed
    price: 0.0816 # [currency]/kWh

meters:
- name: grid1
  type: movingaverage
  decay: 1 # 0.5
  meter:
    type: custom
    power:
      source: mqtt
      topic: mbmd/solaredge1-1-1/Power
      scale: -1  # Die Leistung wird in Watt erwartet. Falls z.b. kW geliefert wird, kann dies hier mit dem Wert 1000 angepasst werden
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    energy:
      source: mqtt
      topic: mbmd/solaredge1-1-1/Import
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    currents:
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L1
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L2
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L3
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
- name: pv1
  type: custom
  power:
    source: mqtt
    topic: mbmd/solaredge1-1/Power
    timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
  energy:
    source: mqtt
    topic: mbmd/solaredge1-1/Export
    timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
  currents:
    - source: mqtt
      topic: mbmd/solaredge1-1/Current/L1
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    - source: mqtt
      topic: mbmd/solaredge1-1/Current/L2
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    - source: mqtt
      topic: mbmd/solaredge1-1/Current/L3
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist

chargers:
- name: pmcc1
  type: template
  template: pmcc
  ski: xxx  # Die SKI der Wallbox, ist üblicherweise im Web Interface der Wallbox zu finden
- name: openwb1
  type: openwb
  phases1p3p: true
  broker: openwb.xxx.de # openWB IP
  id: 1                      # loadpoint ID
- name: openwb2
  type: openwb
  phases1p3p: true
  broker: openwb.xxx.de # openWB IP
  id: 2                      # loadpoint ID

vehicles:
- name: ev1
  type: template
  template: porsche
  title: Das Batmobil  # Wird in der Benutzeroberfläche angezeigt
  user: xxx  # Benutzername um auf das Gerät zuzugreifen
  password: xxx  # Passwort des Benutzerkontos
  vin: xxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind
  capacity: 17.9  # Akku-Kapazität in kWh
  phases: 2
  maxCurrent: 16
  cache: 20m

loadpoints:
- title: Carport
  charger: openwb2
  resetOnDisconnect: true
  mode: pv
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true
  phases: 0
  mincurrent: 6
  maxcurrent: 32
  enable:
    threshold: -600
    delay: 1m
  disable:
    threshold: 800
    delay: 3m
  guardduration: 5m
- title: Straße
  charger: openwb1
  resetOnDisconnect: true
  mode: pv
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true
  phases: 0
  mincurrent: 6
  maxcurrent: 32
  enable:
    threshold: -600
    delay: 1m
  disable:
    threshold: 800
    delay: 3m

site:
  title: XXX
  meters:
    grid: grid1
    pvs:
    - pv1
  residualPower: -150
#  maxCurrent: 63

eebus:
  interfaces:
  - ens160
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          XXX
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          XXX
          -----END CERTIFICATE-----
```

### Log details

```text
Anhang
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.110.0

andig, 2022-12-22:
Da fehlt der Anhang :O

MarkusGH, 2022-12-22:
Du bist zu schnell :-)

andig, 2022-12-22:
Sieht prinzipbedingt aus bei der OpenWB. Leider gibts nix um den "wahren Status" zu erkennen. Die Umschaltung auf disable wäre schon im Schritt vorher erfolgt, da passte der Status aber. Anscheinend hat openWB ein Update verpasst. Da wir das nicht zurück lesen können ist das best effort. Lohnt wohl nicht, da Aufwand rein zu stecken falls openWB kein besseres Api bietet.

MarkusGH, 2022-12-23:
Verstehe ich nicht - die openWB meldet doch alles relevante:

recv openWB/lp/2/boolChargeStat: '1'
recv openWB/lp/2/W: '3548'

andig, 2022-12-23:
Das hat nix mit „enabled“ zu tun.

MarkusGH, 2022-12-23:
Aber wenn er lädt, dann ist er ja offensichtlich enabled. Anders herum geht es nicht, das ist schon klar.

MarkusGH, 2022-12-28:
Bei der eebus charger Implementation gibt es doch genau dasselbe Problem, die dortige Logik sollte man doch fast 1:1 übernehmen können:

````
func (c *EEBus) updateState() (api.ChargeStatus, error) {
	currentState, err := c.emobility.EVCurrentChargeState()
	if err != nil {
		return api.StatusNone, err
	}

	if !c.connected {
		return api.StatusNone, fmt.Errorf("%s charger reported as disconnected", c.ski)
	}

	switch currentState {
	case emobility.EVChargeStateTypeUnknown, emobility.EVChargeStateTypeUnplugged: // Unplugged
		c.expectedEnableState = false
		return api.StatusA, nil
	case emobility.EVChargeStateTypeFinished, emobility.EVChargeStateTypePaused: // Finished, Paused
		return api.StatusB, nil
	case emobility.EVChargeStateTypeActive: // Active
		if c.isCharging() {
			// we might already be enabled and charging due to connection issues
			c.expectedEnableState = true
			return api.StatusC, nil
		}
		return api.StatusB, nil
	case emobility.EVChargeStateTypeError: // Error
		return api.StatusF, nil
	}

	return api.StatusNone, fmt.Errorf("%s properties unknown result: %s", c.ski, currentState)
}

// Status implements the api.Charger interface
func (c *EEBus) Status() (api.ChargeStatus, error) {
	return c.updateState()
}

// Enabled implements the api.Charger interface
// should return true if the charger allows the EV to draw power
func (c *EEBus) Enabled() (bool, error) {
	_, err := c.updateState()
	return c.expectedEnableState, err
}
````

andig, 2022-12-28:
Gerne PR wenn Du eine gute Idee hast :)

MarkusGH, 2022-12-28:
Das stellen eines PR kostet mich 20x mehr Zeit als die Implementation.

Bitte sei doch so nett...

````
func (m *OpenWB) Enabled() (bool, error) {
	switch Status() {
	case api.StatusA:
		m.enabled = false
	case api.StatusC:
		m.enabled = true
	return m.enabled, nil
}
````

premultiply, 2022-12-28:
Das ist definitiv keine Lösung.
Enabled hat nichts mit dem Ladezustand zu tun.

MarkusGH, 2022-12-28:
Weder eebus, noch die OpenWB kennen einen enabled/disabled State.
Diese Logik wird genau so in eebus.go als Workaround verwendet.
Sehe keinen Grund warum das hier nicht funktionieren sollte.

andig, 2022-12-28:
> Das ist definitiv keine Lösung. Enabled hat nichts mit dem Ladezustand zu tun.
> 
> Daher bitte getesteten PR.

Ne, die Idee ist schon gut da wir "enabled" nicht kennen. Wenn er lädt muss es ja enabled sein. Find ich smart.

@MarkusGH Du kannst die openwb.go direkt in Github bearbeiten und speichern. Ohne git und alles!

MarkusGH, 2022-12-28:
Danke für die Hilfestellung. Ich entschuldige mich schon jetzt für die Fehler (go ist nicht meine Sprache).

closed 2022-12-29 by andig