evcc-io/evcc #5546: WARN und ERROR werden im UI nicht mehr angezeigt

VolkerK62, 2022-12-22:
### Describe the bug

Wenn ich z.B. die Ladung stoppe, erscheint im Log immer die (einmalige) Warnung "charger logic error: disabled but charging".
Früher wurde diese Warnung auch im UI angezeigt. Jetzt nicht mehr.

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.110.0 (80ff5516)

andig, 2022-12-22:
Was siehst Du wenn Du /api/state aufrufst?

/cc @naltatis

VolkerK62, 2022-12-23:
/api/state liefert  
`SyntaxError: JSON.parse: unexpected end of data at line 1 column 1 of the JSON data`

andig, 2022-12-23:
Probiers bitte mal in anderem Browser. Anscheiend ist das schon der Fehler- falsches JSON, wobei das wirklich schwer vorzustellen ist.

VolkerK62, 2022-12-23:
Firefox bringt den Fehler. Edge bringt nichts. Und die Smartphone App "http requests" sagt "Antwort ist leer". Raspi wurde auch schon neu gestartet.

andig, 2022-12-23:
Kannst Du das mal mit `curl` auf der Kommandozeile auf dem Raspi machen?

VolkerK62, 2022-12-23:
nix, keine Antwort. Habe zum Test noch andere Eingaben gemacht.

```
pi@raspberrypi:~ $ curl raspberrypi:7070/api/state
pi@raspberrypi:~ $ curl -X POST http://raspberrypi:7070/api/loadpoints/0/mode/off
{"result":"off"}
pi@raspberrypi:~ $ curl raspberrypi:7070/api/health
OK
```

andig, 2022-12-23:
curl -v http://raspberrypi:7070/api/state

VolkerK62, 2022-12-23:
```
pi@raspberrypi:~ $ curl -v http://raspberrypi:7070/api/state
*   Trying 127.0.1.1:7070...
* Connected to raspberrypi (127.0.1.1) port 7070 (#0)
> GET /api/state HTTP/1.1
> Host: raspberrypi:7070
> User-Agent: curl/7.74.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Content-Type: application/json; charset=UTF-8
< Vary: Accept-Encoding
< Date: Fri, 23 Dec 2022 08:38:25 GMT
< Content-Length: 0
< 
* Connection #0 to host raspberrypi left intact

```

andig, 2022-12-23:
Und evcc läuft? Keine Fehler im Logfile? Dann weiss ichs auch nicht :O

VolkerK62, 2022-12-23:
evcc läuft ohne Probleme. 

Ein Fehler im Log, aber der ist "richtig". Der Server für meinen Speicher-Soc ist momentan nicht erreichbar.
`Dez 23 09:38:11 raspberrypi evcc[2539]: [influx] ERROR 2022/12/23 09:38:11 is NaN`

Den bekomme ich jetzt auch im UI angezeigt
`httpd: failed to encode JSON: json: unsupported value: NaN`

andig, 2022-12-23:
Sehr gut, das dürfte das Problem sein. Schaue ich mir später an!

VolkerK62, 2022-12-23:
Hintergrund: Ich hole den Hausspeicher-Soc per script von einem Server und stelle ihn evcc per mqtt zur Verfügung. Dort kommt seit gestern Abend nur noch "NaN" an, weil der Server down ist.

andig, 2022-12-23:
@premultiply ich denke wir sollten `inf` und `nan` auf Rest als nil ausgeben. Auf mqtt wörtlich als String?

VolkerK62, 2022-12-23:
genau das war das Problem. Ich fange den Fehler jetzt im script ab und gebe "0" aus und schon funktioniert auch `/api/state` wieder.

andig, 2022-12-24:
Fixed in master. Frohe Weihnachten!

closed 2022-12-24 by andig

VolkerK62, 2022-12-30:
@andig 
Problem der fehlenden Warnungen besteht immer noch.
Auto lädt und ich habe mich gewundert, dass der Soc sich nicht erhöht.
Im Log war der ERROR "vehicle soc timeout" zu sehen, der im UI nicht angezeigt wurde.
Ich habe dann hyundai: trace eingestellt unde neu gestartet. Danach ist der Fehler leider nicht mehr aufgetreten.

Keine AHnung, ob das damit zusammenhängt, aber im MQTT Explorer bekomme ich unter dem topic `evcc/site/error` den Hinweis:
`version check failed: Get \"https://api.github.com/repos/evcc-io/evcc/releases/latest\": dial tcp: lookup api.github.com on 192.168.178.1:53: dial udp 192.168.178.1:53: connect: network is unreachable (installed: 0.110.1)`
Die IP ist die von meiner FritzBox.

andig, 2022-12-30:
Docker? Du hast anscheinend DNS kaputt gemacht.

andig, 2022-12-30:
@VolkerK62 was ist jetzt mit /api/state?

VolkerK62, 2022-12-31:
kein Docker. normal Debian

Api/state funktioniert.

`{"result":{"auth":{"vehicles":{}},"battery":[{"power":-47.97674418604651,"soc":40,"capacity":7.5}],"batteryCapacity":8,"batteryConfigured":true,"batteryPower":-47.97674418604651,"batterySoc":40,"bufferSoc":110,"currency":"EUR","gridConfigured":true,"gridCurrents":[15.55,15.71,15.34],"gridEnergy":299.8811,"gridPower":10762.04,"homePower":414.06325581395504,"loadpoints":[{"chargeCurrent":16,"chargeCurrents":[15.100000381469727,15,14.899999618530273],"chargeDuration":7870000000000,"chargePower":10300,"chargeRemainingDuration":167000000000,"chargeRemainingEnergy":477.72894965277544,"chargeTotalImport":1353.5,"chargedEnergy":22599.9755859375,"charging":true,"connected":true,"connectedDuration":0,"enabled":true,"maxCurrent":16,"minCurrent":6,"minSoc":0,"mode":"pv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":3,"phasesConfigured":null,"phasesEnabled":1,"planActive":true,"pvAction":"inactive","pvRemaining":0,"targetEnergy":0,"targetSoc":60,"targetTime":"2022-12-31T05:00:00Z","targetTimeActive":true,"targetTimeProjectedStart":"2022-12-31T05:57:06.666666667+01:00","title":"Carport","vehicleCapacity":72,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleIcon":"car","vehicleOdometer":11509,"vehiclePresent":true,"vehicleRange":236,"vehicleSoc":59,"vehicleTargetSoc":100,"vehicleTitle":"(T)Raumschiff"}],"prioritySoc":0,"pv":[{"power":0}],"pvConfigured":true,"pvPower":0,"residualPower":100,"savingsAmount":2.0967849833963017,"savingsEffectivePrice":0.27355399117735674,"savingsGridCharged":68.294892176932,"savingsSelfConsumptionCharged":10.99060596602587,"savingsSelfConsumptionPercent":13.862063332450733,"savingsSince":1670934918,"savingsTotalCharged":79.28549814295786,"siteTitle":"Haus","sponsor":"VolkerK62","tariffFeedIn":0.1095,"tariffGrid":0.3,"vehicles":["(T)Raumschiff"],"version":"0.110.1 (bf41bec1)"}}`

VolkerK62, 2023-01-03:
@andig  wegen des ERROR der im MQTT abgezeigt wird .... der taucht im trace-Log nicht auf (wo kommt der her?)
```
Jan 03 17:26:31 raspberrypi evcc[10651]: [main  ] TRACE 2023/01/03 17:26:31 GET https://api.github.com/repos/evcc-io/evcc/releases/latest

Jan 03 17:26:31 raspberrypi evcc[10651]: [main  ] TRACE 2023/01/03 17:26:31 {"url":"https://api.github.com/repos/evcc-io/evcc/releases/87052177","assets_url":"https://api.github.com/repos/evcc-io/evcc/releases/87052177/assets","upload_url":"https://uploads.github.com/repos/evcc-io/evcc/releases/87052177/assets{?name,label}","html_url":"https://github.com/evcc-io/evcc/releases/tag/0.110.1","id":87052177,"author":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","site_admin":false},"node_id":"RE_kwDODX4bUs4FME-R","tag_name":"0.110.1","target_commitish":"master","name":"0.110.1","draft":false,"prerelease":false,"created_at":"2022-12-23T07:30:29Z","published_at":"2022-12-23T08:47:42Z","assets":[{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/89249325","id":89249325,"node_id":"RA_kwDODX4bUs4FUdYt","name":"checksums.txt","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","site_admin":false},"content_type":"text/plain; charset=utf-8","state":"uploaded","size":754,"download_count":4,"created_at":"2022-12-23T09:01:49Z","updated_at":"2022-12-23T09:01:49Z","browser_download_url":"https://github.com/evcc-io/evcc/releases/download/0.110.1/checksums.txt"},{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/89249323","id":89249323,"node_id":"RA_kwDODX4bUs4FUdYr","name":"evcc_0.110.1_amd64.deb","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","site_admin":false},"content_type":"application/vnd.debian.binary-package","state":"uploaded","size":17538898,"download_count":7,"created_at":"2022-12-23T09:01:47Z","updated_at":"2022-12-23T09:01:48Z","browser_download_url":"https://github.com/evcc-io/evcc/releases/download/0.110.1/evcc_0.110.1_amd64.deb"},{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/89249324","id":89249324,"node_id":"RA_kwDODX4bUs4FUdYs","name":"evcc_0.110.1_arm64.deb","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","site_admin":false},"content_type":"application/vnd.debian.binary-package","state":"uploaded","s [... cut ...] thub.com/users/andig/received_events","type":"User","site_admin":false},"content_type":"application/gzip","state":"uploaded","size":16842061,"download_count":13,"created_at":"2022-12-23T09:01:42Z","updated_at":"2022-12-23T09:01:43Z","browser_download_url":"https://github.com/evcc-io/evcc/releases/download/0.110.1/evcc_0.110.1_linux_amd64.tar.gz"},{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/89249312","id":89249312,"node_id":"RA_kwDODX4bUs4FUdYg","name":"evcc_0.110.1_linu
```

VolkerK62, 2023-01-07:
@andig 
Diese Fehlermeldung https://github.com/evcc-io/evcc/issues/5546#issuecomment-1368051741 taucht nach "clear this topic" nicht mehr auf. 

Ich bekomme aber weiterhin keine Warn/Error Meldungen im UI. 
Auch im /api/state sind die nicht drin. 
Sind nur im log sichtbar.