evcc-io/evcc #5551: Vehicle Template für psa_car_controller

mdkeil, 2022-12-23:
Da sich der Aufwand in Grenzen haltten sollte, möchte ich mich gerne für die Implementierung eines vehicle-templates für den [psa_car_controller](https://github.com/flobz/psa_car_controller) stark machen. Meiner Ansicht nach ist nur eine überschaubare Anpassung des PSA-templates notwendig, da die API-Rückgaben, sofern ich es korrekt überblickt habe, identisch sind. Die Authentifizierung zu den PSA-Servern übernimmt der psa_car_controller.

Welche Informationen benötigt das Template:

1: IP + Port (default 5050) eines laufendes PSA-Car-Controllers.
2: VIN

Folgende API-Abfrage wird benötigt:
http://IP:PORT/get_vehicleinfo/VIN?from_cache=1

wie oft eine Abfrage an die PSA-Server erfolgt, regelt der Controller, daher "from_cache=1"; Anbei die entsprechende API-Rückgabe (Auto angeschlossen an der WB, ohne Ladevorgang)

[psa_controller.json.txt](https://github.com/evcc-io/evcc/files/10295077/psa_controller.json.txt)

andig, 2022-12-23:
Dann müsste immer noch der Status übersetzt werden und dafür brauchts wieder eine eigene Implementierung. Daher leider wontfix.

closed 2022-12-23 by andig

mdkeil, 2022-12-23:
Meiner Ansicht nach stimmt die "Status-Übersetzung" mit der vorhandenen PSA-Implementierung überein.