evcc-io/evcc #5560: SunSpec Soc no such point

premultiply, 2022-12-23:
### Discussed in https://github.com/evcc-io/evcc/discussions/5558

<div type='discussions-op-text'>

<sup>Originally posted by **StevieC121176** December 23, 2022</sup>
seit der neuesten nightly bleibt der SOC meines Hausakkus bei 0%

an der evcc.yaml hab ich nichts geändert.

log gibt folgendes aus:

```
pi@raspberrypi:~ $ evcc meter
[main  ] INFO 2022/12/23 16:57:46 evcc 0.110.1 (e61ec4fe)
[main  ] INFO 2022/12/23 16:57:46 using config file: /etc/evcc.yaml
[db    ] INFO 2022/12/23 16:57:46 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2022/12/23 16:57:46 connecting evcc-774030689 at tcp://192.168.178.77:1883
grid3
-----
Power:          560W
Energy:         11403.1kWh
Current L1..L3: 0.88A 2.48A 0.58A

pv1
---
Power: 0W

battery2
--------
Power: 37W
.Soc:  no such point

pi@raspberrypi:~ $ sudo journalctl -fau evcc
-- Journal begins at Wed 2022-09-07 00:00:06 CEST. --
Dec 23 16:58:51 raspberrypi evcc[203063]: [site  ] ERROR 2022/12/23 16:58:51 battery 1 soc: no such point
Dec 23 16:59:02 raspberrypi evcc[203063]: [site  ] ERROR 2022/12/23 16:59:02 battery 1 soc: no such point
Dec 23 16:59:12 raspberrypi evcc[203063]: [site  ] ERROR 2022/12/23 16:59:12 battery 1 soc: no such point

```
[image]
</div>

premultiply, 2022-12-23:
Vermutlich Problem durch #5549

andig, 2022-12-24:
Um welches Gerät gehts denn hier?

andig, 2022-12-24:
Fehler gefunden. Diese Templates müssen nochmal mit dem PR abgeglichen werden:

```
5 results - 5 files

evcc • templates/definition/meter/kostal-plenticore.yaml:
  43    power: 802:W # sunspec model 802 battery
  44:   soc: 802:Soc # sunspec model 802 battery
  45    {{- end }}

evcc • templates/definition/meter/sma-datamanager.yaml:
  80      register:
  81:       address: 30845 # Battery Soc, %
  82        type: holding

evcc • templates/definition/meter/solarmax-maxstorage.yaml:
  44      register: # manual non-sunspec register configuration
  45:       address: 122 # Batterie Soc
  46        type: input

evcc • templates/definition/meter/solarwatt-myreserve-matrix.yaml:
  28      uri: http://{{ .host }}:{{ .port }}/
  29:     jq: .SData.Soc
  30    {{- end }}

evcc • templates/definition/meter/victron-energy.yaml:
  107      register:
  108:       address: 843 # Soc
  109        type: input
```

andig, 2022-12-24:
Fixed in https://github.com/evcc-io/evcc/commit/d936492fd5966262ae63a604c961152be184dc47

closed 2022-12-24 by andig