evcc-io/evcc #5573: Crash beim Zielladen, wenn Tesla schläft

Hofyyy, 2022-12-26:
### Describe the bug

EVCC crash during testing planning, in WebUI

### Steps to reproduce

1. Nightly with newest planner
2. Choose Tesla with 0 / Unknown SoC because its sleeping
3. Set planner for 8 o'clock in the next morning
--> I assumed it would plan 62 KWH, but it seems to be an endless loop, because I did not see logging.
4. Then I switched to guest vehicle. and then it crashed

### Configuration details

```yaml
Tibber Plan
Tesla Vehicle
```

### Log details

```text
[lp-1  ] DEBUG 2022/12/26 08:58:35 charger status: B
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x28 pc=0x1459f20]
goroutine 124 [running]:
github.com/evcc-io/evcc/core/soc.(*Estimator).RemainingChargeEnergy(...)
	/build/core/soc/estimator.go:67
github.com/evcc-io/evcc/core/soc.(*Estimator).RemainingChargeDuration(0x0, 0x64, 0x40caf40000000000)
	/build/core/soc/estimator.go:58 +0x28
github.com/evcc-io/evcc/core.(*Loadpoint).plannerActive(0x456a000)
	/build/core/loadpoint.go:767 +0x170
github.com/evcc-io/evcc/core.(*Loadpoint).Update(0x456a000, 0x4097740000000000, 0x0)
	/build/core/loadpoint.go:1822 +0x90c
github.com/evcc-io/evcc/core.(*Site).update(0x48324b0, {0x1fb1958, 0x456a000})
	/build/core/site.go:521 +0x140
github.com/evcc-io/evcc/core.(*Site).Run(0x48324b0, 0x48c0ac0, 0x2540be400)
	/build/core/site.go:608 +0x218
github.com/evcc-io/evcc/cmd.runRoot.func3()
	/build/cmd/root.go:263 +0x3c
created by github.com/evcc-io/evcc/cmd.runRoot
	/build/cmd/root.go:262 +0x153c
```

### What type of operating system are you running?

Linux

### Version

nightly

Hofyyy, 2022-12-26:
Ich kann es reproduzieren:
- Ein Auto anschliessen und warten bis es schläft, also nicht mehr antwortet.
- Dann EVCC neustarte und das Auto auswählen (minsoc 0%, damit er nicht anfängt zu laden)
- Dann den Ladeplaner aktivieren, das sorgt glaube ich für eine Endlosschleife.
- Dann auf das Gastfahrzeiug wechseln. Boom

Hofyyy, 2022-12-26:
Ok Grund gefunden.
- Zielladen auf ein korrektes Fahrzeug einstellen, also auch den Tesla der nicht schläft.
- Dann auf das Gastfahrzeug wechsel und Crash.

- Wenn man vorher das Zielladen wieder deaktiviert crashed nichts.

Ich werde mal testen wie es sich nach 13 Uhr verhält, wenn er die Kosten für nächsten Tag schon hat.

andig, 2022-12-26:
Closed in https://github.com/evcc-io/evcc/commit/36be9668a8a0e087e50a4d2e4c8e6b50ee6283a7

closed 2022-12-26 by andig