evcc-io/evcc #5581: 1p Alfen charger failed to charge due to: switch phases: modbus: exception '3' (illegal data value), function '6'

ikruglov, 2022-12-27:
### Describe the bug

I have a single-phase Alfen Eve Single Pro charger. I have no PV at this moment.
When I tried to connect it to evcc and start charging it gave me the following error:
```
switch phases: modbus: exception '3' (illegal data value), function '6'
```

After some debugging and tracing, I discovered that the problem is in [scalePhases](https://github.com/evcc-io/evcc/issues/new?assignees=&labels=&template=bug_report.yaml) function. The `lp.GetPhases()` returns 0 despite `phases: 1` in the config. Hence the code tries to scale phases which Alfen rejects.

P.S. I think I know where the problem is. I'm going to prepare a PR for a fix.

### Steps to reproduce

1. Alfen Eve Single Pro-line with 1p connection
2. Install and configure evcc. Set `phases: 1` in `loadpoint`
3. Start evcc
4. Switch to "Fast" sharing in the UI.
5. Observe the error

### Configuration details

```yaml
network:
  schema: http
  host: ...
  port: 80

interval: 10s # control cycle interval

log: info

meters:
  - name: vattenfall
    type: template
    template: dsmr
    usage: grid
    host: localhost
    port: 3300

chargers:
  - name: alfen
    type: template
    template: alfen
    id: 1
    modbus: tcpip
    host: x.x.x.x
    port: 502 # Port

site:
  title: Home # display name for UI
  meters:
    grid: vattenfall # grid meter

loadpoints:
  - title: Parking # display name for UI
    charger: alfen
    mode: "off"
    resetOnDisconnect: true # set defaults when vehicle disconnects
    phases: 1
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (default 5m)
    minCurrent: 6
    maxCurrent: 16

# the rest of config should not matter
```

### Log details

```text
Dec 27 13:48:16 evcc evcc[29306]: [main  ] INFO 2022/12/27 13:48:16 evcc 0.110.1 (23553f7e)
Dec 27 13:48:16 evcc evcc[29306]: [main  ] INFO 2022/12/27 13:48:16 using config file: ....../evcc.yaml
Dec 27 13:48:16 evcc evcc[29306]: [main  ] INFO 2022/12/27 13:48:16 starting ui and api at :80
Dec 27 13:48:17 evcc evcc[29306]: [db    ] INFO 2022/12/27 13:48:17 using sqlite database: ...../evcc.sqlite.db
Dec 27 13:48:17 evcc evcc[29306]: [mqtt  ] INFO 2022/12/27 13:48:17 connecting evcc-1273340474 at tls://x.x.x.x:1883
Dec 27 13:48:20 evcc evcc[29306]: [lp-1  ] WARN 2022/12/27 13:48:19 locking phase config to 1p for switchable charger
Dec 27 13:48:20 evcc evcc[29306]: [site  ] INFO 2022/12/27 13:48:20 site config:
Dec 27 13:48:20 evcc evcc[29306]: [site  ] INFO 2022/12/27 13:48:20   meters:      grid ✓ pv ✗ battery ✗
Dec 27 13:48:20 evcc evcc[29306]: [site  ] INFO 2022/12/27 13:48:20     grid:      power ✓ energy ✗ currents ✓
Dec 27 13:48:20 evcc evcc[29306]: [lp-1  ] INFO 2022/12/27 13:48:20 loadpoint 1:
Dec 27 13:48:20 evcc evcc[29306]: [lp-1  ] INFO 2022/12/27 13:48:20   mode:        off
Dec 27 13:48:20 evcc evcc[29306]: [lp-1  ] INFO 2022/12/27 13:48:20   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
Dec 27 13:48:20 evcc evcc[29306]: [lp-1  ] INFO 2022/12/27 13:48:20   meters:      charge ✓
Dec 27 13:48:20 evcc evcc[29306]: [lp-1  ] INFO 2022/12/27 13:48:20     charge:    power ✓ energy ✓ currents ✓
Dec 27 13:48:21 evcc evcc[29306]: [lp-1  ] INFO 2022/12/27 13:48:21 car connected
Dec 27 13:48:21 evcc evcc[29306]: [lp-1  ] INFO 2022/12/27 13:48:21 start charging ->
Dec 27 13:48:22 evcc evcc[29306]: [lp-1  ] ERROR 2022/12/27 13:48:22 switch phases: modbus: exception '3' (illegal data value), function '6'
Dec 27 13:48:31 evcc evcc[29306]: [lp-1  ] INFO 2022/12/27 13:48:31 stop charging <-
Dec 27 13:48:32 evcc evcc[29306]: [lp-1  ] ERROR 2022/12/27 13:48:32 switch phases: modbus: exception '3' (illegal data value), function '6'
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.110.1 (23553f7e)

ikruglov, 2022-12-27:
IMO the fundamental issues is that evcc assumes that all Alfens can scale phases. Whereas 1 phase Alfen (apparently?) cannot do so.

To resolve the problem for myself, I tried two solutions which both worked. But I'm not sure which one is more correct.
1. setting `lp.phases` explicitly`. Implemented in this PR: https://github.com/evcc-io/evcc/pull/5582
2. Completely commenting out `Phases1p3p` logic core in `alfen.go`. 

As I said, both of them worked. I'm not sure if the suggested PR for (1) is the right general solution. If so, happy to merge. 
An alternative would be to somehow autodetect or just manually create a param in Alfen spec which would disable attempts to scale phases. 

In any way, I'm happy to contribute a solution with your guidance.

andig, 2022-12-27:
@ikruglov What we need to figure out is how the different Alfen models differentiate themselves. Is you Alfen 1p simply because its only connected 1p or because it is a specific model?

ikruglov, 2022-12-27:
@andig it's phisically 1p model. In other words, there is no L2 and L3 circuits on the board.

andig, 2022-12-27:
> @andig it's phisically 1p model. In other words, there is no L2 and L3 circuits on the board.

@ikruglov All "Alfen Eve Single Pro" I can find are 11kw, hence 3p. Could you please provide a link to the specific 1p model?

@premultiply could we solve this as part of https://github.com/evcc-io/evcc/issues/5369?

ikruglov, 2022-12-27:
I have NG910-60003

[image]

Taken from: https://alfenelkamo.fi/sites/elkamo.fi/files/downloads/Manual-EveSingle-English.pdf

premultiply, 2022-12-27:
Ja, könnte man machen. Das wäre dann in #5539.
Ich muss mich da allerdings auch immer wieder reindenken da es so viele unterschiedliche Fälle gibt und ja auch möglichst nichts kaputt gehen soll...

andig, 2022-12-27:
I see. That's funny:

[image]

@ikruglov any idea if 1p is detectible? If not we'll need to make it a config option :(

premultiply, 2022-12-27:
Maybe it is detectible from voltage metering data too.
Let's check.

ikruglov, 2022-12-27:
@andig, ufff.... I can experiment but not sure where to start looking at. I guess something OCPP-related? Or modbus registers? We can also just hardcode models, it's not a big list of them.

andig, 2022-12-27:
[image]

Could you maybe check those registers? I'd expect the 1p chargers to have 0 in the last two and error trying to set them to anything else?

andig, 2022-12-27:
> We can also just hardcode models, it's not a big list of them.

If you can find a model register?

ikruglov, 2022-12-27:
okay. Thanks for the guidance. Let me see.

While I'm looking. How would you handle the fact that `api.PhaseSwitcher` interface is part of Alfen object? Hence, we cannot "undo it" at runtime so the code in [scalePhases](https://github.com/evcc-io/evcc/issues/new?assignees=&labels=&template=bug_report.yaml) would work correctly? Or would you rather return a error (ErrNotSupported) from `Phases1p3p` method?

andig, 2022-12-27:
I've fixed this in https://github.com/evcc-io/evcc/commit/70454222b01e9375463cfc75df32385ffcef1a6f by adding a 1p template. Forgot to push to branch so there it is. If we can autodetect I'd revert this.

> While I'm looking. How would you handle the fact that api.PhaseSwitcher interface is part of Alfen object? Hence, we cannot "undo it" at runtime so the code in [...]

We need to "decorate" the ability when creating the object.

andig, 2022-12-27:
> Maybe it is detectible from voltage metering data too.
> Let's check.

I think you're right. We should not decorate if L3 doesn't have current.

@ikruglov could you check the voltages? Will L2/L3 even produce a reading or just error?

[image]

ikruglov, 2022-12-27:
Here are my findings. In short:
1. I was only able to read voltages
2. Voltage indeed can be used for 1p vs 3p detection. However, I'm not sure about a case where a 3p charger connected using 1p. My naive thinking tells me that it should work as there is no point to scale to 3p in this case.

```
!!!!!!!! Name modbus: exception '2' (illegal data address), function '3'
!!!!!!!! Manufacturer modbus: exception '2' (illegal data address), function '3'
!!!!!!!! Firmware version  modbus: exception '2' (illegal data address), function '3'
!!!!!!!! Platform type  modbus: exception '2' (illegal data address), function '3'
!!!!!!!! Station serial number modbus: exception '2' (illegal data address), function '3'
!!!!!!!! SCN Max Current per Phase L1 modbus: exception '2' (illegal data address), function '3'
!!!!!!!! SCN Max Current per Phase L2 modbus: exception '2' (illegal data address), function '3'
!!!!!!!! SCN Max Current per Phase L3 modbus: exception '2' (illegal data address), function '3'
!!!!!!!! Voltage Phase V(L1-N) 227.1999969482422
!!!!!!!! Voltage Phase V(L2-N) NaN
!!!!!!!! Voltage Phase V(L3-N) NaN
```

My code:
```
       type item struct {
               name  string
               start uint16
               ln    uint16
               t     string
       }

       items := []item{
               item{"Name", 100, 17, "string"},
               item{"Manufacturer", 117, 5, "string"},
               item{"Firmware version ", 123, 17, "string"},
               item{"Platform type ", 140, 17, "string"},
               item{"Station serial number", 157, 11, "string"},

               item{"SCN Max Current per Phase L1", 1417, 2, "float32"},
               item{"SCN Max Current per Phase L2", 1419, 2, "float32"},
               item{"SCN Max Current per Phase L3", 1421, 2, "float32"},

               item{"Voltage Phase V(L1-N)", 306, 2, "float32"},
               item{"Voltage Phase V(L2-N)", 308, 2, "float32"},
               item{"Voltage Phase V(L3-N)", 310, 2, "float32"},
       }

       for _, i := range items {
               if b, err := wb.conn.ReadHoldingRegisters(i.start, i.ln); err == nil {
                       if i.t == "string" {
                               wb.log.INFO.Println("!!!!!!!!:", i.name, string(b))
                       } else if i.t == "float32" {
                               wb.log.INFO.Println("!!!!!!!!", i.name, rs485.RTUIeee754ToFloat64(b))
                       }
               } else {
                       wb.log.INFO.Println("!!!!!!!!", i.name, err)
               }
```

andig, 2022-12-27:
Ahh, that's nice! I'll provide follow-up PR.

ikruglov, 2022-12-27:
@andig, would you mind letting me prepare the PR. I'm looking into bigger contribution so want to get myself familiar.

ikruglov, 2022-12-27:
@andig, I've implemented https://github.com/evcc-io/evcc/pull/5587/files but I see that you actually merged your Alfen 1p template. I don't understand your logic here. I though that we aim for auto-detection without any new templating.

closed 2022-12-27 by andig

J1Buskens, 2023-02-11:
Hi
I have a 3 phase Alfen charger and I'm also getting `[lp-1  ] ERROR 2023/02/11 10:53:22 switch phases: modbus: exception '3' (illegal data value), function '6'`. is this still supposed to work with the 3 phase version?

andig, 2023-02-13:
@J1Buskens could you please run

    evcc charger --log trace

and show the output? This should definitely still work.

ikruglov, 2023-02-13:
> @ikruglov seems we have merged your change after all :).
@andig you merged 'Alfen 1p template' and I later reverted it by your instructions.

On this matter. I suspect that 3 phase Alfen may have settings allowing it only 1p charging. I recall there is a setting. Can't check right now unfortunately.

J1Buskens, 2023-03-06:
Hi sorry for the late response. In the Alfen software I enabled “Allow 1- and 3-phase charging” and this fixed the problem.