evcc-io/evcc #5584: Discovergy: Negate power for pv usage (production meter)

premultiply, 2022-12-27:
Die Eingabe von

```
  name: pv2
  scale: -1
```
ergibt leider

```
Dez 27 14:27:19 raspberrypi evcc[7261]: [main  ] INFO 2022/12/27 14:27:19 evcc 0.110.1 (48f22d5a)
Dez 27 14:27:19 raspberrypi evcc[7261]: [main  ] INFO 2022/12/27 14:27:19 using config file: /etc/evcc.yaml
Dez 27 14:27:19 raspberrypi evcc[7261]: [main  ] INFO 2022/12/27 14:27:19 starting ui and api at :7070
Dez 27 14:27:19 raspberrypi evcc[7261]: [db    ] INFO 2022/12/27 14:27:19 using sqlite database: /root/.evcc/evcc.db
Dez 27 14:27:19 raspberrypi evcc[7261]: [main  ] FATAL 2022/12/27 14:27:19 cannot create meter 'pv2': cannot create meter 'template': invalid key: scale
Dez 27 14:27:19 raspberrypi evcc[7261]: [main  ] FATAL 2022/12/27 14:27:19 will attempt restart in: 5m0s
```

_Originally posted by @nosuntoday in https://github.com/evcc-io/evcc/discussions/5577#discussioncomment-4502746_

andig, 2022-12-27:
Aus meiner Sicht nicht, es sei denn wir können das unabhängig bestätigen. Grad bei Discovergy haben wir das schon mehrfach hin-und-her gedreht afair.

premultiply, 2022-12-27:
Separaten PV-Produktionszähler hatten wir mWn noch nicht.
Die gelieferten Daten passen aber zur Erwartungshaltung und dem gängigen MSB-Marktmodell.

andig, 2022-12-27:
Siehe Discussion, wäre ja eine Template-only Änderung. `Energy` wäre dann wohl auch falsch wenns das bräuchte, aber die werten wir bei PV (noch...) eh nicht aus.

andig, 2022-12-28:
Closed in https://github.com/evcc-io/evcc/commit/17bfca781261fe5b3e473171c68e03ef586c7ad2

closed 2022-12-28 by andig