evcc-io/evcc #5591: Planner: Estimator does not consider an efficiency factor

Hofyyy, 2022-12-27:
### Describe the bug

@schenlap 
I hope its ok to put your point here.

I think the estimator was written for UI purposes and is now used in a new context, in which precision get more and more important. So maybe we should add the "Ladeverluste" as a percentage.

In the following test we see 5%-10% if we use a default 11KW wallbox.
https://www.adac.de/rund-ums-fahrzeug/elektromobilitaet/laden/ladeverluste-elektroauto-studie/

Because of the reason, that we have further negativ factors, like batterie heating, slower charging at the end and so on,
I would suggest to use 90% efficiency as a default.

One example without battery heating can be found attached in the logfile.

Maybe some ideas to the architecture?
- Should we put the 10% as a default in the vehicle class, with a getter or interface?
- Then we could make in future an optional parameter, because it depends highly on the car itself.

### Steps to reproduce

1. Use new planner feature and check the estimator
2.
3.
...
[logs_all.txt](https://github.com/evcc-io/evcc/files/10309773/logs_all.txt)

### Configuration details

```yaml
...
```

### Log details

```text
d
```

### What type of operating system are you running?

Linux

### Version

nightly

andig, 2022-12-27:
Der Faktor wird schon im Loadpoint angewandt.

closed 2022-12-27 by andig

Hofyyy, 2022-12-27:
Manchmal bist du auch schnell? Ist es vielleicht gut es hier dann umzuschreiben?
Also Auto benötigt 40% SoC bei 62KWh max. Macht bei 11 KWh -> 2.25 Stunden -> 2:15 +-

Loggingzeile aus dem Log passt genau: total plan duration: "2h13m0s"
Wo auch immer das herkommt. Da sollten 10% drauf, da es Netto ist, oder übersehe ich was?

premultiply, 2022-12-27:
Der Estimator berücksichtigt 10% Ladeverluste als Initialwert und passt sich nachfolgend adaptiv an.

andig, 2022-12-27:
> Der Estimator berücksichtigt 10% Ladeverluste als Initialwert und passt sich nachfolgend adaptiv an.

...und darauf schlägt der Loadpoint nochmal 10% drauf bevor es an den Planner geht: https://github.com/evcc-io/evcc/blob/master/core/loadpoint.go#L769

Hofyyy, 2022-12-27:
Ich glaube euch ja. Dann lasst mich naiv fragen.

Im log sehe ich folgendes und zwar von anfang an.
[lp-1  ] DEBUG 2022/12/27 01:45:24 planning 2h13m0s until 2022-12-27 07:00:00 +0000 UTC

Das passt halt zu 40% von 62KWH. Ich sehe da keine 10%, wie das auch immer berechnet wird.
Mit Estimator, Loadpoint o.ä. Oder wo ist mein Denkfehler.

Ich sage also, lasst uns gucken wo die 2h13 herkommen und das anpassen.

Hofyyy, 2022-12-27:
Und es ist auch nicht nur die Zeile. Die Ladefenster sind auch genau 2:13.
[lp-1  ] TRACE 2022/12/27 01:45:24   slot from: 2022-12-27 03:00:00 +0100 CET to 2022-12-27 04:00:00 +0100 CET cost 0.14
[lp-1  ] TRACE 2022/12/27 01:45:24   slot from: 2022-12-27 02:00:00 +0100 CET to 2022-12-27 03:00:00 +0100 CET cost 0.14
[lp-1  ] TRACE 2022/12/27 01:45:24   slot from: 2022-12-27 01:47:00 +0100 CET to 2022-12-27 02:00:00 +0100 CET cost 0.14

andig, 2022-12-27:
Bau Dir gerne in das `plannerActive` noch Logging rein, dann siehst Du wo der Wert her kommt.

Hofyyy, 2022-12-27:
Ich sehe gerade schon wo du geändert hast. Ohne das jetzt zu debuggen.
>> requiredDuration = time.Duration(float64(requiredDuration) / soc.ChargeEfficiency)
vs.
>> 	requiredDuration = time.Duration(float64(requiredDuration) * soc.ChargeEfficiency)

andig, 2022-12-27:
Wieso geändert? Laut Slack hast Du https://github.com/evcc-io/evcc/commit/48f22d5a. Da ist das genau so. Wenn das jetzt wieder ein Suchspiel wird klinke ich mich aus ;)

Hofyyy, 2022-12-27:
Ruhig Blut.
- Ich debugge das bei Zeiten mal und validiere wo das herkommt. Im Log sieht man schön das der Estimator stark daneben liegt. Es sollte sich rausfinden lassen warum.
- Die Slack version stimmt. Ich hatte gerade nur gesehen das du auf dem Master ein commit in der Region gemacht hast. Hat hiermit nichts zu tun.

Also lass es zu und ich mache es ggf. noch mal auf, wenn ich weiss wo die discrepanz herkommt.

Hofyyy, 2022-12-27:
Ich hab leider gerade nur den BMW zum vergleich, aber damit sieht es im Debugger genau so aus wie ihr sagt. Da muss ich wohl am realen Beispiel debuggen. Mal gucken wo der Unterschied herkommt.