evcc-io/evcc #5594: Planner does not show planned start time

naltatis, 2022-12-28:
### Describe the bug

The old planner published the projected time when charging will start. The UI shows this in the status text of the loadpoint. We should also add this to the new planner as well.

`lp.publish("targetTimeProjectedStart", ...)`

Since the new planner may have multiple charging slots I'd expect that `targetTimeProjectedStart` would always contain the time of the next charging slot.

### Steps to reproduce

Set a charging time

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

andig, 2022-12-28:
Ich frag mich ob das Sinn macht. Ohne Planner war das ja einfach Zielzeit - Ladedauer. Jetzt gäbe es ja immer einen Plan falls eine Zielzeit gesetzt ist. Was wollen wir denn damit visualisieren?

/cc @schenlap

schenlap, 2022-12-28:
Prinzipiell finde ich es gut wenn im UI auch erkennbar ist dass der planner seinen Job macht. Aktuell sieht man keine Reaktion, das finde ich nicht gut.
Entweder den nächsten Start anzeigen oder vielleicht auch den "günstigsten" Start inkl. Preisinfo ( da wüsste man dann auch gleicht wie man den Geschirrsspüler programmieren muss um günstig zu waschen ;-)  )

andig, 2022-12-28:
Oder noch fancier: den Plan plus Tarife anzeigen :). Das würde gleich eine neue Welt der Fehlerbehebung eröffnen (oder wohl eher erstmal der Fehlerfindung :)

naltatis, 2022-12-28:
"Plan plus Tarife anzeigen" arbeite ich dran, das wollen wir auf jeden Fall. Nicht nur fürs Debugging. Das würde ich im Zielladen-Dialog ansiedeln.

Ich würde das vorherige Verhalten "Zeitpunkt des nächster Start" aber trotzdem zusätzlich anzeigen. Das ist ein gutes Feedback für den Nutzer was wenig Platz braucht.

andig, 2022-12-29:
Brauchen wir das als Timestamp oder als Duration until? Durations sind etwas einfacher in der Handhabung?

naltatis, 2022-12-29:
`targetTimeProjectedStart` war ein Datum. Fänd auch gut, wenn dir das so lassen. Dann ist das analog zur Zielzeit.

closed 2022-12-29 by andig