evcc-io/evcc #5595: Home Assistant + go-eCharger + evcc + MG4: Force State Umschaltung nach dem Laden führt zu Ladeblockade

Jamhem2010, 2022-12-28:
### Describe the bug

Hallo zusammen,

ich habe seit Anfang Dezember einen MG4 electric. Dieser wird zu Hause mit dem go-eCharger geladen. Im Hintergrund läuft evcc als Addon im Home Assistant.
Mir ist aufgefallen, dass ich nur ein einziges Mal laden konnte und anschließend bis zum Neustart des go-eChargers nicht wieder. Das Fahrzeug wurde nicht erkannt. Erst ein Neustart hat dafür gesorgt, dass das Fahrzeug wieder erkannt und somit auch geladen wird.
Sobald das evcc Addon deaktiviert ist, funktioniert das Laden direkt und immer (auch ohne Neustart des go-eChargers).
Ich war parallel mit dem Support vom go-eCharger im Austausch und wir konnten das Problem eingrenzen.
Evcc stellt den Parameter "frc" vom go-eCharger um. Dieser hat mehrere Zustände: Charge, Don't charge und Neutral.
Sobald man den Lademodus "Aus" im Evcc einstellt, wird der frc auf Don't Charge gestellt. Um das Fahrzeug zu laden muss nun ja wieder ein anderer Lademodus eingestellt werden. Beim Umschalten wird aber der frc Parameter nicht mit umgestellt. Er bleibt bei Don't Charge und sorgt dafür, dass das Fahrzeug trotz MinPV, PV oder Schnell nicht geladen wird.
Wenn ich per Home Assistant den frc wieder auf Neutral stelle, funktioniert auch das Laden wieder direkt.
Solange ich nun nicht mehr in "Aus" wechsle funktioniert auch das Laden weiterhin.
Könnt ihr das bitte einmal prüfen? Liegt hier ein Bug vor?

Viele Grüße
Jan

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: XXX
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
sponsortoken: XXX

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: sma
    type: template
    template: sma-inverter
    usage: pv
    host: XXX # IP-Adresse oder Hostname
    password: XXX # Passwort für Benutzergruppe Benutzer # Optional 
  - name: zaehler
    type: custom # IR-Lesekopf Tasmota
    power:
      source: http
      uri: XXX
      jq: .StatusSNS.Strom.Leistung_akt

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for documentation see https://docs.evcc.io/docs/devices/chargers
chargers:
  - name: go-echarger
    type: template
    template: go-e-v3
    host: XXX # IP-Adresse oder Hostname 

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: MG4
    type: template
    template: offline
    title: # Wird in der Benutzeroberfläche angezeigt # Optional
    capacity: 51 # Akku-Kapazität in kWh # Optional

# site describes the EVU connection, PV and home battery
site:
  title: Home # display name for UI
  meters:
    grid: zaehler # grid meter
    pvs:
      - sma # list of pv inverters/ meters

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Garage # display name for UI
    charger: go-echarger # charger
    meter:  # charge meter
    mode: "minpv" # set default charge mode, use "off" to disable by default if charger is publicly available
    vehicle: MG4 # set default vehicle (disables vehicle detection)
    resetOnDisconnect: true # set defaults when vehicle disconnects
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
        mode: charging
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    phases: 1 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (default 5m)
    minCurrent: 6 # minimum charge current (default 6A)
    maxCurrent: 16 # maximum charge current (default 16A)

# tariffs are the fixed or variable tariffs
# cheap (tibber/awattar) can be used to define a tariff rate considered cheap enough for charging
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price
    type: fixed
    price: 0.25 # EUR/kWh

    # # or variable via tibber
    # type: tibber
    # cheap: 0.2 # EUR/kWh
    # token: "XX" # access token
    # homeid: "XXc" # optional if multiple homes associated to account

    # # or variable via awattar
    # type: awattar
    # cheap: 0.2 # EUR/kWh
    # region: de # optional, choose at for Austria
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.065 # EUR/kWh

# mqtt message broker
mqtt:
  broker: XX
  topic:  # root topic for publishing, set empty to disable
  clientid: evcc
  user: XX
  password: XX

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# eebus credentials
eebus:
  # uri: # :4712
  # interfaces: # limit eebus to specific network interfaces
  # - en0
  # certificate: # local signed certificate, required, can be generated via `evcc eebus-cert`
  #   public: # public key
  #   private: # private key

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: SoC updated
      msg: Battery charged to ${vehicleSoC:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
```

### Log details

```text
n/a
```

### What type of operating system are you running?

Linux

### Version

0.110.1

premultiply, 2022-12-28:
https://github.com/evcc-io/evcc/blob/53982a9b4f77125062cb2f833727ecea6f382914/charger/go-e.go#L120-L133

premultiply, 2022-12-28:
https://github.com/goecharger/go-eCharger-API-v1/blob/master/go-eCharger%20API%20v1%20DE.md
<html><body>

alw | uint8_t | allow_charging: PWM Signal darf anliegen 0:nein 1:ja
-- | -- | --

</body>
</html>

https://github.com/goecharger/go-eCharger-API-v2/blob/main/apikeys-de.md
<html><body>

frc | R/W | uint8 | Config | forceState (Neutral=0, Off=1, On=2)
-- | -- | -- | -- | --

</body>
</html>

iStehle, 2023-01-12:
Ich muss meine Logs und so mal vorbereiten, aber Zielladen fängt bei mir einfach nicht an. Wenn ich den Go-e v3 Neustarten geht es dann los. Ggf auch was mit dem Auto, aber ich kann leider vom neuen tollen zielladen profitieren

Jamhem2010, 2023-01-12:
Nutzt du home assistant, Io broker oder ähnliches? Dann könntest du mal probieren, den frc Parameter manuell auf Charge zu stellen, wenn das Laden nicht startet.
Beim mir klappt es dann auch ohne Neustart des go e

premultiply, 2023-01-12:
Ohne gescheite Logs wird das hier nichts.

iStehle, 2023-01-13:
Diese Nacht wurde wieder nicht geladen:
Evcc sagt nur „188W
[lp-1  ] WARN 2023/01/13 06:25:22 charger out of sync: expected enabled, got disabled“
In den Debug Meldungen stand nichts weiteres drin.

iStehle, 2023-01-14:
Ich hab jetzt wirklich das Verhalten das ich nur noch Strom in meinen iD4 bekomme wenn evcc aus ist. Mit dem nightly zum Testen von Zielladen ging noch alles. Bei GoE hatte ich die 054.11 und die 055.0 installiert. 
Das Log ist nicht weiter aussagekräftig. Auch bei schnell laden passiert nichts. Sondern nur „Zielladen warte auf Fahrzeug“

premultiply, 2023-01-14:
Dir ist aber bekannt dass die VW-MEB-Baukästen nach ein paar Ladeunterbrechungen generell nicht mehr aufwachen?

Ich könnte den frx-Change ins Nightly schieben wenn du das testen magst.
Gegen die VW-Ladesperre würde es aber auch nicht helfen.

iStehle, 2023-01-14:
Könnte ich gerne machen. Bis jetzt hatte ich nie Probleme damit das das Auto anfängt zu laden. Ohne evcc geht der eco Modus (laden nach Strompreisen) des goe ohne Probleme. Bis vor kurzem hatte ich das Zielladen per Automatisierung aus HomeAssistant -> mqtt Evcc selbst gemacht und das hat immer geladen. Ich finde selbst noch nicht was sich da jetzt verändert hab. Ich muss mir zuhause bald mal einen logstash zulegen um das SmartHome zu debuggen

PR #5597 (2023-01-16): go-e: Fix frc operator
May fix #5595
/cc @Jamhem2010
closed 2023-01-16 by premultiply