evcc-io/evcc #5638: DSMR busy loop

ikruglov, 2022-12-31:
### Describe the bug

Hi folks. 

DSMR meter has a problem.
For context: DSMR meter reads from a TCP socket provided by (usually) `ser2net`.
The problem appears if `ser2net` server goes away. In this case, the code starts busy-looping with an EOF error [around Peek() function](https://github.com/evcc-io/evcc/blob/master/meter/dsmr.go#L166). Evcc will not recover from this state since the code needs to reconnect to `ser2net` which it doesn't do.

### Steps to reproduce

1. have setup with DSMR meter via `ser2net`
2. stop/restart `ser2net` while evcc is running

### Configuration details

```yaml
...
meters:
  - name: vattenfall
    type: template
    template: dsmr
    usage: grid
    host: x.x.x.x
    port: x
...
```
```

### Log details

```text
[main  ] INFO 2022/12/31 14:56:38 evcc 0.0.1-alpha
[main  ] INFO 2022/12/31 14:56:38 using config file: ./evcc.yaml
[main  ] INFO 2022/12/31 14:56:38 starting ui and api at :7070
[db    ] INFO 2022/12/31 14:56:38 using sqlite database: /Users/ikruglov/.evcc/evcc.db
[alfen ] DEBUG 2022/12/31 14:56:42 detected 1p alfen
[site  ] INFO 2022/12/31 14:56:42 site config:
[site  ] INFO 2022/12/31 14:56:42   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2022/12/31 14:56:42     grid:      power ✓ energy ✗ currents ✓
[lp-1  ] INFO 2022/12/31 14:56:42 loadpoint 1:
[lp-1  ] INFO 2022/12/31 14:56:42   mode:        now
[lp-1  ] INFO 2022/12/31 14:56:42   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2022/12/31 14:56:42   meters:      charge ✓
[lp-1  ] INFO 2022/12/31 14:56:42     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2022/12/31 14:56:42 phase timer inactive
[lp-1  ] DEBUG 2022/12/31 14:56:42 pv timer inactive
[lp-1  ] DEBUG 2022/12/31 14:56:42 max charge current: 6A
[site  ] DEBUG 2022/12/31 14:56:42 ----
[lp-1  ] DEBUG 2022/12/31 14:56:42 charge power: 2129W
[site  ] DEBUG 2022/12/31 14:56:42 grid power: 2571W
[site  ] DEBUG 2022/12/31 14:56:42 grid currents: [11 0 0]A
[site  ] DEBUG 2022/12/31 14:56:42 pv power: 0W
[site  ] DEBUG 2022/12/31 14:56:42 site power: 2571W
[lp-1  ] DEBUG 2022/12/31 14:56:42 charge voltages: [228 0 0]V
[lp-1  ] DEBUG 2022/12/31 14:56:42 detected phases: 1p
[lp-1  ] DEBUG 2022/12/31 14:56:42 charge currents: [9.38 0 0]A
[main  ] INFO 2022/12/31 14:56:43 new version available: 0.110.1
[lp-1  ] DEBUG 2022/12/31 14:56:43 charger status: C
[lp-1  ] DEBUG 2022/12/31 14:56:43 charge start energy: 1.01e+04kWh
[lp-1  ] INFO 2022/12/31 14:56:43 car connected
[lp-1  ] DEBUG 2022/12/31 14:56:43 pv timer elapse
[lp-1  ] DEBUG 2022/12/31 14:56:43 pv timer inactive
[lp-1  ] DEBUG 2022/12/31 14:56:43 charge start energy: 1.01e+04kWh
[lp-1  ] INFO 2022/12/31 14:56:43 start charging ->
.... (here I inserted 1s pause to not kill my raspberry pi)
[dsmr  ] ERROR 2022/12/31 14:57:35 peek: EOF
[dsmr  ] ERROR 2022/12/31 14:57:36 peek: EOF
[dsmr  ] ERROR 2022/12/31 14:57:37 peek: EOF
[dsmr  ] ERROR 2022/12/31 14:57:38 peek: EOF
[dsmr  ] ERROR 2022/12/31 14:57:39 peek: EOF
[dsmr  ] ERROR 2022/12/31 14:57:40 peek: EOF
[dsmr  ] ERROR 2022/12/31 14:57:41 peek: EOF
[dsmr  ] ERROR 2022/12/31 14:57:42 peek: EOF
```
```

### What type of operating system are you running?

Linux

ikruglov, 2022-12-31:
@andig what would be a good way to fix it? DSMR code clearly needs a reconnection logic. However, I'm not sure if the reconnection should be part of `dsmr.go` itself or if we can just use some outer restart/redo loop?

andig, 2022-12-31:
Set `conn=nil` on EOF? You'd still want to add a backoff...

ikruglov, 2022-12-31:
My main question was about should the reconnection logic be local to DSMR package or there is an external "reconnect/restart" loop. Based on your answer I assume that the reconnection logic should be local to DSMR's code.

ikruglov, 2022-12-31:
Actually, I see what you mean. I've pushed a PR with a simple fix. Need to add a backoff though.

ikruglov, 2022-12-31:
pushed a fix with backoff

closed 2023-01-01 by andig