evcc-io/evcc #5647: Fahrzeug Enyaq wird seit heute nicht mehr erkannt

ammann-ruchti, 2023-01-01:
Nach einigen Stunden rumsuchen vermute ich die Ursache gefunden zu haben. Nachdem mir heute Vormittag EVCC ein ganz komisches Ladeverhalten angezeigt hat (11 kW Laden bei 4 kW Gesamtverbrauch) habe ich EVCC neu gestartet. Danach wurde das Fahrzeug nicht mehr erkannt. Nach langem Suchen und einem Trace-Log des Fahrzeugs faind ich den Fehler: "Invalid identity token" beim Versuch, auf vwgroup.io zu verbinden.
Haben das andere auch erfahren? Was kann man da machen?
Zugriff über die Skoda-App funktioniert.

StefanReppien, 2023-01-01:
Ich habe exakt das selbe Problem seit heute. Rollback auf die vorh. Version 1

Caibuk, 2023-01-02:
Spontane Selbstheilung? Bei mir funktioniert die Abfrage von Enyaq und e-up wieder. Auch in home assistant (hatte heute Morgen auch http-400-Fehler.

andig, 2023-01-02:
VW hat festgestellt, dass der TokenRefreshService wohl doch wichtig ist und ihn repariert :)

closed 2023-01-02 by andig

Farfar, 2023-01-02:
> VW har bestämt att TokenRefreshService trots allt är viktig och fixar det :)

It was probably a planned change that they had to revert. I suggest updating the code to address this issue before VAG decides to implement this change again.
For Skoda I've found that the newer versions of the app use api.connect.-skoda-auto.cz for all token handling (except for the special "MBB" token). See 1.3.0 version of skodaconnect library for URLs etc.
Other brands probably need their specific handling as well.