evcc-io/evcc #5648: Fehler beim Starten

spizzl510, 2023-01-02:
### Describe the bug

cannot create charger 'wallbox'
cannot create charger 'template'
cannot create charger 'easee'
unexpected status: 400

### Steps to reproduce

1. Heute habe ich das Update installiert und danach konnte der Server nicht mehr gestartet werden
2.
3.
...

### Configuration details

```yaml
GNU nano 5.4                                          /etc/evcc.yaml                                                   
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: 

meters:
- type: template
  template: solaredge-inverter
  id: 1
  host: 
  port: 1502
  usage: grid
  modbus: tcpip
  timeout: 10s
  name: grid1
- type: template
  template: solaredge-inverter
  id: 1
  host: 
  port: 1502
  usage: pv
  modbus: tcpip
  timeout: 10s
  name: pv2
- type: template
  template: solaredge-inverter
  id: 3
  host: 
  port: 1502
  usage: pv
  modbus: tcpip
  timeout: 10s
  name: pv3

chargers:
- type: template
  template: easee
  user: 
  password: 
  charger: 
  name: wallbox
```

### Log details

```text
[main  ] INFO 2023/01/02 06:55:36 evcc 0.110.1
[main  ] INFO 2023/01/02 06:55:36 using config file: /etc/evcc.yaml
[main  ] INFO 2023/01/02 06:55:36 starting ui and api at :7070
[db    ] INFO 2023/01/02 06:55:36 using sqlite database: /home/pi/.evcc/evcc.db
[main  ] FATAL 2023/01/02 06:55:40 cannot create charger 'wallbox': cannot create charger 'template': cannot create charger 'easee': unexpected status: 400
[main  ] FATAL 2023/01/02 06:55:40 will attempt restart in: 5m0s
[server] FATAL 2023/01/02 06:55:40 unlinkat /tmp/evcc: operation not permitted
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.110.1

andig, 2023-01-02:
Siehe https://github.com/evcc-io/evcc/discussions/4413

closed 2023-01-02 by andig

spizzl510, 2023-01-02:
Problem gefunden, war etwas anderes. Hätte ich mal gleich etwas besser nachgedacht.
Trotzdem Danke :)

Otle, 2023-01-05:
Hallo, habe das gleiche Problem, was war bei dir die Lösung, wo kann ich besser nachdenken? Wäre für einen kurzen Hinweis dankbar

spizzl510, 2023-01-05:
> Hallo, habe das gleiche Problem, was war bei dir die Lösung, wo kann ich besser nachdenken? Wäre für einen kurzen Hinweis dankbar

Ich hatte meine Passwort in der Easee Cloud/Portal geändert. Beim Update auf die neuste Version von EVCC und den Server Neustart konnte er sich dann wohl nicht mehr einloggen. Nachdem ich das Passwort in der Konfiguration geändert habe, hat wieder alles problemlos funktioniert.

Otle, 2023-01-05:
Hallo vielen Dank für die schnelle Antwort, leider ist mein Problem, denke ich ein anderes, trotzdem vielen Dank