evcc-io/evcc #5656: Erweiterung der API um erkanntes Fahrzeug und Gastfahrzeug, sowie targetEnergy an loadpoints

binderth, 2023-01-03:
**Is your feature request related to a problem? Please describe.**
Ich möchte in meinem smarthome aufgrund des erkannten angeschlossenen Fahrzeugs weitere Aktionen starten. Aktuell zeigt die API nur den "Namen" des Fahrzeugs unter loadpoints an - und auch nur, wenn es nicht das Gastfahrzeug ist.
Außerdem sollte `targetEnergy` auch bei Gastfahrzeugen und nicht gesetztem Wert übertragen werden (analog erkannten Fahrzeugen)

**Describe the solution you'd like**
In der /api/state Antwort sollten zusätzlich folgende Attribute übertragen werden:
`loadpoints.X.vehicleName: VEHICLES.NAME`  
wobei VEHICLES.NAME der Wert aus der evcc.yaml ist, könnte dann "guest" oder ähnliches für das Gastfahrzeug sein.

`loadpoints.X.targetEnergy: 0`  
wobei bei nicht gesetzter targetEnergy ebenfalls `0` übertragen werden sollte (was bei erkannten Fahrzeugen bereits geschieht)

**Additional context**
Ich möchte bei Gastfahrzeugen und erkannten Fahrzeugen dieselben Informationen aus der API erhalten. Daher erstmal keine Alternativen. ;)

andig, 2023-01-03:
> Ich möchte bei Gastfahrzeugen und erkannten Fahrzeugen dieselben Informationen aus der API erhalten. Daher erstmal keine Alternativen. ;)

Bin nicht sicher, ob ichs verstehe: du möchtest eine Fahrzeugliste? Die aktuelle Zielenergie ist wie der aktuelle Zielsoc nur am Ladepunkt, aber nicht je Fahrzeug verfügbar. Da sollte sie allerdings schon sichtbar sein?

andig, 2023-01-03:
> Daher der Wunsch nach loadpoints.X.vehicleName, der dann eben mit dem Namen aus der yaml oder mit "guest" belegt wird.

Alternativ: wenn kein Name, dann Gast -> keine Änderung notwendig.

> wenn man nach loadpoint.X.targetEnergy frägt, das Attribut aber nich immer mitkommt. Sofern es nicht gesetzt wird, dann eben mit dem Standard-Wert vorbelegen oder NULL. 

Das kommt einfach generell nur wenn es gesetzt ist. Immer alle Attribute zu pushen, auch als nil, wäre ne ziemlich große Änderung ich ich im Moment ungern vornehmen würde.

> dafür muss ich dann aber wissen: welche ID (0,1,2,..) das konfigurierte Fahrzeug hat

Du vermischst doch Offline- und Gastfahrzeug? Gastfahrzeug ist genau das- wir wissen es nicht. Da kann auch keine ID angezeigt werden. Die Offlinefahrzeuge sollte alle im API verfügbar sein.

binderth, 2023-01-03:
> Das kommt einfach generell nur wenn es gesetzt ist. Immer alle Attribute zu pushen, auch als nil, wäre ne ziemlich große Änderung ich ich im Moment ungern vornehmen würde.

ok. das muss man dann im Ziel abfangen. Geht auch.

> Du vermischst doch Offline- und Gastfahrzeug? Gastfahrzeug ist genau das- wir wissen es nicht. Da kann auch keine ID angezeigt werden. Die Offlinefahrzeuge sollte alle im API verfügbar sein.

Jein. Durch diverse andere Mechanismen kann ich ein außerhalb evcc erkennen, ob wirklich ein Gast lädt - oder ein Fahrzeug, das ich offline angelegt habe, weil es seine Daten nicht hergibt. Daher würde ich gerne von außen per API das erkannte Offline-Fahrzeug wählen. Ab und zu aber ist es so, dass die Web-API eines anderen Fahrzeugs bissle hinterherhinkt - aber dann ist ja die automatische Erkennung schon wieder aus - also kann ich dann das Online-Fahrzeug später als das aktive setzen, weil z.B. die KIA API ab und zu hängt.

andig, 2023-01-03:
Es tut mir wirklich leid, aber ich verstehe nur Bahnhof. Alle Fahrezeuge einschl. offline sind übers API wählbar und werden bei Erkennung auch angezeigt. Das Andere ist dann eben Guest. Ich verstehe die Lücke nicht.

binderth, 2023-01-03:
hmm. ok. Ich skizziere mal meinen Use Case, vielleicht bin ich einfach zu kompliziert! ;)

Vorausgesetzt:
1. ein Kia EV6 inkl. API-Zugriff (der aber mitunter paar Minuten hinterher hinkt)
2. ein Fiat 500e ohne jegliche API (der Action als absolute Basis-Einsteiger-Version hat gar nix)
3. eine Wallbox
4. evcc 😀

Ich stecke den Fiat an:
1. die Wallbox meldet: "Auto angesteckt"
2. da es der Fiat ist, wird kein Fahrzeug erkannt, evcc stellt auf "Gastfahrzeug"
3. mein openHAB weiß aber, es ist der FIAT, weil der EV6 auf Status "A" steht
4. mein openHAB möchte nun evcc mitteilen, dass der FIAT lädt

Problem:
in der /api/state Response ist `result.vehicles.0` nicht immer dasselbe Fahrzeug. Ein 1:1 Verweis auf vehicleName wie er in der evcc.yaml gepflegt ist, ist nicht möglich, man muss über vehicleTitle gehen. Das hätte ich gern eindeutig auf vehicleName. vehicleTitle ist natürlich ein Umweg, den man gehen kann, aber eindeutig muss ja nur vehicleName sein. 
_Deswegen der Wunsch nach vehicleName als Entscheidungskriterium_

5. nach einer konkreten Fahrzeugwahl ist die automatische Erkennung von evcc natürlich aus (ist ja auch normalerweise sinnvoll)
6. sofern aber z.B. die Kia-Bluelink-API nachhinkt kann es sein, dass nach ein paar Minuten Status "C" beim EV6 aufscheint
7. dann möchte ich nun also wieder auf EV6 wechseln (siehe Problem oben).

Es wäre viel einfacher, könnte man per API-Call das aktive Fahrzeug direkt per `vehicleName` ansteuern. 

So muss man 
* `vehicleName` auf  `vehicleTitle` matchen 
* mit `vehicleTitle` in `result.vehicles` nachsehen, welche ID gerade aktuell vergeben wurde, 
* dann diese ID als aktives Fahrzeug am Ladepunkt zu setzen

andig, 2023-01-03:
> in der /api/state Response ist result.vehicles.0 nicht immer dasselbe Fahrzeug.

Das lässt sich ja leicht ändern. In jedem Fall siehst Du ja aber jetzt schon, welches es ist!

> man muss über vehicleTitle gehen.

Das ist Absicht. Die Names sind ein ausschließlich internes Ordnungskriterium ohne weitere Bedeutung. Da wird sich auch nix ändern.

An der Stelle kennst Du also die ID, musst die Fahrzeuge aber unterschiedlich betiteln (was ohnehin sinnvoll ist...)

> Es wäre viel einfacher, könnte man per API-Call das aktive Fahrzeug direkt per vehicleName ansteuern.

Du kannst die ID nutzen. Wir können gerne dafür sorgen, dass die Reihenfolge statisch bleibt, jedenfalls solange keine weiteren Fahrzeuge dazu kommen.

binderth, 2023-01-03:
> Du kannst die ID nutzen. Wir können gerne dafür sorgen, dass die Reihenfolge statisch bleibt, jedenfalls solange keine weiteren Fahrzeuge dazu kommen.

ok, der `vehicleName` ist doch extra in der evcc.yaml eindeutig. Aber wenn das nicht so einfach geht, den zu verwenden wäre in der Tat eine "statische" ID (am besten nach Reihenfolge in der yaml?) die second-best option! 

danke!

andig, 2023-01-03:
Sind jetzt nach `name` sortiert: https://github.com/evcc-io/evcc/commit/6f55969f940b315714d1ef0c17ae0adf6201fb7a. Im Api hat der nix zu suchen.

andig, 2023-01-03:
Dann machen wir mal zu- danke für die Geduld es zu erklären ;) Und dran denken: im Nightly sind die IDs 1-based!

closed 2023-01-03 by andig