evcc-io/evcc #5671: 2 pvs 2 wallboxen 2 häuser via 1 evcc server?

dc-de, 2023-01-03:
Hallo zusammen,

ich habe ein echtes Luxus problem, ich habe 2 Häuser neben einander mit jeweils einer PV und jweils einer Wallbox, Stromkreise sind über eigene Hausanschlüssel getrennt, gebäude sind über VPN verbunden.
Ich möchte gerne mit einer evcc instance beides managen aber ich weiß nicht wie ich dafür die Konfig hinbekommen soll so das in jedem Haus der Jeweilige Enegerie Verbrauch für die dort vorhandene Wallbox getrennt genutzt wird.
Ja ich habe die Doku gelesen, aber so eine Situation ist da nicht berücksitigt und ja ich habe es mit zwei instancen versucht aber das ist unpraktisch und fühtr zu API timeouts wegen zu vieler anfragen ect. und zwei mal ein System verwalten ist mist, alles an einem ort wäre toll auch wenn man mehrere Häuser hat so wie ich.
Danke für eure Hilfe!

Hier die aktuelle config:
```
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: xx

interval: 10s # control cycle interval

meters:
- type: template
  template: fronius-solarapi-v1
  usage: grid
  host: 192.168.11.xx
  name: grid1
- type: template
  template: fronius-solarapi-v1
  usage: pv
  host: 192.168.11.xx
  name: pv2

#andere meters und PV gebäude 2
#- type: template
#  template: fronius-solarapi-v1
#  usage: grid
#  host: 192.168.22.xx
#  name: grid2
#- type: template
#  template: fronius-solarapi-v1
#  usage: pv
#  host: 192.168.22.xx
#  name: pv3

chargers:
- type: template
  template: ellipro
  ski: 
  ip: 192.168.11.xx
  name: wallbox5
- type: template
  template: ellipro
  ski: 
  ip: 192.168.22.xx
  name: wallbox6

vehicles:
- type: template
  template: id
  title: ID.4 GTX
  name: ev3
- type: template
  template: porsche
  title: Porsche Taycan
  name: ev4

loadpoints:
- title: VW Wallbox Haus 1
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
- title: VW Wallbox Haus 2
  charger: wallbox6
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: EV Laden
  meters:
    grid: grid1
    pvs:
    - pv2

eebus:
  certificate:
      private: |

      public: |

          
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.23 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.123 # [currency]/kWh
  
# 2. Strom Tarif Gebäude 2  
#  tariffs:
#  currency: EUR # (default EUR)
#  grid:
#    type: fixed
#    price: 0.25 # [currency]/kWh

#  feedin:
#    type: fixed
#    price: 0.101 # [currency]/kWh
```

premultiply, 2023-01-04:
Das Szenario ist in der Tat momentan so nicht unterstützt.
evcc verwaltet derzeit pro Instanz nur eine site welche auch einem Hausanschluss als Regelungspunkt entspricht.

Du bräuchtest daher aktuell in deinem Szenario pro Haus eine eigene Instanz mit je eigener Konfig (beim Aufruf mit Parameter `-c <PFAD_KONFIG_DATEI>`).

Beide Instanzen können dabei aber prinzipiell auf einem System laufen wenn man genau darauf achtet dass sie für alles verschiedene Ressourcen verwenden.
Insbesondere werden z. B. verschiedene Ports für HTTP, Topic-Namen für MQTT und Dateinamen für die Datenbank benötigt.

Vielleicht wäre dies sogar mal ein erster, einigermaßen sinnvoller Anwendungsfall für Docker.

dc-de, 2023-01-04:
Danke für die Antwort,
ein resourcen problem habe ich nicht, aktuell habe ich ja zwei Instanz pro Haus und wollte diese nur übersichtlicher in einer Weboberfläche zusammenfassen.
Außerdem scheinen beide Instanzen gegeneinadner zu arbeiten so das es immer wieder zu API timeouts kommt und die Autos verschwinden mal bei der einen mal bei der anderen Instanz bzw werden als unavailable angezeigt.