evcc-io/evcc #5689: Support voor Alfen API

Dominic-070, 2023-01-05:
When Dynamic Load Balancing is switched on in the Alfen charging station with the reading of the P1 meter, Modbus cannot be switched on at the same time.

Use the Alfen API to communicate

https://gitlab.com/LordGaav/alfen-eve/-/tree/develop/alfeneve

andig, 2023-01-05:
> The "Active load balancing" license is required for external Modbus control of the charger. Enable "Active Load Balancing" and select "Energy Management System" as Data Source in the configuration. When using "Double" charger both loadpoints need to be added. The the first port (or single) is accessable on ID 1, second port on ID 2.

This should work

Dominic-070, 2023-01-05:
i have the license. you only have 2 options in the charging station. you enable active load balancing ( p1) or modbus. both can't. when I switch on modbus, the charging station no longer monitors whether I do not exceed my capacity of 3 x 25 A.

ikruglov, 2023-01-06:
I'm an Alfen charger owner which is integrated with evcc. @andig is right. You need the "Energy Management System" option. Not sure which Alfen license is necessary for this.

Dominic-070, 2023-01-06:
I know how to switch on Modbus in the charging station. I also have a license.
The problem is, **Enabling** Modbus automatically **disables** Dynamic load balancing via the P1 meter. The charging station itself is then no longer able to monitor that I do not exceed my max current (in the event of a Home Assistant / EVCC failure).

ikruglov, 2023-01-06:
P1 and "TCP modbus", same as "Energy Management System" are mutually exclusive things. You cannot have them combined.

Dominic-070, 2023-01-06:
I see options for the api here: https://github.com/evcc-io/evcc/discussions/4356
egnerfl Has also forkt a perfectly working API integration for Alfen in Home Assistant.

Dominic-070, 2023-01-06:
Did you manage to integrate the envoy into EVCC?

andig, 2023-01-06:
@Dominic-070 could you clarify with Alfen what license you need to get both Modbus and dynamic load balancing? Seems this is not a question of evcc.

yliefting, 2023-01-10:
I have the same issue. Car will charge at full power (~11kw) even when it should have load balanced. Alfen eve single charger with enabled active load balancing license, set to "Energy management system" and connected to evcc over modbus. This behaviour actually just caused one of my three main fuses (3x25A) to blow...

If I understand correctly, in order to effectively use the load balancing feature of my Alfen charger I need to switch it back to the P1 meter. But then I won't have the nice features of evcc such as solar charging. Or am I missing something here?

at4hawo1, 2023-01-14:
There are different optional licenses available for Alfen charges. If you simply want to use the charger in combination with evcc, you need the "active load balancing" license with data source Energy Management System as described above. Then evcc is able to control the charging station via modbus.
Dynamic load balancing or smart charging network are other Alfen licenses which can be acquired additionaly. These are interesting if you want to use several charging stations at the same time and if you want to balance the charging power between them. Seems to be unnecessary when you use evcc because this is more or less similar to what evcc already does...

andig, 2023-01-22:
> P1 and "TCP modbus", same as "Energy Management System" are mutually exclusive things. You cannot have them combined.

@ikruglov what does that mean in terms of evcc? I find the discussion hard to follow. Are we lacking a feature and if so which one?

ikruglov, 2023-01-27:
@andig, Alfen has multiple ways of achieving load-balancing. In general, [here is a short summary](https://knowledge.alfen.com/space/IN/639762449/How+do+I+configure+the+Smart+Charging+functionalities+such+as+Active+Load+Balancing+and+Smart+Charging+Network%3F) from Alfen. But I'll explain it a more straightforward ways below to the best of my understanding. In my explanation under "active load balancing" I'll mean a mode where the charger reduces currents based on home consumption to avoid overloading main fuses:

1. Active load balancing. This mode has several flavors:
(a) Smart Meter. In this mode, Alfen reads data from an external meter and makes a decision about local currents based on that. So, the decision point of what current power should be is the charger itself. This has two modes:
      - P1 or DSMR meter. This is a Dutch thing where Alfen connects to a smart meter and reads currents from there. 
      - TCP ModBus meter. Same as (a) but the source of currents is a TCP ModBus-capable meter.
(c) External Management System (EMS). This is where Alfen becomes a "passive device" following orders from an external management system such as evcc. The protocol used for such communication is ModBus.

Note #1. "External Management System" and "SmartMeter TCP ModBus" both use Modbus for communication but they are not the same. The difference is who makes a decision. It's evcc (for instance) in the former and Alfen itself in the latter case. These two modes are usually mixed.

Note #2. All of the modes above are mutually exclusive. You cannot have EMS and SmartMeter load balancing at the same time.

Note #3. Since evcc doesn't currently implement load-balancing, the recommended Alfen setup via EMS doesn't implement load-balancing. Hence, messaging about overloaded main fuses.

Note #4. One needs to base the "Active Load balancing" license from Alfen (cost ~200 euro) to run any of the modes above.

2. Smart Charging Network. In this mode, a group of Alfen boxes can self-organize and (re)distribute power from the same source between them. This mode is mutually exclusive with (1) and requires a separate license: "Smart Charging Network". Cost around 100 euro.

3. There is also "Standard load balancing" and "Suspend Charging Mode". The former is when double-socket Alfen can (re)distribute power between its socket. The latter is when one can give Alfen an external signal via special wiring to pause charging. Note that "pause" means: stop-charging in this case. IMO both of these modes are not interesting for evcc and hence I won't go into details. I can read about them using the link I gave above.

4. OCPP smart charging profiles. I don't know much about this mode but, I assume, there is an OCPP standard that allows configuring the charging station including setting max allowed currents, etc. The application to configure Alfen and, I think, Alfen's mobile app uses OCPP for settings. I think that the topic started suggested using this mode to implement load-balancing. I don't know whether this is possible. Perhaps yes via a poor-man-change-settings-every-30s-or-so but this seems the wrong way of doing things.

Anyway, I hope that my message clarifies things. Please note that this explanation is to the best of my knowledge. I'm not an Alfen expert.

ikruglov, 2023-01-27:
I realized that I didn't answer the question about what this all means for evcc. In my opinion, as soon as evcc starts having load-balancing we're all will be good. By "we" I mean Alfen owners who want to use evcc for smart charging from solar panels (or other cool things evcc gives you) but also want to reduce charging currents automatically in cases of high consumption by home.

myhacsint, 2023-01-31:
I would appreciate having an API integration into evcc as well. I don't have a modbus license for my Alfen Single Pro-Line WB and I have no way of purchasing one (the supplier is not handing out these licenses, it's complicated...) -  in addition, OCPP doesn't work for me either since it is already in use by the provider for reimbursement purposes. I think the only restriction with the API method is switching between 3 phases and 1 phase operation, which might not be possible. Other than that, it should work (using egnerfl ' s Home Assistant integration works fine).

andig, 2023-01-31:
To summarise: using the HTTP over the Modbus API helps a) save money on licenses b) use Alfens loadbalancing while *still* using evcc and c) solves some edge cases like cannot purchase license.

Given that I don't see a huge showstopper per se plus we already have a working integration (modbus) I'd be happy to accept a PR but probably won't invest more efforts in the (somewhat peculiar HTTP api). The evcc codebase has a plethora of HTTP-based integrations that could serve as starting point for anyone who wants to try.

Thanks @ikruglov for the detailed explanation.

closed 2023-01-31 by andig

MarcelDelhez, 2023-04-08:
I have read your posts about the Alfen charging stations (I also have one connected to P1 port of my meter) and, it confirms me I am going into the right direction by connecting the chairging station to an ESP32 in place of the digital meter.  The ESP reads a MQTT topic and sends a fake P1 telegram to the Afen charger.  Will learn go and evcc code ... if I have the time ... to write a new interface to that fake charger (the ESP) or ... if I have less time, I will write only for my usage in python and a shelly button to select the mode of usage.
Advantage of the ESP way of doing, it works for any charger with load balancing capabilities.