evcc-io/evcc #5690: FOX ESS Battery

grusslev, 2023-01-05:
This manufacturer seems not yet included in the list of devices. Interfaces are RS485, DRM and CAN.

Is any other device similar or can somebody design the connection for EVCC?

andig, 2023-01-05:
Does is have Modbus over RS485? Do you have docs?

grusslev, 2023-01-07:
this is the only details I could find regarding the interface:

[Fox ESS.pdf](https://github.com/evcc-io/evcc/files/10366081/Fox.ESS.pdf)

andig, 2023-01-08:
That is unfortunately not nearly enough. Happy to look into this again if you can find a description of the modbus registers.

andig, 2023-01-08:
Actually, this looks decent: https://github.com/StealthChesnut/HA-FoxESS-Modbus/wiki/Data-Register-Reference---H1-AC1

grusslev, 2023-01-08:
Super! Danke 
    	Mit besten Grüßen / With best regardsMartin Gruß / GrussMobile +4915781590839

closed 2023-01-09 by andig