evcc-io/evcc #5696: Chargin sessions in .csv file show sometimes wrong values for charged energy

SirkoVZ, 2023-01-06:
### Describe the bug

It seems that somehow the charged energy of the previous session is added to current session when the odometer hasn't changed.
Here some real world sample data from my session.csv, please see the 3rd line, last column that just subtracts the 
Endzählerstand - Anfangszählerstand - Energie 
This should be 0, but it shows about -5,986 kWh, similar to the value of the session from the day before (line 4).
The Kilometerstand of both sessions is the same (12000).
Probably here is the issue.
But in line 8 there is also a small difference (2,241 kWh), here were 4 km driven.

The charged energy value sent as messages directly after the charging (or when disconnecting) is fine. So it seems that only the database contains wrong values, or the csv export is erroneous.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File href="file:///G:/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List href="file:///G:/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Startzeit | Endzeit | Ladepunkt | Kennung | Fahrzeug | Kilometerstand (km) | Anfangszählerstand (kWh) | Endzählerstand (kWh) | Energie (kWh) | Formel: H-G-I
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
06.01.2023 08:51 | 06.01.2023 09:15 | Carport |   | Ioniq 5 | 12140 | 1800,4 | 1804,678 | 4,301 | -0,023
05.01.2023 22:09 | 05.01.2023 23:05 | Carport |   | Ioniq 5 | 12117 | 1790,332 | 1800,4 | 10,113 | -0,045
04.01.2023 15:25 | 04.01.2023 16:09 | Carport |   | Ioniq 5 | 12000 | 1782,371 | 1790,332 | 13,947 | -5,986
03.01.2023 12:22 | 03.01.2023 17:00 | Carport |   | Ioniq 5 | 12000 | 1776,448 | 1782,371 | 5,949 | -0,026
28.12.2022 14:01 | 28.12.2022 15:37 | Carport |   | Ioniq 5 | 11935 | 1759,231 | 1776,448 | 17,301 | -0,084
27.12.2022 10:35 | 27.12.2022 13:59 | Carport |   | Ioniq 5 | 11915 | 1746,52 | 1759,231 | 12,774 | -0,063
21.12.2022 20:13 | 21.12.2022 22:28 | Carport |   | Ioniq 5 | 11792 | 1722,071 | 1746,52 | 26,69 | -2,241
18.12.2022 10:14 | 18.12.2022 12:58 | Carport |   | Ioniq 5 | 11788 | 1716,417 | 1719,957 | 3,552 | -0,012
17.12.2022 11:24 | 17.12.2022 13:58 | Carport |   | Ioniq 5 | 11756 | 1712,002 | 1716,417 | 4,423 | -0,008
14.12.2022 14:08 | 15.12.2022 07:20 | Carport |   | Ioniq 5 | 11660 | 1691,491 | 1712,002 | 20,617 | -0,106
10.12.2022 09:39 | 10.12.2022 11:50 | Carport |   | Ioniq 5 | 11601 | 1668,049 | 1691,491 | 23,567 | -0,125

</body>

</html>

### Steps to reproduce

I can only guess, but it seems that when there are two charging sessions with the same odometer value (means: not driven in between) then the charged energy of both is added.

### Configuration details

```yaml
I can provide, if necessary
```

### Log details

```text
I can provide, if necessary
```

### What type of operating system are you running?

Linux

### Version

0.110.1

andig, 2023-01-06:
> or the csv export is erroneous.

Das kannst Du prüfen wenn Du in die Sqlite Datenbank schaust. Um hier sonst weiter zu kommen brauchen wir ein Logfile in dem wir sehen, welche Werte wir von den Zählern erhalten (log: trace). Ich mache erstmal zu, bis wir ein passendes Logfile zu einem Fehlerfall haben.

andig, 2023-01-06:
I see. We're creating new sessions even when the vehicle is not unplugged.

andig, 2023-01-06:
@SirkoVZ could you share a logfile where- without unplugging- a new charging session is generated? I'd like to fix the root cause. As workaround we could also- if a meter is available- just use the meter difference to recalculate instead of the session energy which may for one reason or other always be wrong. Seems we should do both.

SirkoVZ, 2023-01-06:
I can't find useful debug information in the log.
What log level should I use? core? site? ...?

Yes, I think meter start/stop difference would be a good idea.

andig, 2023-01-06:
Trace log für Loadpoint, `db` und Wallbox bitte

SirkoVZ, 2023-01-06:
Nach dem Anstecken wird in Zeile 24 die Session angelegt (in Zeile 29 wieder beendet, weil noch nicht geladen werden sollte).
In Zeile 103 wird die Ladung gestartet, in Zeile 168 beendet.
In Zeile 206 wird die 2. Ladung gestartet, ohne eine neue Session anzulegen, sondern die 1. Session weiter verwendet.
In Zeile 277 wird die 2. Session beendet.

[trace2.log](https://github.com/evcc-io/evcc/files/10362388/trace2.log)

andig, 2023-01-06:
Siehst du, wo die 2. Session her kommt die dann zu gemacht wird?

SirkoVZ, 2023-01-06:
ich meinte, es ist immer eine Session, ich hab nur 2 Ladungen gestartet und beendet.
Meiner Meinung nach sollte in Zeile 206 eine neue Session angelegt werden (ein neues CREATE statement)

andig, 2023-01-08:
Du schriebst in 277 wird die 2. Session beendet. Du meinst 2. Ladung?

SirkoVZ, 2023-01-09:
Korrkt, die 2. Ladung, ich habs oben ein "Edit:" dran geschrieben

andig, 2023-01-22:
> ich meinte, es ist immer eine Session, ich hab nur 2 Ladungen gestartet und beendet.
> Meiner Meinung nach sollte in Zeile 206 eine neue Session angelegt werden (ein neues CREATE statement)

Ich denke es ist andersrum: solange das Auto nicht abgestöpselt wird sollte es eine Session sein. Ich frag mir nur, warum bei Dir 2 geschrieben werden? Hast Du eine Idee?

Bleibt noch das Problem, des Neustarts zwischendurch. In dem Fall- so ein Zähler vorhanden ist- sollten wir den nutzen um die Session Energy zu korrigieren. Dann enthalten alte Session vor Neustart und neue Session jeweils den korrekten Teil. Ohne Zähler ist das nicht entscheidbar.

SirkoVZ, 2023-01-24:
Nur um sicher zugehen, bei mir ist:
- eine Session ist vom Anstecken bis Abziehen
- ein Ladung ist innerhalb der Session, 2 Ladungen sind dann von 17-19 Uhr und die zweite von 7-9 Uhr. Dann wird abgezogen und die Session ist auch beendet.

So war das bei mir auch, 
- ich stecke an
- die Sonne scheint und die Ladung beginnt
- es wird dunkel, die Ladung wird beendet
- am nächsten Tag scheint die Sonne wieder und die 2. Ladung wird gestartet
- die 2. Ladung wird beendet und abgezogen

Jetzt hab ich beide Ladungen im Ladelog, wobei die 2. die Werte der ersten nochmal enthält.

foto-andreas, 2023-02-10:
Hi,

ich habe auch doppelte Ladungen in der CSV-Datei und in der GUI. Log etc. Betrifft die Ladungen am 8.2.

Andreas

[session-2023.02.csv](https://github.com/evcc-io/evcc/files/10710646/session-2023.02.csv)

[trace1.log](https://github.com/evcc-io/evcc/files/10710640/trace1.log)

andig, 2023-02-11:
In #6197 gibts jetzt einen zusätzlichen Test der das Verhalten überprüft. Ergebnis: bei mehreren Start/Stop Zyklen wird genau eine Session erzeugt. Das Verhalten, dass 2 Sessions generiert werden kann ich nicht nachvollziehen. Da liegt der Ball leider bei Euch.

Was bleibt ist der Fehler der doppelten Energiemenge: der entsteht, wenn evcc neu gestartet wird, das Auto aber *nicht* getrennt. Das sieht für evcc wie ein neuer Ladevorgang aus. Der lässt sich über den Zähler plausibilisieren falls einer eingebaut ist- das werde ich gleich noch im PR ergänzen.

@foto-andreas gibts interessante Timestamps im Logfile die anzuschauen lohnen würde?

andig, 2023-02-11:
> Jetzt hab ich beide Ladungen im Ladelog, wobei die 2. die Werte der ersten nochmal enthält.

Die einzige Erklärung die ich habe wäre Neustart- das sorgt für 2 Sessions, mit dem aktuellen Fix aber korrekter Energiemenge. Die 2 Sessions lassen sich nicht verhindern.

closed 2023-02-11 by andig

foto-andreas, 2023-02-11:
@andig wenn ich das Trace-Log vom richtigen Tag genommen hätte, hätten wir beide gesehen, dass da in der Tat ein Restart von evcc gewesen ist. Dann wäre das Problem aus meiner Sicht dann auch erledigt und ich suche mir bessere Zeiten für Updates... Danke

SirkoVZ, 2023-02-12:
Von meiner Seite auch alles gut: Sowohl nur noch ein Session, als auch exakte kWh.
Perfekt! Danke! :-)

andig, 2023-02-12:
Sehr schön- an der Anzahl der Sessions hab ich allerdings nix geändert ;)

andig, 2023-04-14:
Seit https://github.com/evcc-io/evcc/pull/6940 wird eine Sessionenergie>0 beim Neustart ignoriert. Das ist eine alternative Lösung für dieses Issue. Aufgrund #7214 wird die Lösung hier zurück gebaut- die Zählerstände selbst sind nicht hilfreich.