evcc-io/evcc #5727: Sofar Solar HYD 10-20kW + smart Chint DTSU666

tomipaci, 2023-01-08:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

premultiply, 2023-01-09:
Please provide all communication interface and implementation details like API description, Modbus register mapping table etc. from vendors.

We'll close here until is information are available.

sahomm, 2023-02-27:
Erst einmal herzlichen Dank an die große Mühe die sich hier gemacht wird! Sponsoring ist bereits eingeleitet :) 

Meine Anlage soll kommende Woche ebenso mit einem Sofar HYD 15kw und GTX 3000 Batterien in Betrieb gehen und ich wäre hier sehr an einer Lösung interessiert. Die Tabelle mit den Modus Registern liegt mir vor. Ich sehe mich jedoch leider nicht imstande hier zur Implementation effektiv beizutragen. 

Wo bzw auf welchem Weg kann ich diese Tabelle zur Verfügung stellen?

premultiply, 2023-02-27:
Einfach hier reinziehen oder direkten Link einfügen.

sahomm, 2023-02-27:
Hallo und vielen dank für schnelle Antwort!

hier gewünschten Dateien:
[SOFARSOLAR ModBus-RTU Communication Protocol.pdf](https://github.com/evcc-io/evcc/files/10841440/SOFARSOLAR.ModBus-RTU.Communication.Protocol.pdf)

[211116_SOFAR HYD-3PH and SOFAR -G3 Modbus Protocol 2021-10-14_Client.xlsx](https://github.com/evcc-io/evcc/files/10841443/211116_SOFAR.HYD-3PH.and.SOFAR.-G3.Modbus.Protocol.2021-10-14_Client.xlsx)

Eine Implementierung wäre für mich sehr wünschenswert. Auch denke ich, dass dieser Typ in naher Zukunft häufiger auftauchen wird.

Grüße 

Saho

premultiply, 2023-02-27:
https://community.openenergymonitor.org/uploads/short-url/dF8j79PsvtCS0xQ5aPxHXO4HZDE.pdf

sahomm, 2023-02-27:
> Thanks!

> https://community.openenergymonitor.org/uploads/short-url/dF8j79PsvtCS0xQ5aPxHXO4HZDE.pdf

thanks!

andig, 2023-02-28:
@sahomm jetzt müsstest Du mal die relevanten Register raus suchen und schauen ob die Werte stimmen. Entweder baust Du Dir dafür einen „custom“ meter in der evcc config oder benutzt irgendein anderes Tool zum ausprobieren (mbrtu, Homeassisstant etc). Wenn die Werte klar sind können wir das in ein Template giessen.

sahomm, 2023-02-28:
@andig vielen Dank! Ich bekomme das System leider erst in den kommenden 1-2 Wochen installiert und kann dann erst aktiv beginnen zu unterstützen. Gerne suche ich dann die meines Erachtens relevanten Register raus.

Wäre diese Konfiguration unter "meters" so richtig gem. meins o.g. xls?
`meters:  
  name: test Sofar
  type: custom
  power: # Leistung (W)
    source: modbus
	device: /dev/ttyUSB0
	baudrate: 9600
	comset: "8N1"
		register:
  		address: 0485
  		type: holding # holding or input
  		decode: int32 # int16|32|64, uint16|32|64, float32|64 and u|int32s + float32s
  `

Wo würde ich dann die Daten im evcc sehen? Im Log?

cschlipf, 2023-03-11:
Ich habe meinen Sofar HYD 10KTL-3PH nun installiert und würde sehr gerne helfen. Bin allerdings totaler EVCC Anfänger.

@sahomm - was Du hier konfigurierst ist die serielle Schnittstelle. Der Datenlogger redet doch aber hoffentlich via TCP/IP. Zumindest lässt dieses Projekt darauf schließen: https://github.com/MichaluxPL/Sofar_LSW3. Schau Dir mal die Config Datei hier an:

```
[SofarInverter]
inverter_ip=X.X.X.X             # data logger IP
inverter_port=8899              # data logger port
inverter_sn=XXXXXXXXXX          # data logger S/N
register_start1=0x0000          # Inverter register's first MODBUS address for the first register's range.
register_end1=0x0027            # Inverter register's last MODBUS address for the first register's range
register_start2=0x0105          # Inverter register's first MODBUS address for a second register's range
register_end2=0x0114            # Inverter register's last MODBUS address for a second register's range
registerhw_start=0x2000         # Like above, but for InverterHWData.py
registerhw_end=0x200D           # Like above, but for InverterHWData.py
```

Ich kann bestätigen, dass der Port 8899 offen ist und habe es mal Naiv mit diesem YAML probiert:

```
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: ad7c947eccf9f6f4dfc367c3ed4ee3c72700b82bf2cf06076af66bd8d1a0e683

interval: 10s # control cycle interval

site:
  title: MyHome
  meters:
    grid: sofar-inverter

meters:
  - name: sofar-inverter
    type: modbus
    power:
      source: modbus
      id: 1
      uri: 192.168.158.58:8899
      register:
        address: 0485
        type: holding
        decode: int32
```
Erhalte damit aber folgende Fehlermeldung von evcc:

```
[main  ] INFO 2023/03/11 22:35:26 evcc 0.114.1
[main  ] INFO 2023/03/11 22:35:26 using config file: C:\Users\cschlipf\private\app\evcc\evcc.yaml
[main  ] INFO 2023/03/11 22:35:26 starting ui and api at :7070
[db    ] INFO 2023/03/11 22:35:26 using sqlite database: C:\Users\cschlipf\.evcc\evcc.db
[main  ] FATAL 2023/03/11 22:35:26 cannot create meter 'sofar-inverter': cannot create meter 'modbus': 1 error(s) decoding:

* 'Power' expected type 'string', got unconvertible type 'map[string]interface {}', value: 'map[id:1 register:map[address:485 decode:int32 type:holding] source:modbus uri:192.168.158.58:8899]'
[main  ] FATAL 2023/03/11 22:35:26 will attempt restart in: 5m0s
```

Also offensichtlich habe ich hier noch irgendwo einen Fehler drin. @andig - Ein paar Beispiel YAMLs wären toll. Die Doku erklärt zwar ganz ausführlich die einzelnen Settings - aber meist fehlt der Kontext.

premultiply, 2023-03-11:
type: custom

cschlipf, 2023-03-12:
Vielen Dank, @premultiply - Damit bin ich nun etwas weiter gekommen. Allerdings hänge ich nun hier fest und ich fürchte da könnt ihr auch nicht weiterhelfen. Wenn ich evcc meter ausführe erhalte ich das hier:

`Power: read failed: read tcp 192.168.158.80:59850->192.168.158.58:8899: i/o timeout`

Timeout habe ich schon auf 10s hochgesetzt.

Das passt auch zum Tool von MichaluxPL, das ich oben erwähnt habe. Wenn ich dieses ausführe, hängt das Tool und kommt nicht zurück - vermutlich wohl auch wegen Timeouts.

Nun bietet Sofar ja noch eine API an:
[SolarmanOpenAPI(v1.1.0)-EN.pdf](https://github.com/evcc-io/evcc/files/10950986/SolarmanOpenAPI.v1.1.0.-EN.pdf)
Allerdings ist diese relativ nutzlos, da hier die Daten nur alle 5min aktualisiert werden und gerade dieses Delay mein Grund ist mich mit EVCC zu befassen.

cschlipf, 2023-03-12:
Ein Firmware Upgrade des Wechselrichters hat nicht geholfen. Der Stick Logger LSW-3 hat allerdings auch eine recht alte Firmware bei mir. Habe mal Sofar gebeten mir das Update aufzuspielen. Eventuell ändert das etwas.

premultiply, 2023-03-12:
Der Wechselrichter kann nur RS485.
Mit Modbus TCP brauchst du also gar nichts zu probieren.

Wenn dann müsstest du dieses Logger abfragen. Aber das ist ein ganz anderes Protokoll und Produkt und hat nichts mit dem WR zu tun.

Dafür ggf. anderes Issue.

cschlipf, 2023-03-12:
Hm, dann habe ich wohl dieses Projekt hier falsch verstanden: https://github.com/MichaluxPL/Sofar_LSW3: "Get MODBUS data from Sofar (K-TLX) inverter through LSW-3 or LSE module"

cschlipf, 2023-03-12:
Sehe gerade den PR... Wow! Kann ich irgendwie helfen das zu testen?

premultiply, 2023-03-12:
> Hm, dann habe ich wohl dieses Projekt hier falsch verstanden: https://github.com/MichaluxPL/Sofar_LSW3: "Get MODBUS data from Sofar (K-TLX) inverter through LSW-3 or LSE module"

Ja, das entscheidende Wort ist **through**

andig, 2023-03-13:
Ist jetzt im Nightly, in 15min zum Download bereit

closed 2023-03-13 by andig

cschlipf, 2023-03-13:
> Ist jetzt im Nightly, in 15min zum Download bereit

Sorry für die dumme Frage... wo kann ich den Nightly (hier aktuell noch Windows) herunterladen?

cschlipf, 2023-03-13:
Habe nun evcc unter WSL installiert und habe das unstable APT Repository eingetragen. Evcc läuft unter WSL auch (die Demoseite kam auf Anhieb hoch). Wenn ich aber `evcc configure` eingebe fehlt hier noch die SofarSolar Auswahl:
```
 Wähle ein PV System:  [Use arrows to move, type to filter]
  SENEC .Home
  SMA Data Manager
  SMA Smart Energy Hybrid-Wechselrichter
> SMA Sunny Home Manager 2.0
  SolarEdge Hybrid Inverter
  SolarEdge Wechselrichter
  Solarlog
```

Habe ich doch nicht den nightly build?

```
$ evcc --version
evcc version 0.114.1 (3c6e2ca3)
```

cschlipf, 2023-03-13:
Ach halt, unter Netz-Stromzähler habe ich ihn gefunden. Aber leider kann er nicht angesprochen werden:

```
- Konfiguration Netz-Stromzähler

 Wähle einen Netz-Stromzähler: SofarSolar Inverter / Hybrid Inverter

Führe folgende Einstellungen durch:

 Wähle die ModBus Schnittstelle aus TCP/IP
 Modbus ID 1
 IP-Adresse oder Hostname 192.168.158.58
 Port 502

Teste die grid Konfiguration von SofarSolar Inverter / Hybrid Inverter ...
   Fehler: read failed: dial tcp 192.168.158.58:502: i/o timeout

 Der Test der grid Konfiguration von SofarSolar Inverter / Hybrid Inverter ist fehlgeschlagen. Soll grid trotzdem in die Konfiguration aufgenommen werden? No

 Möchtest du erneut ein Gerät aus der Liste auswählen und einrichten? Yes

 Wähle einen Netz-Stromzähler: SofarSolar Inverter / Hybrid Inverter

Führe folgende Einstellungen durch:

 Wähle die ModBus Schnittstelle aus TCP/IP
 Modbus ID 1
 IP-Adresse oder Hostname 192.168.158.58
 Port 8899

Teste die grid Konfiguration von SofarSolar Inverter / Hybrid Inverter ...
   Fehler: read failed: read tcp 172.21.94.62:38254->192.168.158.58:8899: i/o timeout

 Der Test der grid Konfiguration von SofarSolar Inverter / Hybrid Inverter ist fehlgeschlagen. Soll grid trotzdem in die Konfiguration aufgenommen werden? (y/N)
```

cschlipf, 2023-03-13:
OK, nach weiterem einlesen in die Materie habe ich gemerkt, dass ich wohl auf dem Holzweg bin und die ganze Zeit immer noch über den Logger versuche. Hatte noch nicht kapiert, dass RS485 Hardware Bastelei am Wechselrichter braucht. Phew! Da muss ich mich erstmal mehr mit befassen.

cschlipf, 2023-03-13:
Sorry, für meine dummen Fragen, aber bräuchte ich jetzt sowas hier um den RS485 Port vom Wechselrichter an das Ethernet anzuschließen um dann mit "Serial (Ethernet-RS485 Adapter)" darauf zugreifen zu können?

premultiply, 2023-03-13:
https://eckstein-shop.de/QITAUSBRS485KonverterBusAdapterSeriellSchnittstelleRaspberryPi

premultiply, 2023-03-13:
Oder wenns unbedingt aufwändiger und teurer sein soll: Protoss PE11-H bei Alibaba

cschlipf, 2023-03-13:
> Protoss PE11-H

Vielen Dank!

cschlipf, 2023-03-14:
RS485 Adapter und Raspberry PI sind bestellt. Ich werde berichten.

Wenn ich mir das Template anschaue, habe ich noch ein paar Zweifel ob das zum meinem Sofar Hybrid Wechselrichter  HYD 10 KTL passen wird. Ich glaube das sind die Register aus dem PDF, welches die älteren Wechselrichter von Sofar anspricht. Die neueren Hybridwechselrichter haben die Adressen aus dem Excel, das hier schon angehängt wurde: 
[211116_SOFAR.HYD-3PH.and.SOFAR.-G3.Modbus.Protocol.2021-10-14_Client.xlsx](https://github.com/evcc-io/evcc/files/10968831/211116_SOFAR.HYD-3PH.and.SOFAR.-G3.Modbus.Protocol.2021-10-14_Client.xlsx)

Ich vermute die register addressen wären beim Hybridwechselrichter demnach:

- Feed in/out power / ActivePower_Output_Total: 0485, int16, 0.01
- Grid A Current / Current _Output_L1N: 04B1, uint16, 0.01
- Grid B Current / Current_Output_L2N: 04B6, uint16, 0.01
- Grid C Current / 
- Total energy buy from grid / Energy_Purchase_Total: 068E, uint32, 0.1
- Total generation / PV_Generation_Total: 0686, uint32, 0.1
- Battery Charge/Discharge current / Current_Bat1: 0605, int16, 0.01
- The residual capacity of battery / 

Ich habe da mal als Laie ins Blaue geschossen und hoffe, dass da zumindest ein paar Treffer dabei sind.

PeterPablo, 2023-03-15:
Ein Freund von mir betreibt einen solchen SofarSolar-Wechselrichter mit Akku mit evcc und zwar rein per TCP/IP (kein Modbus, keine Konverter). Wir haben die von evcc benötigten Werte per mqtt aus HomeAssistant exportiert und lesen diese in evcc ein.
Vorteil: Geht bereits heute ohne HW-Gebastel.
Nachteil: Setzt eine HomeAssistant-Installation voraus.

```
meters:
[...]
- name: pv
  type: custom
  power:
    source: mqtt
    topic: PV_W

- name: battery
  type: custom
  power:
    source: mqtt
    topic: Batt_W
    scale: -1
  soc:
    source: mqtt
    topic: soc
```

Im HomeAssistant haben wir über ein kleines Skript die mqtt topics veröffentlicht. Wie das SofarSolar-"plugin" von HomeAssistant arbeitet kann ich aus dem Stehgreif nicht sagen.

PeterPablo, 2023-03-15:
Dem nicht so tief versierten Anwender mag das nicht klar sein, dass man auch anders ans Ziel kommen kann. Der Code von Solarman mag außerdem für jemanden (TM) als Grundlage für eine Implementierung des TCP/IP-Ansatzes in evcc dienen. Ich wollte lediglich helfen. Darf gerne als "offtopic" markiert werden.

andig, 2023-03-15:
Danke dir- wollte nur erklären, warum wir das nicht machen solange es eine (gute) Alternative gibt 👍🏻

premultiply, 2023-03-16:
Die Solarman-Logger sprechen ein ganz eigenes Protokoll. Das müsste man von Grund auf in Go entwickeln.
Letztendlich werden darüber indirekt als Proxy-Protokoll exakt die gleichen Modbus-Abfragen an das Gerät gestellt wie evcc sie direkt stellt. Ist nur noch mehr Verpackung aussen herum.

Hier die Protokollbeschreibung:
https://pysolarmanv5.readthedocs.io/en/latest/solarmanv5_protocol.html

cschlipf, 2023-03-19:
Die Solarman Logger kann man auch wie oben schon gesagt ziemlich vergessen. Updates kommen da minimal alle 5 min - oft sogar noch Zeitverzögerter. Das ist für mich der Grund, warum ich clever-pv (die Solarman unterstützen) den Rücken kehren will.

Raspberry PI und RS485 Dongle sind da. Ich muss jetzt nur noch die Zeit finden und die Register herausfinden.

cschlipf, 2023-03-19:
> @andig vielen Dank! Ich bekomme das System leider erst in den kommenden 1-2 Wochen installiert und kann dann erst aktiv beginnen zu unterstützen. Gerne suche ich dann die meines Erachtens relevanten Register raus.
> 
> Wäre diese Konfiguration unter "meters" so richtig gem. meins o.g. xls?
> 
> ```yaml
> meters:  
> name: test Sofar
>   type: custom
>   power: # Leistung (W)
>     source: modbus
> 	device: /dev/ttyUSB0
> 	baudrate: 9600
> 	comset: "8N1"
> 		register:
>   		address: 0485
>   		type: holding # holding or input
>   		decode: int32 # int16|32|64, uint16|32|64, float32|64 and u|int32s + float32s
> ```
> 
> Wo würde ich dann die Daten im evcc sehen? Im Log?

Versuche es mal mit `evcc meter`

cschlipf, 2023-03-19:
OK, ich bin weitergekommen und kann die ersten Register auslesen. Das Excel stimmt jedenfalls für die HYD Wechselrichter. Da hier die älteren Sofar Solar Inverter hinzugefügt worden sind, habe ich für die neueren Wechselrichter ein neues Issue aufgemacht: #6953

PR #6808 (2023-03-13): Add Sofarsolar template
Fixes #5727