evcc-io/evcc #5732: Tasmota Error

Kai9555, 2023-01-09:
### Describe the bug

Leider erhalte ich keine Daten von meinem Balkonkraftwerk, welches über einen Shelly mit Tasmota auslesen wird. Log siehe unten. 

Hat jemand eine Idee was der Fehler bedeuten könnte?

Vielen Dank
Kai

### Steps to reproduce

.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: xxx

interval: 5s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sungrow-inverter 
  id: 1  
  host: 192.168.x.xx 
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- name: tasmota
  type: custom
  power:
     source: http
     uri: http://192.168.x.xx/cm?cmnd=Status%208
     jq: .StatusSNS.MT175.P
- name: Growatt
  type: template
  template: tasmota
  usage: pv
  host: 192.168.x.xx

vehicles:
  - name: my_car
    type: template
    template: tesla
    title: Tesla Model Y
    accessToken: xxxxx
    capacity: 79

chargers:
- type: template
  template: go-e-gemini 
  host: 192.168.x.xx 
  name: wallbox4

loadpoints:
- title: Carport
  charger: wallbox4
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Mein Zuahuse
  meters:
    grid: tasmota
    pvs: 
    - pv1
    - Growatt
```

### Log details

```text
pi@homebridge:~ $ evcc meter
[main  ] INFO 2023/01/09 14:22:12 evcc 0.110.1 (4b9e5700)
[main  ] INFO 2023/01/09 14:22:13 using config file: /etc/evcc.yaml
[db    ] INFO 2023/01/09 14:22:13 using sqlite database: /home/pi/.evcc/evcc.db
pv1
---
Power:  465W
Energy: 161.0kWh

tasmota
-------
Power: -114W

Growatt
-------
Power:  json: cannot unmarshal array into Go struct field .StatusSNS.Energy.Power of type int
Energy: json: cannot unmarshal array into Go struct field .StatusSNS.Energy.Power of type int
```

### What type of operating system are you running?

Linux

### Version

110.1

andig, 2023-01-09:
@thierolm könntest Du hier mal wieder rein schauen? Scheint, als gäbe es mal wieder eine neue Tasmota Firmware?

andig, 2023-01-09:
@Kai9555 kannst Du das bitte nochmal mit `--log trace` machen damit wir sehen, was Tasmota da schickt? Was für eine Version ist das?

Kai9555, 2023-01-09:
> --log trace

Natürlich gern - Version 12.2.0.4

Hätte eine Anpassung von eurer Seite Auswirkung auf ältere Tasmota Versionen beim Auslesen?

```
pi@homebridge:~ $ evcc meter --log trace
[main  ] INFO 2023/01/09 18:57:39 evcc 0.110.1 (4b9e5700)
[main  ] INFO 2023/01/09 18:57:39 using config file: /etc/evcc.yaml
[db    ] INFO 2023/01/09 18:57:40 using sqlite database: /home/pi/.evcc/evcc.db
[db    ] TRACE 2023/01/09 18:57:40 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/01/09 18:57:40 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/01/09 18:57:40 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/01/09 18:57:40 SELECT * FROM `settings` 0 <nil>
pv1
---
[modbus] TRACE 2023/01/09 18:57:40 modbus: send 00 01 00 00 00 06 01 04 13 a6 00 02
[modbus] TRACE 2023/01/09 18:57:40 modbus: recv 00 01 00 00 00 07 01 04 04 00 00 00 00
[modbus] TRACE 2023/01/09 18:57:40 modbus: send 00 02 00 00 00 06 01 04 13 8b 00 02
[modbus] TRACE 2023/01/09 18:57:40 modbus: recv 00 02 00 00 00 07 01 04 04 00 a1 00 00
Power:  0W
Energy: 161.0kWh

tasmota
-------
[http  ] TRACE 2023/01/09 18:57:40 GET http://192.168.2.65/cm?cmnd=Status%208
[http  ] TRACE 2023/01/09 18:57:40 {"StatusSNS":{"Time":"2023-01-09T18:57:40","MT175":{"E_in":1111.805,"P":890,"L1":109,"L2":57,"L3":723,"E_out":62.511,"Meter_id":""}}}
Power: 890W

Growatt
-------
[tasmota] TRACE 2023/01/09 18:57:40 GET http://192.168.2.90/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2023/01/09 18:57:40 {"StatusSNS":{"Time":"2023-01-09T18:57:39","Switch1":"ON","Switch2":"OFF","ANALOG":{"Temperature":49.6},"ENERGY":{"TotalStartTime":"2023-01-09T13:59:15","Total":0.077,"Yesterday":0.000,"Today":0.077,"Power":[0,0],"ApparentPower":[5,0],"ReactivePower":[4,0],"Factor":[0.00,0.00],"Frequency":50,"Voltage":237,"Current":[0.000,0.000]},"TempUnit":"C"}}
[tasmota] TRACE 2023/01/09 18:57:40 GET http://192.168.2.90/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2023/01/09 18:57:40 {"StatusSNS":{"Time":"2023-01-09T18:57:39","Switch1":"ON","Switch2":"OFF","ANALOG":{"Temperature":49.6},"ENERGY":{"TotalStartTime":"2023-01-09T13:59:15","Total":0.077,"Yesterday":0.000,"Today":0.077,"Power":[0,0],"ApparentPower":[5,0],"ReactivePower":[4,0],"Factor":[0.00,0.00],"Frequency":50,"Voltage":237,"Current":[0.000,0.000]},"TempUnit":"C"}}
Power:  json: cannot unmarshal array into Go struct field .StatusSNS.Energy.Power of type int
Energy: json: cannot unmarshal array into Go struct field .StatusSNS.Energy.Power of type int
```

andig, 2023-01-09:
Danke, das hilft. Muss allerdings sagen, dass mich die Leistungswerte etwas ratlos machen. Was sollen die zwei Werte bedeuten?

Kai9555, 2023-01-10:
Ih hoffe ich verstehe deine Frage richtig...

Grundsätzlich handelt es sich um einen Shelly 2.5 - also können 2 Leitungen gemessen werden. In meinem Fall ist natürlich nur eine Leitung (Balkonkraftwerk) angeschlossen.

Warum der Shelly 2.5?
Shelly 1PM misst keine Spannung = Ergebnisse ungenau 
Shelly EM misst mit 1% Genauigkeit = bei der kleinsten 50A Klemme eine Ungenauigkeit von bis zu 30% bei den geringen Strömen eines Balkonkraftwerk

Lösung:
Shelly 2.5 kann Spannung und Stromstärke messen und das mit einer Genauigkeit von 0,5%.

andig, 2023-01-10:
@Kai9555 d.h. im Falle oben hättest Du eine Leistung von 5W auf L1?

Kai9555, 2023-01-10:
> @Kai9555 d.h. im Falle oben hättest Du eine Leistung von 5W auf ~L1~ dem ersten Kanal?

Nein, zu dem Zeitpunkt war die Anlage bereits aus - zu wenig Sonne - ;)

`"ApparentPower":[5,0]` ist die Scheinleistung. Die Wirkleistung wird unter `"Power":[0,0]` ausgegeben. In dem Fall also 0W.

Hier mal der Auszug von gerade eben. Dort ist zu sehen das gerade 7W unter `"Power":[7,0],"` ausgegeben werden. Unter `"ApparentPower":[149,0]` grenzt sich jetzt deutlicher die Scheinleistung mit 149 VA ab.

```
Growatt
-------
[tasmota] TRACE 2023/01/10 10:00:48 GET http://192.168.2.90/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2023/01/10 10:00:48 {"StatusSNS":{"Time":"2023-01-10T10:00:47","Switch1":"ON","Switch2":"OFF","ANALOG":{"Temperature":39.1},"ENERGY":{"TotalStartTime":"2023-01-09T13:59:15","Total":0.079,"Yesterday":0.077,"Today":0.002,"Power":[7,0],"ApparentPower":[149,0],"ReactivePower":[136,0],"Factor":[0.05,0.00],"Frequency":50,"Voltage":236,"Current":[0.570,0.000]},"TempUnit":"C"}}
[tasmota] TRACE 2023/01/10 10:00:48 GET http://192.168.2.90/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2023/01/10 10:00:48 {"StatusSNS":{"Time":"2023-01-10T10:00:48","Switch1":"ON","Switch2":"OFF","ANALOG":{"Temperature":39.1},"ENERGY":{"TotalStartTime":"2023-01-09T13:59:15","Total":0.079,"Yesterday":0.077,"Today":0.002,"Power":[7,0],"ApparentPower":[149,0],"ReactivePower":[136,0],"Factor":[0.05,0.00],"Frequency":50,"Voltage":236,"Current":[0.570,0.000]},"TempUnit":"C"}}
Power:  json: cannot unmarshal array into Go struct field .StatusSNS.Energy.Power of type int
Energy: json: cannot unmarshal array into Go struct field .StatusSNS.Energy.Power of type int
```

thierolm, 2023-01-10:
@Kai9555 , die aktuell von mir realisierte Tasmota Lösung unterstützt nur Tasmota Devices mit einem Messkanal. Deswegen auch der Fehler... .
Die Erweiterung auf zwei Kanäle wird etwas dauern und ich habe auch kein entsprechendes Testgerät...

ABER: Da du ja einen Shelly nutzt und offensichtlich mit Tasmota geflasht hast: Warum flashst du nicht wieder die original Shelly Software? Diese habe ich schon so implemtiert, dass due den genutzten Messkanal konfigurieren kannst ...

Kai9555, 2023-01-10:
> @Kai9555 , die aktuell von mir realisierte Tasmota Lösung unterstützt nur Tasmota Devices mit einem Messkanal. Deswegen auch der Fehler... . Die Erweiterung auf zwei Kanäle wird etwas dauern und ich habe auch kein entsprechendes Testgerät...
> 
> ABER: Da du ja einen Shelly nutzt und offensichtlich mit Tasmota geflasht hast: Warum flashst du nicht wieder die original Shelly Software? Diese habe ich schon so implementiert, dass du den genutzten Messkanal konfigurieren kannst ...

Hi @thierolm das hat Kompatibilitätsgründe, denn unter solaranzeige.de wird der shelly 2.5 nicht nativ unterstützt, jedoch geflasht  mit Tasmota. Der kleinste gemeinsame Nenner war hier einfach Tasmota. Jedenfalls dachte ich das. Zu Gunsten der Messegenauigkeit (siehe Erklärung oben) kann ich mich mit einer Wartezeit der Implementierung anfreunden. Gern kann ich beim Testen der Umsetzung unterstützen.

thierolm, 2023-01-10:
Hm, wenn ich dass richtig sehe, und du die Dose schon mit Solaranzeige.de nutzt, könntest du dir das meter auch mit dem `mqtt` plugin definieren und aus deinem Solaranzeige Mosquitto Server auslesen ... Siehe: https://docs.evcc.io/docs/reference/plugins/#mqtt-lesenschreiben

Kai9555, 2023-01-10:
> Hm, wenn ich dass richtig sehe, und du die Dose schon mit Solaranzeige.de nutzt, könntest du dir das meter auch mit dem `mqtt` plugin definieren und aus deinem Solaranzeige Mosquitto Server auslesen ...
> Siehe:
> https://docs.evcc.io/docs/reference/plugins/#mqtt-lesenschreiben und
> https://docs.evcc.io/docs/reference/configuration/mqtt

Danke für den Hinweis. Ich versuche mich mal einzulesen. Beim ersten Überfliegen kommen mir aber schon Zweifel ob ich es aus der Kalten schaffe. Ich habe nur rudimentäre Kenntnisse und würde mich eher als Rookie bezeichnen.

Würdest du im Zweifel eine Integration dennoch in Betracht ziehen?

thierolm, 2023-01-10:
Info aus der Solaranzeige Doku:
[image]

thierolm, 2023-01-10:
> Würdest du im Zweifel eine Integration dennoch in Betracht ziehen?

Ja klar, ich schaue mir das an.

Kai9555, 2023-01-10:
> Info aus der Solaranzeige Doku:
> [image]

Ich habe nach kurzer Recherche folgenden Hinweis gefunden:

„evcc hat im Standard ein Regelinterval von 10 Sekunden, deine Solaranzeige braucht die Daten alle 60s wenn ich das richtig sehe
D.h. evcc sollte die Daten lesen und Solaranzeige sollte die Daten von evcc holen
Wenn du die Daten von Solaranzeige an evcc übergibts, werden die Werte in der evcc Oberfläche immer falsch sein, weil ein Teil alle 10s gelesen wird, ein anderer alle 60s, und das passt eben nicht zusammen“

Es ist zwar möglich aber offensichtlich nicht der saubere Weg. Ich scheue mich nicht vor Arbeit, möchte aber das sich am Ende die Mühen lohnen und alles sauber läuft.

Derzeit habe ich schon einen Sponsor Token und habe auch fest eingeplant diesen zu erweitern wenn am Schluss alles sauber läuft. Ich würde mich sehr über eine saubere Integration von euch in evcc freuen. Gern kann ich dir zum Testen auch einem Shelly 2.5 mit tasmota zukommen lassen wenn es dir hilft.

thierolm, 2023-01-10:
Alles gut! :-) Ich schaue mir das die Tage an und gebe Feedback, sobald ich eine testbare Multi-Meter Tasmota Version habe.

thierolm, 2023-01-11:
@Kai9555 , der PR #5748 oben sollte dein Problem lösen.
Sobald @andig den PR gemerged hat, könntest du bitte das darauffolgende nightly Release testen?

Kai9555, 2023-01-11:
> @Kai9555 , der PR #5748 oben sollte dein Problem lösen.
> Sobald @andig den PR gemerged hat, könntest du bitte das darauffolgende nightly Release testen?

Danke für die schnelle Realisierung! Mache ich sofort 👍🏻 @andig erhalte ich eine Info von dir?

thierolm, 2023-01-11:
Wichtig, wie im PR kurz beschrieben, wird bei mehreren Meterkanälen immer nur der erste Meter-Kanal genutzt.
Falls wirklich einmal ein Szenario existiert, in dem der für evcc relevante Kanal konfigurierbar sein muss, muss dies noch angefordert/umgesetzt werden.

Kai9555, 2023-01-11:
> Wichtig, wie im PR kurz beschrieben, wird bei mehreren Meterkanälen immer nur der erste Meter-Kanal genutzt.
> Falls wirklich einmal ein Szenario existiert, in dem der für evcc relevante Kanal konfigurierbar sein muss, muss dies noch angefordert/umgesetzt werden.

Super passt 👍🏻

closed 2023-01-12 by andig

Kai9555, 2023-01-13:
Gerade getestet - funktioniert!! Super Arbeit - Dankeschön :)