evcc-io/evcc #5733: evcc beginnt im PV Modus auch ohne PV Erzeugung spontan zu laden

MarkusGH, 2023-01-09:
### Describe the bug

Fahrzeug ist angesteckt, wolkiger Tag,, keine PV Erzeugung.
evcc gibt um 16:43 spontan den Ladestrom frei, erst um 17:06 wird das Laden beendet.

Auffällig ist, dass im Log in dem Bereich wo geladen wird die "site power" Einträge fehlen, diese fangen erst 17:03 wieder an.

### Steps to reproduce

Nicht reproduzierbar

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: localhost
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 3s # control cycle interval

log: info
levels:
  porsche: trace
  lp-1: debug
  lp-2: debug
  openwb: trace

mqtt:
  broker: mqtt.xxx.de:1883
  topic: evcc

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.1570 # [currency]/kWh
  feedin:
    type: fixed
    price: 0.0816 # [currency]/kWh

meters:
- name: grid1
  type: movingaverage
  decay: 1 # 0.5
  meter:
    type: custom
    power:
      source: mqtt
      topic: mbmd/solaredge1-1-1/Power
      scale: -1  # Die Leistung wird in Watt erwartet. Falls z.b. kW geliefert wird, kann dies hier mit dem Wert 1000 angepasst werden
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    energy:
      source: mqtt
      topic: mbmd/solaredge1-1-1/Import
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    currents:
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L1
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L2
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
[... 203 lines cut ...]
Jan  9 17:03:26 homebridge evcc[815]: [lp-1  ] DEBUG 2023/01/09 17:03:26 charger status: C
Jan  9 17:03:26 homebridge evcc[815]: [lp-1  ] DEBUG 2023/01/09 17:03:26 next soc poll remaining time: 59m54s
Jan  9 17:03:26 homebridge evcc[815]: [lp-1  ] DEBUG 2023/01/09 17:03:26 soc estimated: 1.01% (vehicle: 1.00%)
Jan  9 17:03:26 homebridge evcc[815]: [lp-1  ] DEBUG 2023/01/09 17:03:26 vehicle soc: 1%
Jan  9 17:03:26 homebridge evcc[815]: [lp-1  ] DEBUG 2023/01/09 17:03:26 vehicle range: 0km
Jan  9 17:03:26 homebridge evcc[815]: [lp-1  ] DEBUG 2023/01/09 17:03:26 pv charge current: 0A = 6A + -12.1A (2772W @ 1p)
Jan  9 17:03:26 homebridge evcc[815]: [lp-1  ] DEBUG 2023/01/09 17:03:26 climater active: false, target temp: 20.0°C, outside temp: 20.0°C
Jan  9 17:03:26 homebridge evcc[815]: [lp-1  ] DEBUG 2023/01/09 17:03:26 site power 2772W >= 800W disable threshold
Jan  9 17:03:26 homebridge evcc[815]: [lp-1  ] DEBUG 2023/01/09 17:03:26 pv disable in 2m58s
Jan  9 17:03:26 homebridge evcc[815]: [lp-1  ] DEBUG 2023/01/09 17:03:26 pv disable timer remaining: 2m58s
Jan  9 17:03:26 homebridge evcc[815]: [openwb] TRACE 2023/01/09 17:03:26 send openWB/set/isss/heartbeat: '1'
Jan  9 17:03:26 homebridge evcc[815]: [openwb] TRACE 2023/01/09 17:03:26 send openWB/set/isss/heartbeat: '1'
Jan  9 17:03:27 homebridge evcc[815]: [openwb] TRACE 2023/01/09 17:03:27 recv openWB/lp/2/kWhCounter: '1416.763'
Jan  9 17:03:27 homebridge evcc[815]: [openwb] TRACE 2023/01/09 17:03:27 recv openWB/lp/2/W: '1331'
Jan  9 17:03:29 homebridge evcc[815]: [lp-1  ] DEBUG 2023/01/09 17:03:29 charge power: 1331W
Jan  9 17:03:30 homebridge evcc[815]: [openwb] TRACE 2023/01/09 17:03:30 recv openWB/lp/2/kWhCounter: '1416.764'
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.110.1

andig, 2023-01-09:
Klar:

> climater active: true

closed 2023-01-09 by andig

MarkusGH, 2023-01-10:
My bad. Ich würde ja sagen RTFM, aber ich glaube das ist nirgends dokumentiert. :-)
Das wurde schon mal diskutiert, habe es wieder gefunden. Für das Feature gibt es sinnvolle use cases, in anderen Fällen ist es komplett unerwünscht. Könnte man da nicht einen Schalter in die Konfiguration einbauen um das für den PV Modus an/abschalten zu können? Denn die Anforderungen bei einem PHEV sind andere als bei einem BEV, und dort hängt es dann auch wieder davon ab wie die Nutzung ist und wie hoch die PV Erzeugung.