evcc-io/evcc #5744: Audi e-tron/Elli charger: vehicle soc: parsing time "" as "2006-01-02T15:04:05Z07:00": cannot parse "" as "200

joslippert, 2023-01-11:
### Describe the bug

Seit evcc version 0.111.0 taucht folgende Fehlermeldung auf:

[lp-1  ] ERROR 2023/01/11 11:39:39 vehicle odometer: parsing time "" as "2006-01-02T15:04:05Z07:00": cannot parse "" as "2006"

Platform: Ubuntu 22.04.1 LTS

### Steps to reproduce

1. starte evcc version 0.111.0
2. Audi e-tron

### Configuration details

```yaml
chargers:
- type: template
  template: ellipro
  ski: mySKI
  ip: x.x.x.x 
  name: Elli-Wallbox

vehicles:
- name: EV1
  type: template
  template: etron
  title: Audi_e-tron
  user: myEmail
  password: myPWD
  vin: WAU....
  capacity: 86  # KWH
```

### Log details

```text
[lp-1  ] ERROR 2023/01/11 11:39:39 vehicle odometer: parsing time "" as "2006-01-02T15:04:05Z07:00": cannot parse "" as "2006"
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.111.0

andig, 2023-01-11:
Bitte detailliertes Logfile oder Zugangsdaten an info@evcc.io

joslippert, 2023-01-11:
Trace-Mode:

evcc@evcc:~$ /home/evcc/evcc.0.111.0  -l trace vehicle
[main  ] INFO 2023/01/11 12:06:21 evcc 0.111.0
[main  ] INFO 2023/01/11 12:06:21 using config file: /home/evcc/evcc.yaml
[db    ] INFO 2023/01/11 12:06:21 using sqlite database: /home/evcc/.evcc/evcc.db
[db    ] TRACE 2023/01/11 12:06:21 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/01/11 12:06:21 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/01/11 12:06:21 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/01/11 12:06:21 SELECT * FROM `settings` 6 <nil>
[eebus ] INFO 2023/01/11 12:06:21 Local SKI:  e56132a0d82c528c490fb90e4a0942f494a5383f
[eebus ] DEBUG 2023/01/11 12:06:21 mdns: announce
[eebus ] DEBUG 2023/01/11 12:06:21 mdns: using avahi
[eebus ] DEBUG 2023/01/11 12:06:21 starting websocket server on :4712
[etron ] TRACE 2023/01/11 12:06:21 GET https://identity.vwgroup.io/oidc/v1/authorize?client_id=f4d0934f-32bf-4ce4-b3c4-699a7049ad26%40apps_vw-dilab_com&code_challenge=gD7nI6-CJQgXhPoREBYVc6NuFJCCQ4V4N7s3pKnC2Oo&code_challenge_method=S256&prompt=login&redirect_uri=myaudi%3A%2F%2F%2F&response_type=code+id_token+token&scope=openid+profile+mbb&ui_locales=de-DE&nonce=CYIGDuposBodvdVdqqDvidlQRcEPTPzdkNWhQmmEowB&state=5f509175-585d-4c58-b746-eabfda937160
[etron ] TRACE 2023/01/11 12:06:21 GET https://identity.vwgroup.io/signin-service/v1/signin/f4d0934f-32bf-4ce4-b3c4-699a7049ad26@apps_vw-dilab_com?relayState=911371b85ae8f577777aa986fe70cff426beda78
[etron ] TRACE 2023/01/11 12:06:21 <!DOCTYPE html>
<html lang="de" xmlns="http://www.w3.org/1999/html">
<head>
    <meta charset="utf-8"/>
    <meta name="viewport"
          content="width=device-width, height=device-height, initial-scale=1, maximum-scale=1, user-scalable=no"/>
    <meta name="identitykit" content="loginIdentifier"/>
    
<script type="text/javascript">(window.NREUM||(NREUM={})).init={privacy:{cookies_enabled:true},ajax:{deny_list:["bam.eu01.nr-data.net"]},distributed_tracing:{enabled:true}};(window.NREUM||(NREUM={})).loader_config={agentID:"535875755",accountID:"2547829",trustKey:"2217462",xpid:"VgMDVl5RARAEUlFQBQQOXlY=",licenseKey:"NRJS-ecd5929e8859410bbd9",applicationID:"460225982"};/*! For license information please see nr-loader-spa-1220.min.js.LICENSE.txt */
!function(e,t){"object"==typeof exports&&"object"==typeof module?module.exports=t():"function"==typeof define&&define.amd?define([],t):"object"==typeof exports?exports.NRBA=t():e.NRBA=t()}(self,(function(){return function(){var e,t,n={9034:function(e,t,n){"use strict";var r=n(4168);t.Z=(0,r.ky)(16)},5973:function(e,t,n){"use strict";n.d(t,{I:function(){return r}});var r=0,o=navigator.userAgent.match(/Firefox[\/\s](\d+\.\d+)/);o&&(r=+o[1])},4280:function(e,t,n){"use strict";n.d(t,{H:function(){return o}});var r=document.createElement("div");r.innerHTML="\x3c!--[if lte IE 6]><div></div><![endif]--\x3e\x3c!--[if lte IE 7]><div></div><![endif]--\x3e\x3c!--[if lte IE 8]><div></div><![endif]--\x3e\x3c!--[if lte IE 9]><div></div><![endif]--\x3e";var o,i=r.getElementsByTagName("div").length;o=4===i?6:3===i?7:2===i?8:1===i?9:0},5955:function(e,t,n){"use strict";n.d(t,{I:function(){return r}});var r=function(e,t){var n=this;return e&&"object"==typeof e?t&&"object"==typeof t?(Object.assign(this,t),void Object.entries(e).forEach((function(e){var t=e[0],r=e[1];n[t]=r}))):console.error("setting a Configurable requires a model to set its initial properties"):console.error("setting a Configurable requires an object as input")}},441:function(e,t,n){"use strict";n.d(t,{C:function(){return c},L:function(){return u}});var r=n(1424),o=n(5955),i={beacon:r.ce.beacon,errorBeacon:r.ce.errorBeacon,licenseKey:void 0,applicationID:void 0,sa:void 0,queueTime:void 0,applicationTime:void 0,ttGuid:void 0,user:void 0,account:void 0,product:void 0,extra:void 0,jsAttributes:{},userAttributes:void 0,atts:void 0,transactionName:void 0,tNamePlain:void 0},a={};function c(e){if(!e)throw new Error("All info objects require an agent identifier!");if(!a[e])throw new Error("Info for "+e+" was never set");return a[e]}function u(e,t){if(!e)throw new Error("All info objects require an agent identifier!");a[e]=new o.I(t,i),(0,r.Qy)(e,a[e],"info")}},1476:function(e,t,n){"use strict";n.d(t,{Dg:function(){return u},Mt:function(){return s},P_:function(){return c}});var r=n(1424),o=n(5955),i={privacy:{cookies_enabled:!0},ajax:{deny_list:void 0,enabled:!0},distributed_tracing:{enabled:void 0,exclude_newrelic_header:void 0,cors_use_newrelic_header:void 0,cors_use_tracecontext_headers:void 0,allowed_origins:void 0},ssl:void 0,obfuscate:void 0,jserrors:{enabled:!0},metrics:{enabled:!0},page_action:{enabled:!0},page_view_event:{enabled:!0},page_view_timing:{enabled:!0},session_trace:{enabled:!0},spa:{enabled:!0}},a={};function c(e){if(!e)throw new Error("All configuration objects require an agent identifier!");if(!a[e])throw new Error("Configuration for "+e+" was never set");return a[e]}function u(e,t){if(!e)throw new Error("All configuration objects require an agent identifier!");a[e]=new o.I(t,i),(0,r.Qy)(e,a[e],"config")}function s(e,t){if(!e)throw new Error("All configuration objects require an agent identifier!");var n=c(e);if(n){for(var r=t.split("."),o=0;o<r.length-1;o++)if("object"!=typeof(n=n[r[o]]))return;n=n[r[r.length-1]]}return n}},2085:function(e,t,n){"use strict";n.d(t [... cut ...]
Range:         parsing time "" as "2006-01-02T15:04:05Z07:00": cannot parse "" as "2006"
Odometer:      parsing time "" as "2006-01-02T15:04:05Z07:00": cannot parse "" as "2006"
Finish time:   parsing time "" as "2006-01-02T15:04:05Z07:00": cannot parse "" as "2006"
Climater:      parsing time "" as "2006-01-02T15:04:05Z07:00": cannot parse "" as "2006"
Target Soc:    parsing time "" as "2006-01-02T15:04:05Z07:00": cannot parse "" as "2006"
[eebus ] DEBUG 2023/01/11 12:06:28 mdns: stop announcement

closed 2023-01-11 by andig