evcc-io/evcc #5749: Error with meter template solax-hybrid-cloud

ioniks, 2023-01-11:
### Describe the bug

When i configure solax-hybrid-cloud template i have error in my log every 5 secdonds

### Steps to reproduce

1. configure a meter with template "solax-hybrid-cloud"
Like this 
`- type: template
  template: solax-hybrid-cloud
  usage: grid
  tokenid: XXXXX
  serial: XXXXX
  name: grid
- type: template
  template: solax-hybrid-cloud
  usage: pv
  tokenid: XXXXXX
  serial: XXXXX
  name: pv1
- type: template
  template: solax-hybrid-cloud
  usage: battery
  tokenid: XXXXXX
  serial: XXXXXXX
  name: battery`

### Configuration details

```yaml
EVCC 0.111.0 now and 0.110.1 before have the same problem. 
In a docker container
```

### Log details

```text
`[site  ] ERROR 2023/01/11 13:10:52 pv 1 power: add[0]: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] ERROR 2023/01/11 13:10:52 battery 1 power: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] ERROR 2023/01/11 13:10:52 battery 1 soc: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] ERROR 2023/01/11 13:10:53 grid meter: strconv.ParseFloat: parsing "<nil>": invalid syntax`
```

### What type of operating system are you running?

Docker container

### Version

0.111.0

andig, 2023-01-11:
Bitte detailliertes Logfile (--log trace)

ioniks, 2023-01-11:
With i active trace, i have an error TokenId missing, and the tokenId is a wird int.

Si i have ot add "TokenId", not TokenId because it's a number :/

Mybad

closed 2023-01-11 by ioniks