evcc-io/evcc #5780: Memory Leak in provider/cache

nename0, 2023-01-14:
Hallo,

Beim Implementieren meines eigenen Chargers, bin ich auf ein Memory Leak gestoßen in Zusammenhang mit dem `EventBus` den `provider/cache.go` verwendet.
Die 'übliche' Implementation eines Chargers hat eine `reset` Funktion die dann `Cached` [aufruft](https://github.com/evcc-io/evcc/blob/master/charger/zaptec.go#L122), um einen neuen gecached Getter für den Zustand zu bauen. Bei Statusänderungen wird dann `reset` [aufgerufen](https://github.com/evcc-io/evcc/blob/master/charger/zaptec.go#L186) um den Cache zurückzusetzen.

Wenn man dieses Vorgehen jetzt nimmt und in eine Schleife packt:
```golang
var cacheTimeout = time.Second
for i := 0; i < 1_000_000; i++ {
	if i%1000 == 0 {
		provider.ResetCached()
	}

	var getter = provider.Cached(func() (string, error) {
		return "foo", nil
	}, cacheTimeout)

	_, _ = getter()
}
```
bekommt man schnell viel Speichernutzung. Das liegt an den ganzen Aufrufen von
https://github.com/evcc-io/evcc/blob/6e60acfb09a01fc4c5f40a90de26daf4fceffb1a/provider/cache.go#L39
die nie aufgeräumt werden.

Denke es gibt da verschiedene Ansätze, dass zu lösen.
1) Eine etwas unschöne Lösung mit `runtime.SetFinalizer` um `Unsubscribe` aufzurufen wenn der zurückgegebene Getter nicht mehr reachable ist:
    <details>
     <summary>Implementierung</summary>
      
    ```golang
    type finalizableGetter[T any] struct {
	    getFunc func() (T, error)
    }
    
    func (g *finalizableGetter[T]) Get() (T, error) {
	    return g.getFunc()
    }
    
    // Cached wraps a getter with a cache
    func Cached[T any](g func() (T, error), cache time.Duration) func() (T, error) {
	    c := ResettableCached(g, cache)
	    // Because we need to unsubscribe from the event bus again
	    // once the method value returned from here is gargabe collected,
	    // create a dummy object to pass to SetFinalizer which just forwards the get call
	    var fg = &finalizableGetter[T]{
		    getFunc: c.Get,
	    }
	    var resetFunc = c.Reset
	    _ = bus.Subscribe(reset, resetFunc)
	    runtime.SetFinalizer(fg, func(fg2 *finalizableGetter[T]) {
		    _ = bus.Unsubscribe(reset, resetFunc)
	    })
	    return fg.Get
    }
    ```
      
    </details>

2) Eine Lösung um die Anzahl der Objekte für den GC zu reduzieren: Statt jedes mal einen neuen gecachten Getter zu bauen alle(!) Verwendungen von `Cached` durch `ResettableCached` ersetzen und dann jeweils in der `reset` Methode darauf `Reset` aufrufen. Den EventBus müsste man dann in `ResettableCached` subscriben und der Memory Leak wäre an sich noch da, aber kaum zu merken. Man merkt denn Leak ja scheinbar jetzt auch noch nicht wirklich...

Ich kann auch gerne ein PR machen je nachdem welche Variante ihr bevorzugt.

andig, 2023-01-14:
Ich verstehe nicht was Du meinst. Der Getter wird nur 1x beim Start erzeugt, daran ändert auch `provider.ResetCached()` nichts. Könntest Du einen Go Benchmark teilen an dem man steigenden Speicherverbrauch ablesen kann?

> Eine Lösung um die Anzahl der Objekte für den GC zu reduzieren

Das ist hier kein Problem. GC hat erstmal nix mit Speicherverbrauch zu tun.

andig, 2023-01-14:
Oder meinst Du dieses Pattern hier https://github.com/evcc-io/evcc/blob/master/charger/daheimladen.go#L76 wo neue `Cacheable`s erzeugt werden? Das wäre in der Tat leaky da die alten immer noch am Bus hängen. Good catch!

M.E. wär die richtige Lösung dafür statt

```go
// Cached wraps a getter with a cache
func Cached[T any](g func() (T, error), cache time.Duration) func() (T, error) {
	c := ResettableCached(g, cache)
	_ = bus.Subscribe(reset, c.Reset)
	return c.Get
}

// Cacheable is the interface for a resettable cache
type Cacheable[T any] interface {
	Get() (T, error)
	Reset()
}
```

eine neue Funktion einzuführen die das `Cacheable` zurück gibt so dass das `Reset()` aufgerufen werden kann anstatt immer neu `Cached()` zu verwenden.

PR wäre Klasse!

nename0, 2023-01-14:
Genau ich beziehe mich auf das Pattern, dass einige Charger (zaptec, daheimladen, ...) immer wieder ihre `reset` Methode aufrufen dort dann wieder neue Cacheables erzeugen, die im Bus hängen.

andig, 2023-01-14:
Magst Du PR machen? Im Prinzip das 

    _ = bus.Subscribe(reset, c.Reset)

nach `ResettableCached` verschieben und das `Cacheable` an allen relevanten Stellen verwenden.

Good catch- wie hast du den gefunden???

nename0, 2023-01-14:
> Magst Du PR machen? 

Ja, bin dran.

Bin jetzt mal grob durchgegangen, was 'betroffen' ist:
Charger: daheimladen, mystrom, smaevcharger, zaptec
Meter: Nix
Vehicle: Nix
 
> Good catch- wie hast du den gefunden???

Habe copy-pasta vom daheimladen charger gemacht für meinen eigenen Charger (der PR kommt auch noch ;-))
und mir dann die Cache-Implementierung angeschaut und mich iwann gefragt, wann dass wieder unsubscribed wird....

closed 2023-01-14 by andig