evcc-io/evcc #5784: planner: GUI zeigt Zielladen obwohl TargetTime in der Vergangenheit

schenlap, 2023-01-14:
### Describe the bug

Zielladen, z.b TargetTime 14.1.2023 7:00
Auto lädt wie gewünscht.
Auto abstecken, fahren und wieder anstecken (z.B 14.1.2023 um 16:00)

GUI schreibt: Zielladen geplant. Ladung startet heute 06:01 Uhr.
6:01 ist in der Vergangenheit

[image]
[image]

Erwartung: Nach aktivem Zielladen verschwindet die GUI Anzeige.

### Steps to reproduce

1. Zielladen
2. abstecken fahren
3. nach der alten Zielzeit wieder einstecken
...

### Configuration details

```yaml
Verwendung des Planners
```

### Log details

```text
siehe oben
```

### What type of operating system are you running?

Linux

### Version

v0.111.1

andig, 2023-01-15:
Der Fehler liegt nicht im UI. Hier dürfte gar kein Plan für einen Zeitpunkt in der Vergangenheit erzeugt werden. Hast Du evtl schon eine Idee, warum das passiert? Sollten wir bei https://github.com/evcc-io/evcc/pull/5649 gleich mit beheben.

andig, 2023-01-15:
Gleiche Beobachtung hier (es ist Sonntag):
[image]

schenlap, 2023-01-15:
Der planner.go ist es nicht, der müsste mit false aussteigen.
```
	// target charging without tariff or late start
	if t.tariff == nil || afterStart {
		return latestStart, targetTime, afterStart && beforeTarget, nil
	}
```

Also bleibt nur der loadpoint über. Ich vermute dass es daran liegt dass wie die targetTime nicht löschen. Woher weiß dann der Loadpoint dass alles vorbei ist?

VolkerK62, 2023-01-15:
hat evtl damit zu tun?
mir ist augefallen, wenn ich eine Zielzeit eingestellt habe und diese wieder lösche, dass die in MQTT unter loadpoint `targetTimeProjectedStart` immer noch angezeigt wird. Der Wert wird dann nciht mehr aktualisiert.

andig, 2023-01-15:
Das sollte per https://github.com/evcc-io/evcc/pull/5649/commits/a83a104f6b8dfa821a299c7240f72c131e1ec7b3 gelöst sein?

schenlap, 2023-01-17:
Ich habe es mit https://github.com/evcc-io/evcc/pull/5649/commits/8f02b4c22090f1d947c2985053b3a713edb0e3ef getestet, da geht es jetzt. Du kannst dann gerne hier schließen oder auch erst wenn pr gemerged...

closed 2023-01-19 by andig