evcc-io/evcc #5787: Variable vehicleSoc in Dokumentation falsch eingetragen

ammann-ruchti, 2023-01-14:
### Describe the bug

Guten Tag,

In der Dokumentation im Bereich Referenz > evcc.yaml > messaging steht im Kapitel Fahrzeug die Variable **vehicleSoC**. Könnt ihr diese bitte ändern auf vehicleSoc - mit kleinem c am Schluss. Mit grossem C gibt es einen Fehler.

Vielen Dank und Gruss
Hans-Jakob

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
nicht notwendig
```

### Log details

```text
Jan 14 20:58:11 evcc-raspi evcc[23700]: [push  ] ERROR 2023/01/14 20:58:11 invalid message template for start: wanted: [vehicleSoC], got: [savingsAmount vehicleIcon savingsTotalCharged batteryCapacity batteryConfigured pvRemaining vehicles connected currency vehicleCapacity vehiclePresent targetEnergy phaseRemaining siteTitle savingsSelfConsumptionPercent battery batteryPower savingsSince batterySoc phasesConfigured sponsor phasesEnabled vehicleSoc minCurrent vehicleOdometer prioritySoc version pvConfigured vehicleTargetSoc savingsEffectivePrice phasesActive phaseAction savingsGridCharged gridConfigured tariffGrid enabled savingsSelfConsumptionCharged chargeCurrent title vehicleDetectionActive pvPower auth minSoc bufferSoc targetSoc homePower pv vehicleRange chargedEnergy charging pvAction planActive maxCurrent mode chargePower tariffFeedIn chargeRemainingEnergy chargeDuration residualPower gridPower vehicleFeatureOffline connectedDuration targetTimeActive vehicleTitle]
```

### What type of operating system are you running?

Linux

VolkerK62, 2023-01-15:
Guter Hinweis. Doku ändern ist relativ einfach. Den PR kannst du selber machen. Probier es mal aus ;)

andig, 2023-01-15:
Ich mach hier schonmal zu da das Thema ausschließlich die Doku betrifft. In der `evcc.dist.yaml` ist das in den Beispielen bereits richtig drin. Danke für den Hinweis!

closed 2023-01-15 by andig