evcc-io/evcc #5811: Automatisches Zielladen

Fantus62, 2023-01-17:
Hallo Community,

nachdem ich mit dem SMA Solarmanager und einer Mennekes Charge Control Ladebox bezüglich PV-Überschussladen gescheitert bin, habe ich in EVCC ein tolles Tool gefunden. Dieses ist nun in unserem Haushalt im täglichen Einsatz und funktioniert hervorragend. 

Aktuell (Winter) lade ich hauptsächlich mit Nachtstrom. Damit dieses zuverlässig funktioniert, wird der Lademodus von EVCC nachts (22:00 Uhr) mit Hilfe eines externen Cron-Jobs auf Schnell gestellt und morgens (07:00 Uhr) wieder zurück auf PV.

Schöner wäre es, das Feature "Zielladen" für diesen Zweck zu verwenden. Hierzu müsste allerdings der Bediener nach dem Anstecken den Zielzeitpunkt jedes Mal manuell setzen. Das wird schon mal vergessen und dann ist das Auto am nächsten Morgen nicht geladen.

Ich schlage deshalb vor, ein Feature für das automatische Zielladen beim Einstecken des Autos zu implementieren.  

Eine Einstellung des PV oder des MIN-PV Modes bei Identifikation des Fahrzeuges ist ja bereits möglich. Es bräuchte einen zusätzlichen Parameter (z.B. targetTime), der die Zielzeit und optional die Wochentage beschreibt, an denen das Auto im konfigurierten Ladezustand benötigt wird.

Beispiel:
targetTime: 07:00 Mo-Fr 
targetTime: 10:00 Sa
targetTime: 08:00

Wird kein Wochentag angegeben, bezieht sich die Zielzeit auf alle Wochentage.

Da ich aus dem IT-Bereich komme, würde ich mich auch gerne selbst an der Entwicklung beteiligen. Mit git habe ich bisher noch nicht gearbeitet. Könnte mir jemand einen Tip geben, wie ich unter Linux eine Entwicklungsumgebung aufsetze?

Viele Grüße
Stephan

VolkerK62, 2023-01-17:
schau mal https://github.com/evcc-io/evcc/issues/5492

Fantus62, 2023-01-17:
Danke! Ich wusste, dass ich das irgendwo gesehen hatte, konnte es aber nicht mehr finden. Dann schließe ich dieses issue wieder.

closed 2023-01-17 by Fantus62