evcc-io/evcc #5819: template variable vehicleTitle not available

GrimmiMeloni, 2023-01-17:
### Describe the bug

I have tried to use the messaging/events section to signal charge start/stop to a script.
As a starter I have simply copied the example message template from the documentation, e.g.

```
      msg: ${vehicleTitle} connected on wallbox ${title} at ${pvPower:%.1fk}kW PV
```

However, this seems to not work, I am getting an error by EVCC claiming "vehicleTitle" does not exist.

`[push  ] ERROR 2023/01/17 18:35:47 invalid message template for connect: wanted: [vehicleTitle], got: [savingsSelfConsumptionCharged pvRemaining phasesConfigured residualPower savingsTotalCharged siteTitle batteryPower tariffGrid savingsEffectivePrice pvAction maxCurrent sponsor version gridPower bufferSoC minCurrent tariffFeedIn homePower charging phasesEnabled pvConfigured gridConfigured phasesActive connectedDuration targetSoC enabled chargePower chargeDuration phaseRemaining savingsSince savingsAmount availableVersion chargeCurrents targetTimeProjectedStart vehicles mode prioritySoC savingsGridCharged currency vehicleDetectionActive chargedEnergy releaseNotes batterySoC targetTime minSoC chargeTotalImport title phaseAction chargeCurrent savingsSelfConsumptionPercent pvPower batteryConfigured auth connected]`

I took a quick glance at the codebase and it seems the variable(s) are emitted by their respective subsystems. That said, my combination of charger (Easee) and car (Škoda Enyaq) does not allow direct vehicle detection via the Wallbox connection. Therefore I rely on the identification of the vehicle via the car's API. 

I think this is therefore a sequencing problem. At the time of the connect event (i.e. immediately after plugging in) my car is still considered a "guest". It is only after the vehicle API update, that my vehicle is correctly detected. Looking at the logs provided below it can be seen that the event is emitted **before** the vehicle identification is completed. I think it is reasonable to assume that only after the vehicle identification the vehicleTitle is available and hence the variable expansion fails.

### Steps to reproduce

1. Setup a msg template for the **connect** event using the vehicleTitle variable (see my config provided below)
2. Configure a car that is **identified via the vehicle manufacturer API** not the charger/loadpoint
3. Plugin the vehicle
4. Error message should appear as shown above when the template is expanded

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: ds218.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  easee: debug
  enyaq: debug

# unique installation id
plant: ***
sponsortoken: ***

interval: 10s # control cycle interval

meters:
- type: template
  template: tesla-powerwall
  usage: grid
  host: ***
  password: ***
  name: grid1
- type: template
  template: tesla-powerwall
  usage: pv
  host: ***
  password: ***
  name: pv2
- type: template
  template: tesla-powerwall
  usage: battery
  host: ***
  password: ***
  name: battery3

site:
  title: Zu Hause
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3
  residualPower: 100
  prioritySoC: 70 # Hausbatterie bekommt bis zum SoC 70% Priorität beim laden

chargers:
  - name: Easee
    type: template
    template: easee
    user: ***
    password: ***
    charger: ***

loadpoints:
  - title: Carport # display name for UI
    charger: Easee
    mode: off
    resetOnDisconnect: true

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.3796 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.082 # [currency]/kWh

vehicles:
  - name: Enyaq
    type: template
    template: enyaq
    title: Enyaq
    user: ***
    password: ***
    vin: *** # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 82 # Akku-Kapazität in kWh # Optional
    mode: minpv

messaging:
  events:
    start:
      title: Charging started
      msg: Wallbox ${title} started charging ${vehicleTitle} in ${mode} mode
    stop:
      title: Charging stopped
      msg: Wallbox ${title} finished charging ${vehicleTitle} with ${chargedEnergy:%.1fk}kWh in ${chargeDuration}
    connect:
      title: Vehicle connected
      msg: ${vehicleTitle} connected on wallbox ${title} at ${pvPower:%.1fk}kW PV
    disconnect:
     title: Vehicle disconnected
     msg: ${vehicleTitle} disconnected of wallbox ${title} after ${connectedDuration}

  services:
  - type: script
    cmdline: /root/scripts/eventHandler.sh
    timeout: 30s
```

### Log details

```text
[site  ] DEBUG 2023/01/17 18:35:37 ----
[lp-1  ] DEBUG 2023/01/17 18:35:37 charge power: 0W
[site  ] DEBUG 2023/01/17 18:35:37 pv power: 0W
[site  ] DEBUG 2023/01/17 18:35:37 battery power: 490W
[site  ] DEBUG 2023/01/17 18:35:37 grid power: -6W
[site  ] DEBUG 2023/01/17 18:35:37 battery soc 0: 50%
[site  ] DEBUG 2023/01/17 18:35:37 site power: 584W
[lp-1  ] DEBUG 2023/01/17 18:35:37 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/01/17 18:35:37 charger status: A
[site  ] DEBUG 2023/01/17 18:35:47 ----
[lp-1  ] DEBUG 2023/01/17 18:35:47 charge power: 4271W
[site  ] DEBUG 2023/01/17 18:35:47 pv power: 0W
[site  ] DEBUG 2023/01/17 18:35:47 battery power: 4620W
[site  ] DEBUG 2023/01/17 18:35:47 grid power: 160W
[site  ] DEBUG 2023/01/17 18:35:47 battery soc 0: 50%
[site  ] DEBUG 2023/01/17 18:35:47 site power: 4880W
[lp-1  ] DEBUG 2023/01/17 18:35:47 charge currents: [5.99 6.08 5.94]A
[lp-1  ] DEBUG 2023/01/17 18:35:47 charger status: C
[lp-1  ] INFO 2023/01/17 18:35:47 car connected
[lp-1  ] DEBUG 2023/01/17 18:35:47 vehicle api refresh
[lp-1  ] DEBUG 2023/01/17 18:35:47 pv timer elapse
[lp-1  ] DEBUG 2023/01/17 18:35:47 pv timer inactive
[lp-1  ] INFO 2023/01/17 18:35:47 start charging ->
[push  ] ERROR 2023/01/17 18:35:47 invalid message template for connect: wanted: [vehicleTitle], got: [savingsSelfConsumptionCharged pvRemaining phasesConfigured residualPower savingsTotalCharged siteTitle batteryPower tariffGrid savingsEffectivePrice pvAction maxCurrent sponsor version gridPower bufferSoC minCurrent tariffFeedIn homePower charging phasesEnabled pvConfigured gridConfigured phasesActive connectedDuration targetSoC enabled chargePower chargeDuration phaseRemaining savingsSince savingsAmount availableVersion chargeCurrents targetTimeProjectedStart vehicles mode prioritySoC savingsGridCharged currency vehicleDetectionActive chargedEnergy releaseNotes batterySoC targetTime minSoC chargeTotalImport title phaseAction chargeCurrent savingsSelfConsumptionPercent pvPower batteryConfigured auth connected]
[push  ] ERROR 2023/01/17 18:35:48 invalid message template for start: wanted: [vehicleTitle], got: [chargeCurrents title phaseAction availableVersion savingsSelfConsumptionCharged savingsGridCharged targetTimeProjectedStart batteryPower savingsAmount maxCurrent batterySoC vehicles targetTime chargeTotalImport connected phasesEnabled gridPower batteryConfigured savingsSince phasesConfigured phasesActive enabled auth chargeDuration mode chargeCurrent minCurrent tariffGrid pvPower pvConfigured version savingsSelfConsumptionPercent currency connectedDuration pvRemaining gridConfigured siteTitle prioritySoC phaseRemaining releaseNotes pvAction charging bufferSoC tariffFeedIn chargePower savingsEffectivePrice minSoC savingsTotalCharged homePower sponsor vehicleDetectionActive targetSoC residualPower chargedEnergy]
[main  ] DEBUG 2023/01/17 18:35:49 vehicle status: C (Enyaq)
[lp-1  ] INFO 2023/01/17 18:35:49 vehicle updated: unknown -> Enyaq
[lp-1  ] DEBUG 2023/01/17 18:35:49 set charge mode: minpv
[lp-1  ] DEBUG 2023/01/17 18:35:49 pv timer elapse
[lp-1  ] DEBUG 2023/01/17 18:35:49 pv timer inactive
[lp-1  ] DEBUG 2023/01/17 18:35:50 vehicle status: C
[lp-1  ] DEBUG 2023/01/17 18:35:50 vehicle soc: 56%
[lp-1  ] DEBUG 2023/01/17 18:35:50 vehicle target soc: 80%
[lp-1  ] DEBUG 2023/01/17 18:35:50 vehicle range: 243km
[lp-1  ] WARN 2023/01/17 18:35:50 charger out of sync: expected disabled, got enabled
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.110.1

andig, 2023-01-17:
Good report, thank you. I'll need to investigate.

andig, 2023-01-19:
> Looking at the logs provided below it can be seen that the event is emitted before the vehicle identification is completed. I think it is reasonable to assume that only after the vehicle identification the vehicleTitle is available and hence the variable expansion fails.

Genau. `vehicleTitle` ist nur dann verfügbar, wenn auch ein Fahrzeug zugeordnet ist. Bei Gastfahrzeug ist das z.B. nicht der Fall. Ein bisschen unglücklich ist, dass ein leerer Titel dazu führt, dass das Property aus `/api/state` verschwindet.

In Deinem Fall: 

```
[lp-1  ] INFO 2023/01/17 18:35:47 start charging ->
[push  ] ERROR 2023/01/17 18:35:47 invalid message template for connect: wanted: [vehicleTitle], got: ...
[main  ] DEBUG 2023/01/17 18:35:49 vehicle status: C (Enyaq)
[lp-1  ] INFO 2023/01/17 18:35:49 vehicle updated: unknown -> Enyaq
```

Der Ladestart passiert bevor das Auto erkannt wurde. 

> That said, my combination of charger (Easee) and car (Škoda Enyaq) does not allow direct vehicle detection via the Wallbox connection. 

Genau, kein EEbus, kein RFID.

Ich sehe ich allerdings nicht, was wir daran ändern könnten?

GrimmiMeloni, 2023-01-20:
Korrekt - an der Verzögerung beim Fahrzeugnamen lässt sich in diesem Szenario nichts ändern. In der Praxis ist das eigentliche Problem welches aus dieser Situation entsteht, dass der Message Handler nicht aufgerufen wird. Anders gesagt: ein fehlender Wert für eine gültige(!)  Template Variable verhindert am Ende eine Benachrichtigung.

Mein Vorschlag wäre für den Status "Gastfahrzeug" am Loadpoint einen Default vehicleTitle zu definieren. Es würde sich der Wert anbieten, der auch in der UI genutzt wird. Somit wäre die Variable immer verfügbar. Dann wäre zumindest dieser konkrete Fall gelöst.
Alternativ könnte geprüft werden ob der Templating Mechanismus anders konfiguriert werden kann, so daß er fehlende Werte bzw. unbekannte Variablen ignoriert statt mit einem Fehler auszusteigen. Das wäre dann eine allgemeine Lösung für diese Situation.

andig, 2023-01-20:
> Mein Vorschlag wäre für den Status "Gastfahrzeug" am Loadpoint einen Default vehicleTitle zu definieren. Es würde sich der Wert anbieten, der auch in der UI genutzt wird.

@naltatis was meinst Du? Im Backend kennen wir heute keine Sprachen. Ich bin nicht sicher, ob wir damit anfangen wollen.

andig, 2023-01-20:
Update: sollten wir nicht unbedingt tun. Es gibt einen guten Workaround unter Nutzung von Go Templates die auch für andere Fälle anwendbar sind (/cc @VolkerK62):

      msg: Started charging "{{if .vehicleTitle}}{{.vehicleTitle}}{{end}}"

closed 2023-01-20 by andig