evcc-io/evcc #5843: evcc gets stuck connecting to mqtt on startup

thomas-hutterer-tik, 2023-01-19:
### Describe the bug

evcc gets stuck connecting to mqtt on startup
connecting to the mosquitto broker with mosquitto_sub is possible without delay

### Steps to reproduce

1. sudo systemctl restart evcc
2. journalctl -f -u evcc

### Configuration details

```yaml
---
network:
  schema: "http"
  host: "0.0.0.0"
  port: 7070
interval: "2s"
sponsortoken: "..."
log: "info"
meters:
- name: "Test SmartPi"
  type: "custom"
  power:
    source: "mqtt"
    topic: "wa-pi/grid"
- name: "Senec Michael"
  type: "template"
  template: "senec-home"
  usage: "battery"
  host: "helios.ddnss.de:443"
- name: "Fronius Simulator"
  type: "template"
  template: "fronius-solarapi-v1"
  usage: "pv"
  host: "192.168.0.165:8083"
- name: "2nd Fronius for Evcc"
  type: "template"
  template: "fronius-solarapi-v1"
  usage: "pv"
  host: "192.168.0.165:8083"
chargers:
- name: "OCPP-1"
  type: "template"
  template: "ocpp"
  host: "192.168.2.143"
  id: 1
site:
  title: "Wien"
  meters:
    grid: "Test SmartPi"
    pvs:
    - "Fronius Simulator"
    - "2nd Fronius for Evcc"
    batteries:
    - "Senec Michael"
loadpoints:
- title: "OCPP-1"
  charger: "OCPP-1"
  phases: 3
  mode: "minpv"
mqtt:
  broker: "192.168.0.158"
  topic: "evcc"
  user: "..."
  password: "..."
```

### Log details

```text
Jan 19 20:13:13 smartpi systemd[1]: Started evcc.
Jan 19 20:13:18 smartpi evcc[22594]: [main  ] INFO 2023/01/19 20:13:18 evcc 0.111.1
Jan 19 20:13:18 smartpi evcc[22594]: [main  ] INFO 2023/01/19 20:13:18 using config file: /etc/evcc.yaml
Jan 19 20:13:18 smartpi evcc[22594]: [main  ] INFO 2023/01/19 20:13:18 starting ui and api at :7070
Jan 19 20:13:21 smartpi evcc[22594]: [db    ] INFO 2023/01/19 20:13:21 using sqlite database: /root/.evcc/evcc.db
Jan 19 20:13:21 smartpi evcc[22594]: [mqtt  ] INFO 2023/01/19 20:13:21 connecting evcc-409459007 at tcp://192.168.0.158:1883
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.111.1

andig, 2023-01-19:
Please check trace log for whats really going on

thomas-hutterer-tik, 2023-01-19:
Jan 19 21:07:27 smartpi systemd[1]: Started evcc.
Jan 19 21:07:37 smartpi evcc[1331]: [main  ] INFO 2023/01/19 21:07:37 evcc 0.111.1
Jan 19 21:07:37 smartpi evcc[1331]: [main  ] INFO 2023/01/19 21:07:37 using config file: /etc/evcc.yaml
Jan 19 21:07:37 smartpi evcc[1331]: [main  ] INFO 2023/01/19 21:07:37 starting ui and api at :7070
Jan 19 21:07:40 smartpi evcc[1331]: [db    ] INFO 2023/01/19 21:07:40 using sqlite database: /root/.evcc/evcc.db
Jan 19 21:07:40 smartpi evcc[1331]: [db    ] TRACE 2023/01/19 21:07:40 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
Jan 19 21:07:40 smartpi evcc[1331]: [db    ] TRACE 2023/01/19 21:07:40 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
Jan 19 21:07:40 smartpi evcc[1331]: [db    ] TRACE 2023/01/19 21:07:40 SELECT * FROM `settings` LIMIT 1 -1 <nil>
Jan 19 21:07:40 smartpi evcc[1331]: [db    ] TRACE 2023/01/19 21:07:40 SELECT * FROM `settings` 6 <nil>
Jan 19 21:07:40 smartpi evcc[1331]: [mqtt  ] INFO 2023/01/19 21:07:40 connecting evcc-1550975529 at tcp://192.168.0.158:1883
Jan 19 21:07:40 smartpi evcc[1331]: [mqtt  ] DEBUG 2023/01/19 21:07:40 tcp://192.168.0.158:1883 connected
Jan 19 21:07:41 smartpi evcc[1331]: [mqtt  ] TRACE 2023/01/19 21:07:41 recv wa-pi/grid: '7.63'
Jan 19 21:07:41 smartpi evcc[1331]: [ocpp  ] DEBUG 2023/01/19 21:07:41 waiting for chargepoint: 10m0s
Jan 19 21:07:42 smartpi evcc[1331]: [mqtt  ] TRACE 2023/01/19 21:07:42 recv wa-pi/grid: '8.11'

andig, 2023-01-19:
Its stuck at ocpp startup. Should timeout after 5 minutes.

closed 2023-01-19 by andig