evcc-io/evcc #5847: Fix Pracht Alpha XT mit Pracht PNI Modul

premultiply, 2023-01-19:
### Discussed in https://github.com/evcc-io/evcc/discussions/5846

<div type='discussions-op-text'>

<sup>Originally posted by **altibe** January 19, 2023</sup>
Hallo zusammen,

erstmal vielen Dank für das Megaprojekt. Hab bisher die SMA Wechselrichter ohne Probleme eingebunden bekommen.

Allerdings stellt mich die Wallbox vor ein Problem. Ich habe eine Pracht Alpha XT mit dem Pracht PNI Modul für die Anbindung ans Netzwerk. Ich komme auch über den Browser auf die Wallbox. 
Leider aber nicht per EVCC und Modbus.
Laut der Anleitung soll es damit funktionieren. Modbus ist aktiv in der Konfiguration:
Auszug aus der Anleitung
Die Modbus TCP Schnittstelle verwendet den Port 502. Dabei spricht das   standardmäßig die Modbus ID 1 an.

Wenn ich rs485tcpip in ivcc.yaml nutze bekomme ich folgende Meldung:

Charge status: read tcp 192.168.178.169:41002->192.168.178.199:502: i/o timeout
Enabled:           read tcp 192.168.178.169:41010->192.168.178.199:502: i/o timeout

Bei Nutzung von tcpip:
```
[main  ] INFO 2023/01/19 22:24:54 evcc 0.111.1 (4c35451a)
[main  ] INFO 2023/01/19 22:24:55 using config file: /etc/evcc.yaml
[db    ] INFO 2023/01/19 22:25:06 using sqlite database: /home/pi/.evcc/evcc.db
Charge status: modbus: exception '2' (illegal data address), function '4'
Enabled:       modbus: exception '2' (illegal data address), function '4'
```

Ist das Template ggf. nicht korrekt oder liegt es bei mir??

Jemand eine Idee??
</div>

premultiply, 2023-01-19:
Die Modbus-Operatoren in der Implementierung sind offensichtlich falsch.

03: Multiple Holding Registers (FC=03)
04: Input Registers (FC=04)

premultiply, 2023-01-19:
@altibe Ich bräuchte mal bitte einen Remotezugriff auf den Modbus-Port der Box.
Melde die bitte mal unter info@evcc.io

altibe, 2023-01-20:
Email hab ich geschickt. Mit der portweiterleitung auf die Wallbox

andig, 2023-01-20:
Umklassifiziert. Ich sehe keinen Bug in evcc ;)

premultiply, 2023-01-20:
Danke @altibe für den Remotezugang.
Probier mal bitte ab morgen das Nightly Build.

closed 2023-01-20 by premultiply

altibe, 2023-01-20:
Danke. Auch schonmal. Ich Probier es! Dann gibt es Rückmeldung.

altibe, 2023-01-21:
Nochmals Danke. Hat funktioniert. Kurze Frage: wie lese und steuere ich die beiden Ladepunkte in der Box an. Welche Befehle brauch ich dafür?

PR #5854 (2023-01-20): Fix Pracht-Box with PNI
Fixes #5847