evcc-io/evcc #5852: OpenEVSE charge power problem

ioniks, 2023-01-20:
### Describe the bug

In Solar mode, When enough solar power available, the power is progressively set to max charge power.

### Steps to reproduce

1. Connect the vehicle in solar mode
2. Set Manual mode on OpenEVSE

### Configuration details

```yaml
vehicles:
- type: template
  template: renault
  title: Zoe Florian
  user: XXXXX
  password: XXXX
  vin: XXXXX
  capacity: 41
  name: zoeFlorian
  mode: pv
  minCurrent: 8
- type: template
  template: renault
  title: Zoe Noémie
  user: XXXXX
  password: XXXX
  vin: XXX
  capacity: 22
  name: zoeNoemie
  mode: pv
  minCurrent: 8

meters:
- type: template
  template: solax-hybrid-cloud
  usage: grid
  tokenid: "XXXXX"
  serial: XXXXX
  name: grid
- type: template
  template: solax-hybrid-cloud
  usage: pv
  tokenid: "XXX"
  serial: XXX
  name: pv1
- type: template
  template: solax-hybrid-cloud
  usage: battery
  tokenid: "XXXXX"
  serial: XXXX
  name: battery

chargers:
- type: template
  template: openevse
  host: 192.168.0.27
  name: wallbox

loadpoints:
- title: Maison
  charger: wallbox
  mode: pv
  phases: 1
  mincurrent: 8
  maxcurrent: 32
  resetOnDisconnect: true
  vehicles: 
  - zoeNoemie
[... 845 lines cut ...]
[site  ] DEBUG 2023/01/20 11:02:40 grid power: -0W
[site  ] DEBUG 2023/01/20 11:02:40 site power: -2277W
[lp-1  ] DEBUG 2023/01/20 11:02:40 charger status: B
[lp-1  ] INFO 2023/01/20 11:02:40 stop charging <-
[lp-1  ] DEBUG 2023/01/20 11:02:40 soc estimated: 32.64% (vehicle: 32.00%)
[lp-1  ] DEBUG 2023/01/20 11:02:40 vehicle soc: 32%
[lp-1  ] DEBUG 2023/01/20 11:02:40 vehicle range: 66km
[site  ] DEBUG 2023/01/20 11:02:45 ----
[lp-1  ] DEBUG 2023/01/20 11:02:45 charge power: 0W
[site  ] DEBUG 2023/01/20 11:02:45 pv power: 2991W
[site  ] DEBUG 2023/01/20 11:02:45 battery soc: 41%
[site  ] DEBUG 2023/01/20 11:02:45 battery power: -2477W
[site  ] DEBUG 2023/01/20 11:02:45 grid power: -0W
[site  ] DEBUG 2023/01/20 11:02:45 site power: -2277W
[lp-1  ] DEBUG 2023/01/20 11:02:45 charger status: B
[lp-1  ] DEBUG 2023/01/20 11:02:45 next soc poll remaining time: 14m55s
```

### What type of operating system are you running?

Docker container

### Version

0.111.1

premultiply, 2023-01-20:
You are using a cloud-service which is far to slow for regulation loop.
Use a local interface to the grid and battery meter.

closed 2023-01-20 by premultiply

ioniks, 2023-01-20:
@premultiply i don't understand why this is possible if this can't be work ?

it's not possible to add timing delay for getting new information from cloud ?

Thanks

premultiply, 2023-01-20:
Your car, battery inverter and other consumption will change their conumption in parts of seconds so there is no time to delay anything for minutes or hours until a undefined cloud service may be updated.