evcc-io/evcc #5866: seit einiger Zeit werden keine Warnungen/Fehler im UI angezeigt.

VolkerK62, 2023-01-21:
Ich bekomme seit einiger Zeit keine Warnungen/Fehler mehr im UI angezeigt. 
Sind nur im Logfile sichtbar. 
Habe auch schon verschiedene Browser probiert, sowohl auf dem Laptop, als auf dem Raspi.

Hat das schon mal jemand gehabt?

_Originally posted by @VolkerK62 in https://github.com/evcc-io/evcc/discussions/5855_

andig, 2023-01-21:
Was steht in /api/state wenn ein Fehler im Logfile ist?

VolkerK62, 2023-01-21:
Netzwerkkabel von der Wallbox getrennt und Ladung gestartet.

LogFile:
```
Jan 21 19:11:26 raspberrypi evcc[22223]: [daheim] TRACE 2023/01/21 19:11:26 POST https://api.daheimladen.com/v1/cs/229461813211317/change_config
Jan 21 19:11:27 raspberrypi evcc[22223]: [daheim] TRACE 2023/01/21 19:11:27 {"key":"ChargeRate","value":"25"}
Jan 21 19:11:27 raspberrypi evcc[22223]: --
Jan 21 19:11:27 raspberrypi evcc[22223]: error invoking changeconfiguration
Jan 21 19:11:27 raspberrypi evcc[22223]: [lp-1  ] ERROR 2023/01/21 19:11:27 max charge current 25A: unexpected status: 500
Jan 21 19:11:30 raspberrypi evcc[22223]: [site  ] DEBUG 2023/01/21 19:11:30 ----
```

/api/state:
`{"result":{"auth":{"vehicles":{}},"battery":[{"power":-45.33720930232558,"soc":40,"capacity":7.5}],"batteryCapacity":8,"batteryConfigured":true,"batteryPower":-45.33720930232558,"batterySoc":40,"bufferSoc":100,"currency":"EUR","gridConfigured":true,"gridCurrents":[0.75,1.72,0.88],"gridEnergy":583.1971,"gridPower":605.9200000000001,"homePower":560.5827906976745,"loadpoints":[{"chargeCurrent":0,"chargeCurrents":[0,0,0],"chargeDuration":0,"chargePower":0,"chargeRemainingDuration":0,"chargeRemainingEnergy":3200,"chargeTotalImport":36.5,"chargedEnergy":0,"charging":false,"connected":true,"connectedDuration":0,"enabled":false,"maxCurrent":25,"minCurrent":6,"minSoc":0,"mode":"now","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":null,"phasesEnabled":1,"planActive":false,"pvAction":"inactive","pvRemaining":0,"targetEnergy":0,"targetSoc":90,"targetTime":"0001-01-01T00:00:00Z","title":"Carport","vehicleCapacity":72,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleIcon":"car","vehicleOdometer":12479.8,"vehiclePresent":true,"vehicleRange":332,"vehicleSoc":86,"vehicleTargetSoc":100,"vehicleTitle":"(T)Raumschiff"}],"prioritySoc":35,"pv":[{"power":0}],"pvConfigured":true,"pvPower":0,"residualPower":50,"savingsAmount":5.567938016495842,"savingsEffectivePrice":0.2960993978811769,"savingsGridCharged":98.84171980511573,"savingsSelfConsumptionCharged":22.030499408458518,"savingsSelfConsumptionPercent":18.22627196869108,"savingsSince":1670934918,"savingsTotalCharged":120.87221921357425,"siteTitle":"Haus","sponsor":"VolkerK62","tariffFeedIn":0.1095,"tariffGrid":0.424,"vehicles":["(T)Raumschiff","offline"],"version":"0.111.1 (f38353bf)"}}`

andig, 2023-01-22:
> error invoking changeconfiguration

Was ist das denn?

andig, 2023-01-22:
@naltatis kannst Du das nachvollziehen? Ich kann zumindest wenn ich ein 

   	valueChan <- util.Param{Key: "error", Val: "foo"}

einbaue auf die Schneller auch keinen Fehler im UI sehen?

andig, 2023-01-22:
Ich hab jetzt bis v0.90, noch mit dem uralten UI, versucht einen Fehler im UI anzuzeigen und es gelingt mir nicht. Ohne Positivtestcase- egal auf welcher Version- wird das schwierig :(

andig, 2023-01-22:
Hab den Fehler gefunden. Stay tuned.

andig, 2023-01-22:
Closed in https://github.com/evcc-io/evcc/commit/3faef17ffa65ea5742d7e5eb9df6b83445369d31

closed 2023-01-22 by andig