evcc-io/evcc #5887: Climater Status not receiving any more

hlathome, 2023-01-23:
### Describe the bug

The climater status used to come in, but it does not any more.
vehicle is Audi e-tron, using the template "etron".  The other params come in as desired.

/app # evcc vehicle
[main  ] INFO 2023/01/23 14:17:37 evcc 0.111.1
[main  ] INFO 2023/01/23 14:17:37 using config file: /etc/evcc.yaml
Soc:           99%
Capacity:      64.0kWh
Charge status: C
Range:         215km
Odometer:      20595km
Finish time:   2023-01-23 14:20:00 +0100 CET
Climater:      missing climatisation status
Target Soc:    100%

### Steps to reproduce

see description

### Configuration details

```yaml
see description
```

### Log details

```text
see description
```

### What type of operating system are you running?

Linux

### Version

evcc 0.111.1

andig, 2023-01-23:
Probier mal bitte die nightly Version. Angefragt wird der Klimastatus.

hlathome, 2023-01-23:
leider nein: Image ist jetzt andig/evcc:nightly@sha256:f8889f7ba323725e7e18b0d6af90c0364f3f9a8889bd9974e3bf855f509b9804]

Docker:
  evcc:
    image: andig/evcc:nightly
    container_name: evcc
...

Output:
/app # evcc vehicle
[main  ] INFO 2023/01/23 14:52:21 evcc 0.111.1 (d1d88295)
[main  ] INFO 2023/01/23 14:52:21 using config file: /etc/evcc.yaml
Soc:           99%
Capacity:      64.0kWh
Charge status: B
Range:         215km
Odometer:      20595km
Finish time:   2023-01-23 14:17:00 +0100 CET
Climater:      missing climatisation status
Target Soc:    100%

andig, 2023-01-23:
Sieht aus als käme der- selbst wenn explizit angefordert- aktuell einfach nicht mit. Da hilft nur warten auf VAG. Es sei denn, Du kennt eine andere Integration die das Problem nicht hätte?

hlathome, 2023-01-24:
Nein kenne ich leider nicht, auch nach viel Sucherei durch etlich Python Projekte im Github lande ich immer wieder bei Euch :-). Gut dann schaun' mer mal. Danke jedenfalls.

closed 2023-01-24 by andig