evcc-io/evcc #5890: evcc.yaml soc / poll / mode: charging prüft permanent den akku status

ako-3004, 2023-01-23:
### Describe the bug

Ich hab in letzter Zeit das Log beobachtet, da ich den Verdacht hatte dass EVCC den SOC permanent abruft, obwohl nur mode charging in der evcc.yaml eingetragen ist. Hoffe ich hab nicht nur einen Fehler in der config (siehe unten). Im log taucht der request alle 2 Stunden auf, unabhängig ob geladen wird oder nicht. Er requestet also permanent, nicht nur während des Ladens.

### Steps to reproduce

config wie unten, mode charging, intervall 30m bis 120m

### Configuration details

```yaml
loadpoints:
- title: Garage links (Kia)
  charger: wallbox6
  vehicle: ev3
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
  soc:
    poll:
      mode: charging
      interval: 120m
    estimate: true
```
```

### Log details

```text
Jan 23 07:20:51 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 07:20:51 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/xxxxxxxxx/status/latest
Jan 23 07:20:51 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 07:20:51 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":48.147736,"lon":8.543486,"alt":0,"type":0},"h
Jan 23 07:20:51 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 07:20:51 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/xxxxxxxxx/status
Jan 23 07:21:06 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 07:21:06 {"retCode":"S","resCode":"0000","resMsg":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft"
Jan 23 09:21:01 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 09:21:01 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/xxxxxxxxx/status/latest
Jan 23 09:21:01 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 09:21:01 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":48.147736,"lon":8.543486,"alt":0,"type":0},"h
Jan 23 09:21:01 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 09:21:01 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/xxxxxxxxx/status
Jan 23 09:21:17 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 09:21:17 {"retCode":"S","resCode":"0000","resMsg":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft"
Jan 23 11:21:01 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 11:21:01 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/xxxxxxxxx/status/latest
Jan 23 11:21:02 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 11:21:02 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":48.147736,"lon":8.543486,"alt":0,"type":0},"h
Jan 23 11:21:02 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 11:21:02 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/xxxxxxxxx/status
Jan 23 11:21:37 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 11:21:37 {"retCode":"F","resCode":"5031","resMsg":"Unavailable remote control - Service Temporary Unavailable","msgId":"80715dbe-e969-4b98-b6e7-6a7dc29
Jan 23 13:21:11 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 13:21:11 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/xxxxxxxxx/status/latest
Jan 23 13:21:12 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 13:21:12 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":48.147736,"lon":8.543486,"alt":0,"type":0},"h
Jan 23 15:21:21 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 15:21:21 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/xxxxxxxxx/status/latest
Jan 23 15:21:21 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 15:21:21 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":48.147736,"lon":8.543486,"alt":0,"type":0},"h
Jan 23 15:21:21 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 15:21:21 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/xxxxxxxxx/status
Jan 23 15:21:35 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 15:21:35 {"retCode":"S","resCode":"0000","resMsg":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft"
Jan 23 17:21:26 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 17:21:26 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/xxxxxxxxx/status/latest
Jan 23 17:21:27 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 17:21:27 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{"lat":48.147736,"lon":8.543486,"alt":0,"type":0},"h
Jan 23 17:21:27 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 17:21:27 GET https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles/xxxxxxxxx/status
Jan 23 17:21:40 raspberrypi evcc[2327]: [kia   ] TRACE 2023/01/23 17:21:40 {"retCode":"S","resCode":"0000","resMsg":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft"
```

### What type of operating system are you running?

Linux

### Version

0.111.1

tobiasbayer, 2023-01-23:
Die [Doku](https://docs.evcc.io/docs/reference/configuration/loadpoints#pollinterval) sagt dazu:
"`poll.interval`: Definiert, wie oft das Fahrzeug nach neuen Daten abgefragt wird, wenn es **NICHT** lädt."

tobiasbayer, 2023-01-23:
> ```yaml
> mode: charging
>       interval: 120m
> ```

Wobei das nach mehrmaligem Lesen tatsächlich missverständlich sein könnte.
Kann vielleicht @andig was dazu sagen, wie das gedacht ist?

tobiasbayer, 2023-01-23:
Ich habe es mir mal im [Code](https://github.com/evcc-io/evcc/blob/0f07cd612c42c309c34aff404e349dbcf387afdd/core/loadpoint.go#L1601) angesehen.
Hast du in deinem Log auch Ausgaben der Form "next soc poll remaining time:" stehen?

andig, 2023-01-24:
@ako-3004 für eine Analyse brauchen wir Deine Config. In diesem Fall insbesondere die vom Auto. Gerne aktualisieren, dann mache ich hier wieder auf.

closed 2023-01-24 by andig

ako-3004, 2023-01-24:
Okay danke erst mal, der erste Test den ich mache: Stecker ziehen, mal schauen was passiert wenn er nicht lädt. Grade eben gezogen.
Der SoC wird auf jeden Fall richtig angezeigt, ich sehe auch im Log keine Fehler von der Kia-Schnittstelle bisher.
Ein grep mit soc oder poll oder remaining liefert gar keine ergebnisse (sudo cat /var/log/syslog | grep poll)...

Ich muss da zustimmen, nach der Doku verstehe ich es so als würde poll.mode = charging dafür sorgen dass nur während des Ladens gelesen wird. Im gleichen Schritt tut sich dann die Frage auf, wenn poll.intervall den Zyklus angibt während des Nicht-ladens, wie ist dann der Zyklus während des Ladens?

Hinsichtlich log bin ich grad verwirrt :D ich hab gestern auf 240m intervall umgestellt, seither gar keine Einträge mehr im Log (war angesteckt, nicht gefahren, nicht geladen).
Das Log ist mit grep auf "kia" reduziert, endet gestern Abend:
https://mega.nz/file/i3RynT7I#2CjngRjzX1KsXrw1F8XTmQxlXGC_mj-bDNdvHVsKAmI

Die EVCC yaml (ohne pers. Daten) ist aktuell diese:
https://mega.nz/file/ezRTAQjL#hG57JUAg6y9cLnpX6jv4tiMZ7e0WYQB8-XaVgaIDJDw

(Hintergrund des Ganzen ist der: mir ging über Weihnachten die 12V Batterie leer, ich hab mittlerweile einen Microcontroller zur Spannungsüberwachung an der 12V Batterie)

andig, 2023-01-24:
> @ako-3004 für eine Analyse brauchen wir Deine Config. In diesem Fall insbesondere die vom Auto. Gerne aktualisieren, dann mache ich hier wieder auf.

ako-3004, 2023-01-25:
@andig die config yaml hab ich oben als download verlinkt...

Die EVCC yaml (ohne pers. Daten) ist aktuell diese:
https://mega.nz/file/ezRTAQjL#hG57JUAg6y9cLnpX6jv4tiMZ7e0WYQB8-XaVgaIDJDw

andig, 2023-01-25:
Du kannst Dateien direkt hier anhängen, dann müssen wir nicht zu Kim Schmitz. Ich seh in der Config nichts was eine Abfrage alle 15s rechtfertigt. Allerdings fehlt in Deinem Log auch jeglicher Kontext wann/wo/wie das passiert. So ist das leider nicht verwertbar. 

Die komplette poll: Geschichte kannst Du Dir eigentlich sparen weil du mehr oder weniger das Standardverhalten willst. Zur Fehlersuche würde ich vorschlagen mit minimalster Config zu starten (siehe evcc.dist.yaml) und nur die *unbedingt*  notwendigen Ergänzungen zu machen. Dann hast Du deutlich weniger Fehlerquellen.

ako-3004, 2023-01-31:
Danke, ich hab die poll configuration mal komplett auskommentiert, und bei log (ähnlich der evcc.dist.yaml) die beiden lp und die fahrzeuge auf debug gesetzt. Hoffe das war richtig. Mal sehen was die nächsten Tage passiert. Konnte heute zumindest sehen, dass er auch die letzten tage immer wieder die zyklische Abfrage aktiviert, außerhalb des Ladevorgangs. Ich melde mich nach ein paar Tagen wieder.

ako-3004, 2023-02-07:
Hi @andig , ich hatte jetzt einige Tage die poll configuration entfernt (also default Verhalten?). Er greift nach wie vor zyklisch aufs Fahrzeug zu, dabei ist egal ob geladen wird oder nicht. Interessanterweise hört er auch mal für einen Tag zwischendrin einfach so auf, auch das hab ich gesehen, irgendwann fängt er wieder an (wenn, dann nach einer Zustandsänderung, also noch einer Fahrt oder einem Ladevorgang).

Ich hab einen unkonventionellen Weg zu schauen ob er aufs Fahrzeug zugreift, ich sehs am 12Batterie-Log. (Wenn das Fzg aufwacht bricht die Spannung ein, ist sehr einfach zu sehen, in Orange ein Ladevorgang). Ist aber auch im syslog nachzuvollziehen.

[image]

Ansonsten hab ich Euch mal die aktuelle evcc yaml angehängt. Und die Logs (sorry sind einige MB). Kann die gern auf irgendwas kleiner greppen.

[evcc.yaml.txt](https://github.com/evcc-io/evcc/files/10678732/evcc.yaml.txt)

[Fehlerprotokoll 2.zip](https://github.com/evcc-io/evcc/files/10678742/Fehlerprotokoll.2.zip)

Danke, Alex

ako-3004, 2023-02-07:
PS: wie @tobiasbayer oben schreibt könnte die Doku missverständlich sein. Für mich steht lt. der Doku für poll.mode = charging fest, dass nur während des Lades geprüft wird, für poll.mode = connected, always darf auch ausserhalb des Ladens geprüft werden. Sollte das so falsch verstanden sein bitte um Info und Anpassung der Doku. Ich hatte noch nie was anderes als poll.mode = charging, daher dürfte ein Poll ausserhalb des Ladens also nach meinem Verständnis gar nie stattfinden.

andig, 2023-02-08:
> daher dürfte ein Poll ausserhalb des Ladens also nach meinem Verständnis gar nie stattfinden.

Das ist auch so. Wenn nein ists ein Bug. @nename0 Du hattest Dir das in https://github.com/evcc-io/evcc/pull/5991 intensiv angeschaut. Ist da noch ein Fehler drin der das Fahrzeug zu oft aufweckt?

@ako-3004 ich hab leider keine Zeit die Logs zu checken. Wenn Du einen konkreten Fehler siehst wo zu oft abgefragt wird bitte Logzeile/Timestamps rausziehen.

ako-3004, 2023-02-08:
Alles klar, ich fuchse mich mal in die Logs rein.
Wallbox ist übrigens der Go E.

Wenn ich irgendwas weiter testen kann sagt mir bescheid.

nename0, 2023-02-08:
Ok, dann gebe ich auch mal meinen Senf dazu. 
Also zuzunächst die technische Erklärung wie die Bluelink API funktioniert damit wir alle auf dem gleichen Stand sind:

- `.../status/latest` damit kann man den letzten Zustand den das Auto an den Bluelink Server gesendet hat holen. Dabei wird das Auto nicht involviert, wird also auch nicht aufgeweckt. Das Auto sendet aber auch von sich aus Zustände z.B. wenn man das Auto abstellt oder eine Ladung beendet wird. Dann hat man nach dem Abstellen des Autos direkt den letzten SOC in der App. In den App sieht man das immer als aktuellen Zustand inklusive dem Zeitstempel von dem der Zustand stammt.
- `.../status` damit zwingt man das Auto einen neuen Zustand zu liefern. Dabei wird das Auto aufgeweckt. In der App geht das über den Refresh-Knopf oben rechts.

Nun zum SOC-Poll von EVCC. Da scheint die Dokumentation einige Leute zu verwirren. Folgende Default-Konfiguration
```
poll:
        mode: charging
        interval: 60m
```
führt dazu, dass:

- Wenn das Auto lädt wird alle 10s der SOC abgefragt. [Immer](https://github.com/evcc-io/evcc/blob/54321c0cf72ceb0668adf8d58cf985cc0142aab8/core/loadpoint.go#L1316)! Das wird aber durch das Caching von 15min in den Vehicle-Providern wieder entschärft.
- Wenn das Auto nicht lädt aber angesteckt ist, wird alle 60 min der SOC abgefragt. **Das verstehen die meisten Leute falsch**, weil eben ein `mode: charging` doch dazu führt das der SOC auch ohne Ladevorgang geholt wird. 
- Wenn das Auto nicht angesteckt ist, wird der  SOC nicht abgefragt.

So nun zur Bluelink Anbindung von EVCC: Hier scheint es irgendein Szenario zu geben bei dem das Obige nicht funktioniert und man sehr viele 503er bekommt. Das passiert aber meist nur direkt nach dem Starten von EVCC.
Ich habe da die returns von `api.ErrMustRetry` in Verdacht, da diese nämlich das Caching von 15min komplett abschalten.

Ich muss aber zugeben in den Logs von @ako-3004 sehe ich gerade keine Situation wo die API zu häufig benutzt wird....

nename0, 2023-02-08:
Also eig müsste man den `mode: charging` umbenennen in `mode: connectedButBelowTargetSOC`

tobiasbayer, 2023-02-08:
Das Verhalten ist aber dann schon hart an der Doku vorbei. 

"`charging`: aktualisiere die Daten NUR während eines Ladevorgangs (dies ist die empfohlene Standardeinstellung)
`connected`: aktualisiere die Daten, wenn das Fahrzeug mit der Wallbox verbunden ist (nicht nur wenn es lädt); der Parameter interval definiert wie oft"

Da erwartet doch niemand, dass das Fahrzeug dann in "charging" trotzdem wie bei "connected" abgefragt wird, nur weil man ein `interval` angegeben hat.

Aber das gehört eigentlich nicht in dieses Issue.

andig, 2023-02-08:
> Da erwartet doch niemand, dass das Fahrzeug dann in "charging" trotzdem wie bei "connected" abgefragt wird, nur weil man ein interval angegeben hat.

Und genau so ist es halt auch nicht!

nename0, 2023-02-08:
> > Da erwartet doch niemand, dass das Fahrzeug dann in "charging" trotzdem wie bei "connected" abgefragt wird, nur weil man ein interval angegeben hat.
> 
> Und genau so ist es halt auch nicht!

Dann erkläre mir diese Zeile: 
https://github.com/evcc-io/evcc/blob/54321c0cf72ceb0668adf8d58cf985cc0142aab8/core/loadpoint.go#L1310

tobiasbayer, 2023-02-08:
Sieht für mich auch nach eine, Bug aus. Fahrzeug gesteckt, aber nicht bis zum Target SoC geladen, interval = 60m, cache = 15m führt durch die beiden verlinkten Zeile dazu, dass das Fahrzeug alle Stunde aufgeweckt wird.

nename0, 2023-02-08:
Scheint als wurde das hier eingeführt: https://github.com/evcc-io/evcc/commit/05976b479d7c34c7b084a8c8aae72c3f7fe7f227

nename0, 2023-02-08:
Ich muss aber auch sagen persönlich von finde ich diese Verhalten nicht schlecht und benutze es mit interval auf 150 statt 60 min. Es ist halt nur nirgends dokumentiert...