evcc-io/evcc #5896: NRGKick Connect Out-of-Sync Error

maatinh, 2023-01-24:
Hi,
I do get an error when using the NRGKick Connect as a charger:

[lp-1  ] WARN 2023/01/24 13:08:07 charger out of sync: expected disabled, got enabled
[lp-1  ] ERROR 2023/01/24 13:08:07 charger: response: The request contains an entity body but no Content-Type header. The inferred media type 'application/octet-stream' is not supported for this resource.
[lp-1  ] DEBUG 2023/01/24 13:08:07 pv charge current: 0A = 0A + -14.1A (3242W @ 1p)

EVCC can read Charge-Current correctly from NRGKick.

I can change charging-status, charging-current using FHEM and JSON API, so its not an issue with NRGKick/Connect/API.

premultiply, 2023-01-24:
`evcc charger --log trace` ?

andig, 2023-01-24:
Sollte im Nightly in 15min gelöst sein, bitte nochmal testen

maatinh, 2023-01-24:
Der HTTP-Header Fehler ist weg. Aber korrekt ausgelesen wird der Ladestatus noch nicht:

Direkt nach Start - Auto lädt **nicht**:
[site  ] DEBUG 2023/01/24 14:03:03 ----
[lp-1  ] DEBUG 2023/01/24 14:03:03 charge power: 0W
[site  ] DEBUG 2023/01/24 14:03:03 pv power: 357W
[site  ] DEBUG 2023/01/24 14:03:03 grid power: -128W
[site  ] DEBUG 2023/01/24 14:03:03 site power: -128W
[lp-1  ] DEBUG 2023/01/24 14:03:03 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/01/24 14:03:03 charger status: C
[lp-1  ] DEBUG 2023/01/24 14:03:03 charge start energy: 2.87e+03kWh
[lp-1  ] INFO 2023/01/24 14:03:03 car connected
[lp-1  ] DEBUG 2023/01/24 14:03:03 pv timer elapse
[lp-1  ] DEBUG 2023/01/24 14:03:03 pv timer inactive
[lp-1  ] DEBUG 2023/01/24 14:03:03 charge start energy: 2.87e+03kWh
[lp-1  ] INFO 2023/01/24 14:03:03 start charging ->
[lp-1  ] DEBUG 2023/01/24 14:03:04 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1  ] DEBUG 2023/01/24 14:03:04 vehicle soc: 0%
[lp-1  ] WARN 2023/01/24 14:03:04 charger logic error: disabled but charging
[lp-1  ] ERROR 2023/01/24 14:03:04 charger: EOF
[lp-1  ] DEBUG 2023/01/24 14:03:04 pv charge current: 0.557A = 0A + 0.557A (-128W @ 1p)
[lp-1  ] DEBUG 2023/01/24 14:03:04 pv enable timer reset
[lp-1  ] DEBUG 2023/01/24 14:03:04 pv timer inactive
[site  ] DEBUG 2023/01/24 14:03:13 ----

Dann Ladevorgang manuell gestartet:

[site  ] DEBUG 2023/01/24 14:07:33 ----
[lp-1  ] DEBUG 2023/01/24 14:07:33 charge power: 3350W
[site  ] DEBUG 2023/01/24 14:07:34 pv power: 370W
[site  ] DEBUG 2023/01/24 14:07:34 grid power: 3323W
[site  ] DEBUG 2023/01/24 14:07:34 site power: 3323W
[lp-1  ] DEBUG 2023/01/24 14:07:34 charge currents: [15.4 0 0]A
[lp-1  ] DEBUG 2023/01/24 14:07:34 detected active phases: 1p
[lp-1  ] DEBUG 2023/01/24 14:07:34 charger status: C
[lp-1  ] DEBUG 2023/01/24 14:07:34 next soc poll remaining time: 59m48s
[lp-1  ] DEBUG 2023/01/24 14:07:34 soc estimated: 0.53% (vehicle: 0.00%)
[lp-1  ] DEBUG 2023/01/24 14:07:34 vehicle soc: 0%
[lp-1  ] WARN 2023/01/24 14:07:34 charger out of sync: expected disabled, got enabled
[lp-1  ] ERROR 2023/01/24 14:07:35 charger: EOF
[lp-1  ] DEBUG 2023/01/24 14:07:35 pv charge current: 0A = 0A + -14.4A (3323W @ 1p)
[lp-1  ] ERROR 2023/01/24 14:07:35 max charge current 7A: EOF
[site  ] DEBUG 2023/01/24 14:07:43 ----

[lp-1  ] DEBUG 2023/01/24 14:07:53 charge power: 3330W
[site  ] DEBUG 2023/01/24 14:07:53 pv power: 370W
[site  ] DEBUG 2023/01/24 14:07:53 grid power: 3323W
[site  ] DEBUG 2023/01/24 14:07:53 site power: 3323W
[lp-1  ] DEBUG 2023/01/24 14:07:53 charge currents: [15.5 0 0]A
[lp-1  ] DEBUG 2023/01/24 14:07:53 detected active phases: 1p
[lp-1  ] DEBUG 2023/01/24 14:07:53 charger status: C
[lp-1  ] DEBUG 2023/01/24 14:07:53 next soc poll remaining time: 59m58s
[lp-1  ] DEBUG 2023/01/24 14:07:53 soc estimated: 0.60% (vehicle: 0.00%)
[lp-1  ] DEBUG 2023/01/24 14:07:53 vehicle soc: 0%
[lp-1  ] WARN 2023/01/24 14:07:53 charger out of sync: expected disabled, got enabled
[lp-1  ] ERROR 2023/01/24 14:07:53 charger: EOF

Ändern des Stroms funktioniert auch noch nicht.

In FHEM nutze ich folgendes JSON um den Strom zu ändern:
{"Values": {"ChargingCurrent": {"Value": "$val" },"DeviceMetadata":{"Password":"XXXX"}}}

$val = 7,8,9...16

Und folgendes um das Laden zu starten/stoppen:
{"Values": {"ChargingStatus": {"Charging": "$val" },"DeviceMetadata":{"Password":"XXXX"}}}

$val = true|false

andig, 2023-01-24:
@maatinh könntest Du bitte mal das `--log trace` erstellen? Ich vermute, die Parameter müssen als string codiert sein, werden bei uns aber als native Werte raus geschickt. Blick aufs Logfile sollte das bestätigen.

andig, 2023-01-24:
Die Werte sind alle nativ. In Deinem Beispiel oben:

    {"Values": {"ChargingStatus": {"Charging": "$val" },"DeviceMetadata":{"Password":"XXXX"}}}

ist das true/false aber als String codiert. Nativ haben wir es heute im Code- der scheint aber bei Dir nicht zu funktionieren. Könntest Du evtl. mal Remotezugang für info@evcc.io herstellen (Portweiterleitung)?

maatinh, 2023-01-24:
Es funktioniert auch native. Jedenfalls aus FHEM:
```

2023.01.24 17:58:53 4: http_nrgkick: HandleSendQueue sends set03 with timeout 2 to http://192.168.40.132/api/settings/00:1E:C0:59:2E:BD, 
data: {"Values": {"ChargingStatus": {"Charging": true },"DeviceMetadata":{"Password":"XXXX"}}}, 
header: Content-Type: application/json
2023.01.24 17:58:53 5: http_nrgkick: ReadCallback called from __ANON__
2023.01.24 17:58:53 4: http_nrgkick: Read callback: request type was set03 retry 0,
header: HTTP/1.1 202 Accepted
Server: Mono-HTTPAPI/1.0
Date: Tue, 24 Jan 2023 16:58:53 GMT
Content-Length: 0
Connection: close, no body
2023.01.24 17:58:53 5: http_nrgkick: Read callback: body empty
```

Portweiterleitung auf das Host-OS/SSH (evcc läuft im docker-container) oder die EVCC UI?

andig, 2023-01-24:
Portweiterleitung auf den Connect

andig, 2023-01-24:
Ja gerne:

> Remotezugang für [info@evcc.io](mailto:info@evcc.io)

andig, 2023-01-24:
@maatinh mir fällt grad auf, dass die Implementierung niemals abgeschlossen wurde; die Statuserkennung ist gar nicht umgesetzt. Hast Du eine Idee, wie "Fahrzeug verbunden" im API zu erkennen wäre?

maatinh, 2023-01-24:
Leider scheint die API das nicht herzugeben. Habe ich auch schon geschaut. Info_Connected bezieht sich nur auf die Verbindung zwischen Connect und NRGKick Ladegerät. Online bezieht sich auf den Online-Status des Connect.

closed 2023-01-24 by andig