evcc-io/evcc #5908: fehlende UI zur Auswertung Ladeenergie

StevieC121176, 2023-01-26:
### Describe the bug

in der Version `0.111.1 (32f69a01)` fehlt die Anzeige unten rechts zur Auswertung der Ladeenergie %Sonnenenergie

[image]

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

naltatis, 2023-01-27:
@andig Hier landen alten UI Assets im aktuellen Nightly Build. [Diese Zeile](https://github.com/evcc-io/evcc/blame/master/assets/js/components/Savings.vue#L126) wurde vorgestern geändert, ist aber noch in seiner "alten Form" im Build. Ich vermute hier ärgert und das Caching in den Pipelines. Lass uns die assets im Release immer frisch bauen.

andig, 2023-01-27:
Ich ändere das caching auf commit

andig, 2023-01-27:
Sollte behoben sein- bitte in 15min das Nightly Build probieren. Danke!

closed 2023-01-27 by andig

naltatis, 2023-01-29:
Works for me ✅